sparkfun/OpenLog_Artemis #2: OpenLog Artemis - Data Logging Issues

adamgarbo, 2020-04-21:
### Subject of the issue
I've encountered a number of issues when starting to work with the OpenLog Artemis and setting the log rate to longer values (i.e. 5-10 seconds). The two main issues are:
1. OpenLog Artemis does not log data beyond the first observation.
2. OpenLog Artemis does not power down after 2 seconds.

### Your workbench
* What development board or microcontroller are you using?
    * OpenLog Artemis
* What version of hardware or breakout board are you using?
    * Firmware v1.1
* How is OpenLog Artemis wired to your sensor(s)?
    * Directly via USB-C
* How is everything being powered?
    * USB-C or 3.7V LiPo battery
* Are there any additional details that may help us help you?
    * See below for raw data outputs.

### Steps to reproduce
* Connect to OpenLog Artemis over Serial and in the menu change "Set Log Rate in seconds between readings: " to 5 seconds (0.2 Hz) or 10 seconds (0.1 Hz)

### Expected behaviour
* OpenLog Artemis should log data every 5-10 seconds and provide a visual indication of samples being recorded via the STAT LED.
* OpenLog Artemis should power down after 2 seconds of inactivity and have a sleep current of ~250 uA.

### Actual behaviour
* Only the first data sample is recorded to the microSD card. All subsequent data is not recorded.
* Quiescent current does not decrease after 2 s of inactivity and remains at ~1 mA (SD card dependent). 
* When powering the OpenLog Artemis with either a 3.7V LiPo battery or 5V from USB-C, there is brief LED activity after the 45 second timeout but the STAT LED does not blink beyond that.

### Raw data
Contents of dataLog00001.txt

```
SD card online
Data logging online
Serial logging online
IMU online
No sensors detected
rtcDate,rtcTime,aX,aY,aZ,gX,gY,gZ,mX,mY,mZ,imu_degC,output_Hz,
04/21/2020,09:08:15.28,-9.77,-19.04,-1000.00,-1.92,2.16,-1.11,-10.05,87.45,12.90,26.13,333.33,
```

Output of Serial Monitor:
```
09:07:26.486 -> 
09:07:26.660 -> Config file read complete
09:07:26.730 -> Artemis OpenLog v1.10
09:07:26.802 -> Created log file: dataLog00001.TXT
09:07:26.838 -> Created log file: serialLog00002.TXT
09:07:26.940 -> SD card online
09:07:26.940 -> Data logging online
09:07:26.940 -> Serial logging online
09:07:26.940 -> IMU online
09:07:26.977 -> No sensors detected
09:07:26.977 -> 
09:07:26.977 -> Menu: Main Menu
09:07:26.977 -> 1) Configure Terminal Output
09:07:26.977 -> 2) Configure Time Stamp
09:07:26.977 -> 3) Configure IMU Logging
09:07:26.977 -> 4) Configure Serial Logging
09:07:26.977 -> 5) Configure Analog Pin Logging
09:07:26.977 -> 6) Detect / Configure Attached Devices
09:07:26.977 -> r) Reset all settings to default
09:07:26.977 -> x) Return to logging
09:07:39.206 -> 
...
09:07:55.279 -> rtcDate,rtcTime,aX,aY,aZ,gX,gY,gZ,mX,mY,mZ,imu_degC,output_Hz,
09:07:55.279 -> 04/21/2020,09:08:15.28,-9.77,-19.04,-1000.00,-1.92,2.16,-1.11,-10.05,87.45,12.90,26.13,333.33,
09:08:00.608 -> 04/21/2020,09:08:20.62,-5.37,-30.76,-1016.60,-0.98,6.11,2.44,-11.10,87.60,14.70,25.07,666.67,
09:08:00.642 -> ⸮04/21/2020,09:08:25.95,-8.79,-18.07,-1020.51,-0.83,5.68,2.47,-12.15,86.70,12.90,25.31,1000.00,
09:08:05.958 -> ⸮04/21/2020,09:08:31.29,-4.39,-32.23,-1024.41,-1.15,-1.73,-6.09,-12.75,85.35,14.25,25.27,1333.33,
09:08:11.300 -> ⸮04/21/2020,09:08:36.63,-10.74,-25.88,-1008.30,-1.04,7.12,2.26,-11.10,85.80,12.75,25.22,1666.67,
09:08:16.644 -> ⸮04/21/2020,09:08:41.97,5.37,-24.41,-1013.67,-0.70,6.39,4.02,-10.20,85.05,7.05,24.98,2000.00,
09:08:22.001 -> ⸮04/21/2020,09:08:47.31,-5.86,-18.07,-1007.81,1.01,5.07,1.83,-9.90,86.10,7.05,25.12,2333.33,
09:08:27.334 -> ⸮04/21/2020,09:08:52.65,-11.72,-11.72,-1012.21,-1.65,3.35,3.34,-10.20,83.10,5.85,25.07,2666.67,
09:08:32.675 -> ⸮04/21/2020,09:08:57.98,-0.49,-25.88,-1001.46,0.75,-1.63,-3.56,-9.45,85.05,5.10,24.79,3000.00,
09:08:37.996 -> ⸮04/21/2020,09:09:03.32,-19.04,-14.16,-1006.84,-0.83,2.89,4.13,-9.45,82.05,4.35,25.22,3333.33,
09:08:43.352 -> ⸮04/21/2020,09:09:08.66,-26.86,-26.37,-1020.02,0.51,2.17,2.48,-9.45,83.55,2.40,24.88,3666.67,
09:08:48.674 -> ⸮04/21/2020,09:09:14.00,5.37,-34.18,-1010.25,1.62,3.50,1.60,-9.60,83.25,5.25,24.93,4000.00,
09:08:54.022 -> ⸮04/21/2020,09:09:19.34,-17.58,-26.37,-1015.14,0.21,2.69,1.62,-8.85,82.20,1.80,25.07,4333.33,
09:08:59.358 -> ⸮04/21/2020,09:09:24.68,-13.67,-36.13,-1007.32,-1.18,3.15,1.93,-9.00,84.60,5.10,24.93,4666.67,
09:09:04.678 -> ⸮04/21/2020,09:09:30.02,-12.21,-19.53,-1026.37,-1.53,2.41,1.16,-8.25,84.30,5.10,24.93,5000.00,
09:09:10.060 -> ⸮04/21/2020,09:09:35.37,-21.48,-21.00,-1017.09,-1.63,5.92,2.94,-9.90,82.80,5.10,25.22,5333.33,
09:09:15.393 -> ⸮04/21/2020,09:09:40.72,5.86,-17.58,-1009.28,-2.31,2.76,2.45,-8.85,83.70,4.95,24.93,5666.67,
09:09:20.743 -> ⸮04/21/2020,09:09:46.05,-16.11,-25.39,-1017.58,0.52,6.27,5.03,-10.05,84.45,4.20,24.98,6000.00,
09:09:26.080 -> ⸮04/21/2020,09:09:51.39,-5.37,-28.32,-1008.79,-0.09,1.74,2.60,-9.90,82.80,5.85,24.93,6333.33,
09:09:31.400 -> ⸮04/21/2020,09:09:56.73,-15.14,-23.44,-1026.86,-0.87,3.08,1.75,-9.00,83.10,5.10,24.88,6666.67,
09:09:36.729 -> ⸮04/21/2020,09:10:02.07,-2.93,-18.55,-1015.14,0.27,5.70,3.08,-8.40,84.00,4.20,24.79,7000.00,
09:09:42.084 -> ⸮04/21/2020,09:10:07.41,-20.51,-35.16,-1017.58,-2.23,2.76,1.23,-8.10,82.95,6.00,24.88,7333.33,
09:09:47.413 -> ⸮04/21/2020,09:10:12.75,-3.42,-30.27,-1015.62,-0.59,3.02,-0.26,-8.25,85.05,5.10,24.98,7666.67,
09:09:52.763 -> ⸮04/21/2020,09:10:18.09,-11.23,-16.60,-1015.14,-0.40,4.34,5.27,-9.00,83.55,6.60,24.93,8000.00,
09:09:58.087 -> ⸮04/21/2020,09:10:23.43,-7.81,-26.37,-1014.65,-0.15,2.03,4.09,-8.85,85.20,7.20,24.93,8333.33,
09:10:03.429 -> ⸮04/21/2020,09:10:28.76,0.00,-27.34,-1003.91,-0.98,3.43,4.76,-8.70,85.35,6.30,24.98,8666.67,
09:10:08.780 -> ⸮04/21/2020,09:10:34.10,0.98,-17.09,-1014.16,-1.11,2.90,1.11,-8.10,84.90,6.45,25.22,9000.00,
09:10:14.131 -> ⸮04/21/2020,09:10:39.44,-5.86,-25.39,-1010.74,0.41,2.49,1.32,-8.40,84.00,6.45,25.27,9333.33,
09:10:19.453 -> ⸮04/21/2020,09:10:44.78,-108.40,-6.35,-1166.50,-8.48,9.34,112.17,-21.60,74.40,8.85,24.88,9666.67,
09:10:24.795 -> ⸮

```

nseidle, 2020-04-21:
Thanks for the thorough report! I'm attempting to replicate. I haven't been able to yet.

`
x) Return to logging
rtcDate,rtcTime,aX,aY,aZ,gX,gY,gZ,mX,mY,mZ,imu_degC,output_Hz,
04/21/2020,10:22:09.09,-109.86,-8.30,-978.52,0.34,-2.11,2.97,59.85,-27.90,26.25,25.84,250.00,
04/21/2020,10:22:14.36,-112.79,-1.95,-996.09,-0.32,-1.14,5.29,59.10,-26.85,27.00,24.98,666.67,
04/21/2020,10:22:19.62,-132.32,-18.07,-990.23,-2.28,0.82,-4.08,60.00,-25.95,26.10,24.93,750.00,
04/21/2020,10:22:24.88,-109.38,-21.97,-978.52,1.49,-2.76,3.69,59.55,-25.35,27.30,24.88,1333.33,
04/21/2020,10:22:30.15,-108.40,-14.16,-994.63,-0.97,-2.44,-4.56,61.05,-25.05,26.25,24.79,1666.67,
04/21/2020,10:22:35.41,-111.82,-9.77,-1005.86,1.77,-0.18,5.52,58.95,-25.95,26.85,24.74,1500.00,
04/21/2020,10:22:40.67,-111.33,-10.25,-1003.42,1.46,-1.04,4.83,61.80,-26.40,27.30,24.93,2333.33,
04/21/2020,10:22:45.94,-115.72,-15.62,-1015.62,1.69,0.14,3.63,61.95,-25.20,25.65,24.74,2000.00,
04/21/2020,10:22:51.20,-118.65,-16.60,-996.09,0.92,-4.13,5.46,60.30,-26.10,25.20,24.59,3000.00,
04/21/2020,10:22:56.47,-115.23,-8.30,-990.72,1.02,-2.37,1.86,61.05,-26.10,26.25,24.59,3333.33,
04/21/2020,10:23:01.73,-116.21,-18.07,-1001.95,1.03,-1.60,4.77,60.45,-25.95,26.40,24.69,3666.67,
04/21/2020,10:23:07.00,-120.61,-6.35,-1004.39,-3.23,2.18,-7.56,60.90,-26.25,26.85,24.55,4000.00,
04/21/2020,10:23:12.27,-122.56,-19.04,-986.82,1.16,-0.49,1.56,59.70,-25.50,27.60,24.55,4333.33,
04/21/2020,10:23:17.54,-116.21,-5.37,-985.84,1.19,1.51,6.15,61.95,-25.95,26.85,24.55,4666.67,
04/21/2020,10:23:22.80,-109.86,-9.77,-995.61,1.44,0.71,5.44,60.90,-25.20,26.40,24.45,5000.00,
04/21/2020,10:23:28.07,-118.16,-10.25,-985.84,1.91,-1.34,3.46,61.80,-25.05,27.30,24.35,4000.00,
04/21/2020,10:23:33.33,-117.68,-14.16,-1000.49,-2.58,-1.06,-5.40,60.30,-26.10,26.25,24.45,5666.67,
04/21/2020,10:23:38.60,-116.70,-4.39,-993.16,-0.63,-1.80,5.71,60.00,-25.95,27.15,24.45,6000.00,
04/21/2020,10:23:43.86,-123.05,-6.35,-1011.72,-2.27,-0.51,-3.02,59.85,-25.05,27.00,24.26,4750.00,
04/21/2020,10:23:49.13,-115.23,-15.62,-1006.84,0.90,-1.34,3.47,60.00,-25.20,25.65,24.35,6666.67,
04/21/2020,10:23:54.40,-105.96,0.00,-1005.86,1.10,0.74,6.82,60.60,-25.65,27.75,24.45,7000.00,
04/21/2020,10:23:59.66,-121.09,-6.84,-1002.93,-0.92,-1.06,5.63,61.65,-25.50,27.15,24.02,5500.00,
04/21/2020,10:24:04.93,-110.35,-4.88,-1007.81,1.11,-0.56,4.37,60.30,-26.40,25.20,24.35,5750.00,
04/21/2020,10:24:10.20,-115.72,-14.16,-991.21,-2.34,-1.59,-3.62,61.80,-26.10,25.95,24.16,6000.00,
04/21/2020,10:24:15.47,-114.75,-21.00,-997.07,2.01,-1.38,6.11,60.00,-27.00,24.90,23.92,8333.33,
04/21/2020,10:24:20.74,-120.61,-2.44,-1001.46,1.08,-0.38,4.38,60.60,-25.65,25.95,24.16,8666.67,
04/21/2020,10:24:26.01,-110.84,-10.74,-997.56,1.63,0.36,4.32,61.35,-25.65,27.00,24.07,9000.00,
04/21/2020,10:24:31.28,-114.26,-5.86,-998.54,-1.22,0.24,-7.02,61.35,-26.10,27.30,24.26,9333.33,
04/21/2020,10:24:36.55,-117.68,-9.77,-1000.98,0.72,0.44,3.89,61.95,-24.45,26.85,24.16,7250.00,
04/21/2020,10:24:41.83,-109.86,-9.28,-998.05,-2.24,-0.37,1.76,59.70,-27.30,26.10,24.02,2142.86,
04/21/2020,10:24:47.09,-121.09,0.49,-1005.86,0.89,-0.05,4.18,60.90,-26.25,26.40,23.88,7750.00,
04/21/2020,10:24:52.36,-101.07,-15.14,-1001.46,2.42,-1.21,3.98,60.45,-25.20,27.15,24.02,8000.00,
04/21/2020,10:24:57.63,-106.93,-10.74,-1015.14,1.37,-0.38,4.21,61.05,-24.60,25.95,23.92,8250.00,
04/21/2020,10:25:02.90,-116.21,1.95,-992.19,-0.16,-0.82,3.08,60.00,-26.70,26.85,24.02,8500.00,
04/21/2020,10:25:08.17,-101.56,-7.81,-987.79,1.22,-2.44,5.48,60.00,-25.50,26.85,23.92,8750.00,
04/21/2020,10:25:13.44,-109.86,-5.37,-999.02,2.21,0.37,3.48,60.90,-25.50,26.40,24.16,9000.00,
04/21/2020,10:25:18.71,-110.35,-10.74,-999.02,-0.96,-0.89,3.56,60.90,-26.25,28.20,24.16,9250.00,
04/21/2020,10:25:23.98,-113.28,-15.14,-1007.32,-0.16,-1.09,3.94,60.60,-25.35,26.55,23.92,12666.67,
04/21/2020,10:25:29.25,-123.05,-21.48,-998.05,0.01,0.64,4.50,61.35,-25.05,25.50,24.07,13000.00,
04/21/2020,10:25:34.53,-117.19,-25.88,-994.63,-0.34,-3.70,2.76,59.70,-24.45,27.90,24.16,13333.33,
04/21/2020,10:25:39.80,-111.82,-2.93,-1004.39,1.58,-1.76,7.44,59.70,-25.95,27.15,23.92,10250.00,
`

WRT 1mA in sleep current, be sure to cut the jumper to the power LED for maximum reduction of current. We'll move the power LED to a GPIO in the next hardware rev.

Additionally, I see the output Hz is incorrectly reported (9666Hz there at the end). I'll try to fix that shortly.

nseidle, 2020-04-21:
You mention failures at 5-10s between readings. Do you have success at >10s between readings?

Other questions:
* We recently moved to exFat support. What file system format is your SD card? FAT16, FAT32, or exFat?
* Why size and brand of microSD card are you using?
* STAT LED is currently not toggled in low power mode. The only LEDs that may be active are power (if jumper is intact) and TX LED as data is sent to terminal. Are you seeing the STAT LED toggle?
* If you don't mind, can you post your OLA_settings.cfg file so I can setup my OLA identically to yours?

adamgarbo, 2020-04-21:
Hi Nathan,

Thanks for the updates. The microSD cards were formatted using both FAT16 and FAT32. 

I confirmed the behaviour with a 256 MB [industrial-rated micro SD card](https://industrial.apacer.com/en-ww/SSD/CV110-MSD) formatted using FAT16. At 1 Hz, only the first 3 samples are recorded.

Regarding power consumption, I can confirm the PWR jumper is cut and no current is running through it. When the microSD card is removed, Iq drops to ~250 uA after the battery is charged, but never appears to enter low-power while a microSD card is inserted.

Thanks for the clarification on the STAT LED.

Let me know if there's any additional info you need. I'm slowly making my way through the firmware and am happy to test.

nseidle, 2020-04-21:
I believe the issue was [this array](https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/OpenLog_Artemis/logging.ino#L20) was declared too small. The file names are "dataLog00003.txt" which is larger than 13 chars. This function is called every time the unit wakes so over time the bug was propagating all sorts of bad behavior including the stray random characters at power down you (and I) were seeing:

[image]

This bug snuck in because this code comes from the original OpenLog that only supported 8.3 file names.

With the fix in place, everything is behaving much better. Here is a release candidate: [OpenLog_Artemis-v12-RC.zip](https://github.com/sparkfun/OpenLog_Artemis/files/4512651/OpenLog_Artemis-v12-RC.zip). Please load it and kick the tires. I'll be 3s logging tonight to see if I can break it.

adamgarbo, 2020-04-21:
Thanks! I've uploaded the new firmware and at first glance, the data logging appears to be working much more smoothly. I'll also run a test overnight.

Quiescent draw also appears to be improved. Currently showing 330 uA on my Fluke 287 but I suspect the battery needs to recharge. I'll keep an eye on it to see if it drops.

Speaking of the sleep current, I know the AP2112 accounts for ~60 uA but where is the remaining ~180 uA coming from? Any chance of reducing it further in the next hardware rev?

adamgarbo, 2020-04-22:
No issues with the overnight data logging test, aside from clock drift. I'm assuming the plan is to eventually move away from using microseconds as an interval timer and start making use of the Artemis' built-in RTC alarms?

Quiescent draw also stabilized around ~240 uA (1 mW).

If your 3 s test didn't break anything, I think this issue can be closed.

nseidle, 2020-04-22:
Good to know. Thanks Adam! My tests similarly passed but I wasn't watching clock drift. Currently we are using the internal RTC for timestamps. I believe you've done some RTC analysis on Artemis. Let me know if there's anything we can do to shore it up.

~240uA matches exactly what I am seeing. To answer your question, yep AP2112, but also the CH340E, and the MCP LiPo charger. The microSD was also giving me a bit of head scratching and we attempt to totally disconnect it via the PChannel MOSFET but in practice it looks like it is being parasitically powered via some of the SPI pins (although even those are disabled). I welcome the help squeezing every uA we can out of the hardware. If you're up to the challenge, I will replace any hardware that you destroy in testing. Future revs may be lower current but I needed to get this released just to find bugs like you did.

I'll keep this open until I get the release complete.

adamgarbo, 2020-04-22:
Sounds good!

The clock drift is actually due to an accumulation of milliseconds over time. For example, with the interval set to 5 seconds, the log shows the actual time between measurements is 5.16 - 5.18 seconds. This results in the timestamps drifting over time (i.e. maintaining regular intervals of :05, :10, :15, etc. is not possible).

I've started [adding functionality to the Apollo3 RTC library](https://github.com/adamgarbo/Arduino_Apollo3/blob/rtcAlarms/libraries/RTC/src/RTC.h) and have got rolling alarms working well with the Artemis' RTC, but it's a work in progress.

Roger on the quiescent draw! Paul mentioned there were some gremlins.

closed 2020-04-27 by nseidle