sparkfun/OpenLog_Artemis #4: SCD30 autodetection has problems

nseidle, 2020-05-05:
v1.2 OLA firmware, no MicroSD card, 10Hz output, SCD30 with no SEL connection.

SCD30 is usually autodetected correctly, but not always. It's most repeatable when SCD30 is first inserted/powered by OLA. After power-on, OLA can be reset and correctly IDs SCD30.

nseidle, 2020-05-05:
Now replicable using I2C_Detector test sketch.

It turns out the SCD30 wants a lot of time at boot. Boot time is not clearly documented but mentioned in the interface spec as <2s (so not clear at all).

[image]

By power cycling the SCD30 via MOSFET+software I can get the SCD30 to ack but then fail to begin (register checks fail). This is workable. New code is this:

    case ADR_SCD30:
      if (co2Sensor_SCD30.begin(qwiic) == true) //Wire port
        qwiicAvailable.SCD30 = true;
      else
      {
        //Give it 2s to boot and then try again
        delay(2000);
        if (co2Sensor_SCD30.begin(qwiic) == true) //Wire port
          qwiicAvailable.SCD30 = true;
      }
      break;

This code works reliably. Pushing to v13. @jiggermole - please reopen if you continue to have problems.

closed 2020-05-05 by nseidle

nseidle, 2020-05-05:
Further thoughts. This problem is related to:

* How quickly the SD card mounts. If not SD card is present, then sensor auto-detection happens faster than the sensors may be online. We need to ask future users who have issues if they are using an SD card or not.
* The SCD30 requires being powered. It is not really compatible with being power cycled. Perhaps we should add a feature that warns the user that Qwiic bus power cycling is a bad idea.
* We need to ask future users who have issues if they are using an SD card or not.

jiggermole, 2020-06-01:
V1.3 of the firmware did take care of the issue.  the SCD30 detect reliably now.

nseidle, 2020-06-01:
Excellent! Thanks. Please keep an eye on v14 and future firmware. I need to add the ability to set [boot and measurement times](https://github.com/sparkfun/OpenLog_Artemis/issues/5) for individual sensors.