sparkfun/OpenLog_Artemis #5: Create warning for sensors that should avoid power cycling

nseidle, 2020-05-05:
Some sensors have very long boot times and/or need to be powered constantly to run calculations between being queried for data. Powering down the Qwiic bus between sleeps causes these sensors to fail to perform as designed. SCD30 is a prime example but there are others.

Currently (v1.1 and above) Qwiic bus can be kept on between low power sleeps.

If a always-on device is detected, over ride the Qwiic bus power setting and stay on. If setting is over-written then display warning at initial power on banner.

nseidle, 2020-05-27:
Sensors known to be an issue:

* SGP30

Also, we should probably implement a delayBeforeBegin and delayBeforeMeasurement for each class of device.

PaulZC, 2020-09-07:
These two commits in the v16_Candidate branch cover this:
- [Added qwiic bus power up delays for each sensor type](https://github.com/sparkfun/OpenLog_Artemis/commit/90e99761a0feec1870c54c1932ad27aa3fce999c)
- [Updated the power-on delay for the SCD30](https://github.com/sparkfun/OpenLog_Artemis/commit/e20a8bf1955608625420bba0b27d1cc75b375ad4)

If we find any other sluggish sensors, we can give them their own start-up delay.

We can close this once v16 goes live.

PaulZC, 2020-09-08:
Released in [Firmware Version 1.6](https://github.com/sparkfun/OpenLog_Artemis/releases/tag/v1.6)

closed 2020-09-08 by PaulZC