sparkfun/OpenLog_Artemis #6: Support multiple of a device through mux

nseidle, 2020-05-05:
Some I2C devices support only one I2C address but benefit greatly from multiple devices being read. The NAU7802 digital scale is probably at the top of this list.

Additionally, the OLA firmware does not support multiple of a device that *does* have multi-address support (for example, the LPS25HB). It would be nice to support both of these if possible.

artemtk, 2020-05-06:
2nd this request. Support for Sparkfun Qwiic I2C Mux would be great!

nseidle, 2020-05-27:
Alright! I think I've got a release candidate: [OpenLog_Artemis-v14-RC1.zip](https://github.com/sparkfun/OpenLog_Artemis/files/4691625/OpenLog_Artemis-v14-RC1.zip)

The branch is [here](https://github.com/sparkfun/OpenLog_Artemis/tree/v14). This is a *large* change from previous versions of OLA. Instead of creating one class at power on for each library, the classes are created dynamically.

During the detection phase, we create a linked list of detected devices. Once detection is complete, pointers from the linked list are pointed to newly formed classes. Each device has its own config setting pointer - this means you can have 17 NAU7802s and configure each one individually.

This is how the autodection algorithm works:
1) Scan bus for muxes (0x70 to 0x77)
2) Begin() any muxes. This causes them to turn off all their ports.
3) With any possible muxes turned off, finish scanning main branch
4) Any detected device is stored as a node in a linked list containing their address and device type,
5) If muxes are found, begin scanning mux0/port0. Any new device is stored with their address and mux address/port.
6) Begin() and configure all devices in our linked list. Connections through muxes are performed as needed.

What you get is something like this:

[image]

Note in the image above that there are two VL53L1Xs (a device that has only one I2C address). They both play nicely. There are also 2x BME280 and 2x CCS811. These have different address capabilities, but where hung off of various mux ports for testing.

Known issues:

* Power cycling. See issue #9.
* U-blox GPS modules can fail to detect at startup due to too many pull up resistors. See [this issue](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/issues/40) but just know that removing all the I2C pullups on the bus should fix this.
* Some sensors don't like to be power cycled. I still need to implement [a way](https://github.com/sparkfun/OpenLog_Artemis/issues/5) to avoid it.

Overall it's a great improvement to flexibility and usability. Please kick the tires and let me know what you find.

nseidle, 2020-05-27:
More notes:

* This RC has ExFat turned on. I haven't had any logging issues recently but it's still suspect. We can revert back to FAT if anyone has concern.

PaulZC, 2020-08-04:
I had to change the autodetection algorithm slightly:

1. Scan bus for muxes (0x70 to 0x77)
2. Begin() any muxes. This causes them to turn off all their ports.
3. With any possible muxes turned off, finish scanning main branch
4. Any detected device is stored as a node in a linked list containing their address and device type,
5. If muxes are found, begin scanning mux0/port0. **Skip any device addresses found on the main bus (otherwise we end up with ghost duplicates).** Any _new_ device is stored with their address and mux address/port.
6. **Remember to disable mux port7 when we're done scanning ports on that mux.**
7. Begin() and configure all devices in our linked list. Connections through muxes are performed as needed.

Seems to work nicely - with the NAU7802 too. This fix is currently only implemented in the [V15_Candidate branch](https://github.com/sparkfun/OpenLog_Artemis/tree/V15_Candidate).

nseidle, 2020-08-10:
Supported in v15. Closing.

closed 2020-08-10 by nseidle