sparkfun/OpenLog_Artemis #7: Unable to compile OpenLog_Artemis.ino

adamgarbo, 2020-05-16:
### Subject of the issue
OpenLog_Artemis.ino is unable to compile locally. Appears to be due to a conflict between the following two libraries:
* SparkFun_VL53L1X_4m_Laser_Distance_Sensor
* SparkFun_High_Precision_Temperature_Sensor_TMP117_Qwiic

### Your workbench
* Working within Arduino IDE.
* SparkFun Artemis Module board definition selected (is there an OLA definition?)

### Steps to reproduce
* Forked repository on GitHub, cloned locally through GitHub Desktop and modifying files directly. 

### Expected behaviour
* Code would compile without issue.

### Actual behaviour
```
In file included from /Users/adam/Documents/Arduino/libraries/SparkFun_VL53L1X_4m_Laser_Distance_Sensor/src/SparkFun_VL53L1X.h:31,
                 from /Users/adam/Documents/GitHub/OpenLog_Artemis/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino:104:
/Users/adam/Documents/Arduino/libraries/SparkFun_VL53L1X_4m_Laser_Distance_Sensor/src/vl53l1x_class.h:58:30: error: expected unqualified-id before numeric constant
 #define SOFT_RESET           0x0000
                              ^~~~~~
/Users/adam/Documents/Arduino/libraries/SparkFun_High_Precision_Temperature_Sensor_TMP117_Qwiic/src/SparkFun_TMP117.h:39:11: note: in expansion of macro 'SOFT_RESET'
   uint8_t SOFT_RESET : 1;  // Software reset bit
           ^~~~~~~~~~
```

Commenting out all code relating to the VL53L1X allows the code to compile without issue.

adamgarbo, 2020-05-16:
And of course, figured out less than 5 minutes later this collision was already addressed in v1.22:
https://github.com/sparkfun/SparkFun_TMP117_Arduino_Library/compare/v1.2.2...master

This compiling issue looks to be due to the fact that v1.21 is the most recent version of the TM117 library available from the Arduino IDE. Will SparkFun need to push the most recent version?

Cheers,
Adam

nseidle, 2020-05-16:
Yep. I fixed it to make OLA work but failed to release. TMP117 v1.2.3 released. You should see it hit lib manager shortly.

closed 2020-05-16 by nseidle