sparkfun/OpenLog_Artemis #9: OLA & GPS-RTK2 - Hanging at Boot

adamgarbo, 2020-05-23:
### Subject of the issue
I've noticed some strange behaviour with the OpenLog Artemis and the GPS-RTK2 board while booting-up. It's frequently necessary to either re-open the Serial Monitor or hit the reset button in order for the OLA to actually boot up and start logging. When connecting the OLA via USB it appears to get stuck at reading the configuration file (shown below).

```
09:15:50.703 ->
09:15:50.740 -> Config file read complete (Freezes here)
```
When connecting the OLA to external power (i.e. battery), I have to hit the reset button 100% of the time to get the LEDs to start blinking and for logging to start. 

### Your workbench
* What version of firmware are you using? 
  * v1.3
  * Paul's GNSS Logger v1.1
* How is OpenLog Artemis wired to your sensor(s)?
  * OLA + GPS-RTK2 using a single Qwiic connector
* How is everything being powered?
  * USB-C or LiPo
* Are there any additional details that may help us help you?
  * I've observed this behaviour on both of my OLA boards and firmware v1.1 to v1.3

### Steps to reproduce
* USB Issue
  * Connect OLA to GPS-RTK2
  * Open Serial Monitor and confirm OLA is logging properly
  * Unplug and reconnect USB without closing Serial Monitor
  * Observe OLA hang at "Config file read complete"
* Battery Issue
  * Connect OLA to GPS-RTK2
  * Connect battery power
  * Observe no LED activity
  * Hit reset button and observe that LED activity begins

### Expected behaviour
* OLA should be able to boot properly without needing the Serial Monitor to be re-opened or the reset button to be pressed.

### Actual behaviour
* Hangs at boot-up

adamgarbo, 2020-05-26:
After having spoken with Paul regarding this issue, it appears to be due to GPS-RTK2 causing the OLA to experience a brown-out. I expect that pairing the OLA with another similarly high-current device would produce the same behaviour. As I understand it, this should be addressed with the next hardware revision of the OLA (4-layers). Should this issue remain open until then?

nseidle, 2020-05-26:
First - thanks as always for testing a reporting. This is helping a ton.

I concur that it *should* get fixed shortly but let's keep this open until we can get it properly verified with v1.0 hardware.

And just to add to the pontification, We should keep in mind that Artemis is a ~1mA device with 10, 1, and 0.1uF dedicated to it. These caps are protected against drain with D6. I suspect the RTK2 *is* browning out the 3.3V bus but perhaps VCC_1 (the main feed into Artemis) is staying at 3.3V. But the POWER_LOSS circuit may trigger the shutdown of Artemis because 3.3V bus is dropping.

I'll do some testing today.

adamgarbo, 2020-05-27:
I put the OLA and GPS-RTK2 on the scope this morning. I disabled the PIN_POWER_LOSS interrupt in the firmware and measured the 3.3V line during a 15 s logging/15 s sleeping cycle that turned off power to the Qwiic bus. It appears that there is 200 μs voltage drop to 1.4 V when the power to the Qwiic bus is turned back on, and then another drop after it resets itself and tries to start logging again.

[image]

nseidle, 2020-05-27:
This is extremely helpful, thank you! 

Just to be clear, are you measuring the 3.3V bus, not the VCC_1 traces that feed Artemis after the D6 RTC diode?

Perhaps the SD card is (within spec) pulling a few hundred mA during power on startup. Perhaps the 0.1uF decoupling for SD is not enough.

Have you tried moving the [beginQwiic()](https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/OpenLog_Artemis/lowerPower.ino#L196) after the config file calls?

adamgarbo, 2020-05-27:
I was measuring the 3.3V bus, correct. Here is the voltage drop on the VCC_1 trace between D6 and the Artemis. You can see the 1 second writes to the SD card on the left. Is the voltage drop across D6 expected?

Pardon the noise:
[image]

Haven't tried moving beginQwiic() yet, but I can give it a shot!

nseidle, 2020-05-27:
@adamgarbo - Looking at your [second](https://user-images.githubusercontent.com/22924092/83033000-9185cd80-a004-11ea-8331-d76cecd09288.png) image, is the time scale in seconds? So you are manually re-opening the serial monitor to initial a DTR reset of Artemis?

2.7V for Artemis *should* be enough to continue operation (it goes down to 1.8V). Perhaps it's getting stuck trying to talk to the SD card that has locked up due to brown out. I've tried to replicate with a 1GB and 512GB card (exfat). No luck yet.

However, I am doing some tangential testing and am seeing similar madness. I am using a very low power board (LPS25HB 25uA) but [it has](https://cdn.sparkfun.com/assets/e/a/a/4/0/Qwiic_Pressure_Sensor_-_LPS25HB.pdf) a power LED that draws ~2mA as well as a 4.7uF and a 0.1uF.

I'm skipping OLA firmware entirely and just running a simple on/off sketch. Turning off the Qwiic bus correctly brings the power on the peripheral LPS25HB to ~0.15V (note: I am disabling Artemis' internal I2C pullups with setPullups(0), otherwise the SDA/SCL lines leak current). Turning the Qwiic bus back on causes the CH340E USB To serial IC to reset, which in turn causes DTR to toggle as the port re-enumerates, which resets Artemis. Ick.

Adding 10uF to the board reduces reset a bit, 100uF prevents resets and device operates as we want.

We're taking two approaches on the next rev (v10):

* 4 layers with a dedicated 3.3V plane
* Dedicated 3.3V regulator (a 2nd AP1112K) for the Qwiic bus 3.3V. Artemis controls the Vreg Enable pin.

[image]

This *should* move the surge current draw from the 3.3V rail to VIN. It should also allow the 3.3V Artemis (and CH340E) portion of the circuit to have more tolerance for ripple.

adamgarbo, 2020-05-27:
@nseidle 
> @adamgarbo - Looking at your second image, is the time scale in seconds? So you are manually re-opening the serial monitor to initiate a DTR reset of Artemis?

Correct, the time scale is seconds! I'm re-opening Serial Monitor to initiate a DTR reset to attempt to reproduce the behaviour I've been observing thus far (i.e. having to press the reset button or re-open the Serial Monitor to get the OLA to begin logging).

> Turning the Qwiic bus back on causes the CH340E USB To serial IC to reset, which in turn causes DTR to toggle as the port re-enumerates, which resets Artemis. Ick.

I feel like this behaviour really sums up the issue. 

Looking forward to the next hardware revision and am happy to do more testing!

nseidle, 2020-05-27:
I piggy-backed an AP2112K to provide the Qwiic connector with its own 3.3V source from VIN. All problems have been fixed. I can power cycle boards with large capacitance and current draws without issue. I'm a lot more confident this will be fixed soon.

[image]

adamgarbo, 2020-05-27:
Great news!

So, will the plan be to replace Q4 with a second AP2112, controlled using QWIIC_PWR?

With the second AP2112 EN disabled, this also shouldn't appreciably increase the quiescent draw (datasheet shows typical 0.01 uA).

P.S. Those piggy-backed AP2112s are super cute!

nseidle, 2020-05-27:
Yes, that the plan and hope.

PS - Yea, I was looking for a good place to get VIN and GND and realized it was right there on the pre-existing reg!

PaulZC, 2020-06-24:
Hi @adamgarbo ,
We managed to squeeze the extra regulator onto the new V10 PCB and feed it from a beefed-up power polygon on the internal layers. Brown-outs will be a thing of the past. I'll leave this issue open until V10 goes public.
Thanks again for identifying this,
Paul