sparkfun/OpenLog_Artemis #11: Suggestion: Battery Voltage Divider

adamgarbo, 2020-05-29:
@nseidle I'm curious, is adding a voltage divider to the OpenLog Artemis to measure battery voltage still a possibility for the next hardware revision?

As I've mentioned previously, this is an extremely valuable feature for any data logger to have (e.g. Adafruit M0 Adalogger) and it would be great to see it on the OLA. 

Running the math quickly, it could look something like this:

**Voltage input range**: 2.5 - 6.0V (AP2112)
**Analog input range:** 0 - 2.0V
**Resistor divider:** 2MΩ/1MΩ 
**Voltage output range:** 0.83 - 2.0V (1.167V)
**Resolution:** 0.0712 mV
**Current draw:** 2 μA @ 6.0V

The resistor values could be tweaked slightly (e.g. 5/2 MΩ) so as to not max out the analog input. A 0.1-1.0 μF capacitor would also be required.

Of course, the main question is where to stick it? Looking at the schematic, it looked like removing the PWR LED jumper would have opened up some real-estate, but it appears that the jumper has been repurposed.

Cheers!
Adam

PaulZC, 2020-06-01:
Hi @adamgarbo / @nseidle ,
I found the ADC input impedance data hidden in the Apollo3 BLE datasheet (section 21.7):

[image]

I need to look at this a bit more as they only mention 8 inputs (not 10 or 14) and I'm now really not sure if this explains the 9% issue I see on the AGT. If the impedance really is 180MOhm minimum (for ADC CH1-CH7) then a 2M/1M divider should work perfectly...
Cheers,
Paul

PaulZC, 2020-06-01:
Hi @adamgarbo / @nseidle ,
OK. So maybe they are referring just to ADCSE0-ADCSE7. Maybe they don't give the impedance figures for ADCSE8 and ADCSE9 as those are the special pins that can be used as single-ended or differential.

[image]

I'm using pin/pad 13 on the AGT to measure the battery voltage with a 200K/100K divider. That is one of the special pins. Maybe the impedance on that pin is lower for some reason and explains my 9% issue?
So, if my understanding is correct, then you should be perfectly OK to use pins 29/11/31/32/33/34/35 for voltage monitoring with a 2M/1M divider?
Keep away from pin 16 as that should have low input impedance.
And keep away from pins 12 and 13 as we perhaps don't properly understand what their impedance is when used for single-ended A2D?
Definitely worth a test just to be sure!
I'll try this on the ATP tomorrow. Watch this space!
Cheers,
Paul

adamgarbo, 2020-06-01:
@PaulZC Thanks for the sleuthing! Looking forward to your tests on the ATP tomorrow!

PaulZC, 2020-06-01:
Hi @adamgarbo / @nseidle ,
I couldn't wait until morning...!
I made a simple divider from three 1M resistors configured to divide by 3. I've connected it to 5V.
Here are the readings using * 2.0 * 3.0 / 16384.0 to convert to voltage:
- A11 (ADCSE2) 4.11V
- A12 (ADCSE9) 4.15V
- A13 (ADCSE8) 4.15V
- A16 (ADCSE0) 4.17V
- A29 (ADCSE1) 4.12V
- A31 (ADCSE3) 4.16V
- A32 (ADCSE4) 4.18V
- A33 (ADCSE5) 4.19V
- A34 (ADCSE6) 4.11V
- A35 (ADCSE7) 4.05V

So, a bit of variation but reasonably consistent. No evidence of pins 16/12/13 being significantly different.
The ADC reading (int) matches the calculated voltage.
Reading the A2D pin voltage with a multimeter I see exactly 1.6V. Reading the 5V pin with the same meter I see 5.07V.

If I repeat with 200K + 100K I see:
- A11 (ADCSE2) 4.16V
- A12 (ADCSE9) 4.21V
- A13 (ADCSE8) 4.21V
- A16 (ADCSE0) 4.20V
- A29 (ADCSE1) 4.18V
- A31 (ADCSE3) 4.19V
- A32 (ADCSE4) 4.22V
- A33 (ADCSE5) 4.23V
- A34 (ADCSE6) 4.16V
- A35 (ADCSE7) 4.12V

Reading the A2D pin voltage with a multimeter I see exactly 1.67V. Reading the 5V pin with the same meter I see 5.07V. The ~0.06V difference could be down to resistor tolerances.

If I drive the pins directly from a lab power supply set to 1.70V I see the following (the software is still using * 2.0 * 3.0 / 16384.0 to calculate the voltage):
- A11 (ADCSE2) 4.99V
- A12 (ADCSE9) 4.99V
- A13 (ADCSE8) 5.00V
- A16 (ADCSE0) 5.00V
- A29 (ADCSE1) 4.99V
- A31 (ADCSE3) 4.99V
- A32 (ADCSE4) 4.98V
- A33 (ADCSE5) 4.97V
- A34 (ADCSE6) 4.99V
- A35 (ADCSE7) 5.00V

I conclude that using resistor dividers is definitely throwing the measurement off (by close to 20%).
What I don't understand at the moment is how...
Tomorrow I'll measure the current flow into each pin to see if that can help explain what is going on.
Cheers,
Paul

PaulZC, 2020-06-02:
Hi @adamgarbo / @nseidle ,
Morning!
Just so we have the complete picture, here's what I see with 20K/10K (like I use on the AGT):
- A11 (ADCSE2) 4.65V
- A12 (ADCSE9) 4.66V
- A13 (ADCSE8) 4.66V
- A16 (ADCSE0) 4.66V
- A29 (ADCSE1) 4.65V

And here's what I see with 2K/1K:
- A11 (ADCSE2) 5.11V
- A12 (ADCSE9) 5.11V
- A13 (ADCSE8) 5.11V
- A16 (ADCSE0) 5.11V
- A29 (ADCSE1) 5.11V

OK. Good. I'm not completely losing the plot!
Now, let's measure how much current is flowing through the divider and how much is flowing into the A2D pin:

| Pin | ADC | Divider | ADC reading | Current into divider | Current into pin |
|---|---|---|---|---|---|
| A11 | ADCSE2 | 2K/1K | 5.10V | 1.68mA | 0.0uA |
| A11 | ADCSE2 | 20K/10K | 4.66V | 0.16mA | 0.0uA |
| A11 | ADCSE2 | 200K/100K | 4.18V | 16.8uA | 0.0uA |
| A11 | ADCSE2 | 2M/1M | 4.12V | 1.6uA | 0.0uA |
| A13 | ADCSE8 | 2K/1K | 5.10V | 1.68mA | 0.0uA |
| A13 | ADCSE8 | 20K/10K | 4.66V | 0.16mA | 0.0uA |
| A13 | ADCSE8 | 200K/100K | 4.20V | 16.8uA | 0.0uA |
| A13 | ADCSE8 | 2M/1M | 4.16V | 1.6uA | 0.0uA |

OK. Perhaps unsurprisingly the current flowing into the pin is too small to measure with my uA meter. I did notice that the ADC reading was slightly different when the ammeter was in series with the pin though. So what happens if I introduce a little extra resistance in series with the pin? Say 100R?

| Pin | ADC | Divider | ADC reading without 100R in series | ADC reading with 100R in series |
|---|---|---|---|---|
| A11 | ADCSE2 | 2K/1K | 5.11V | 5.10V |
| A11 | ADCSE2 | 20K/10K | 4.65V | 4.66V |
| A11 | ADCSE2 | 200K/100K | 4.17V | 4.24V |
| A11 | ADCSE2 | 2M/1M | 4.10V | 4.19V |

OK. Great. Now we're getting somewhere!

Time for pencil and paper...
More soon!

PaulZC, 2020-06-02:
Hi @adamgarbo / @nseidle ,
OK. I've dug into the literature on this and my conclusion is that the answer is complicated and depends on many things!
E.g.: https://www.st.com/resource/en/application_note/cd00004444-understanding-and-minimising-adc-conversion-errors-stmicroelectronics.pdf
The bottom line is that if you want the A2D reading to be true/accurate then the impedance of the source (our resistor divider) needs to be a few KOhms at most.
As we've seen, higher resistances give progressively worse accuracy.
My conclusion is that it is OK to use a 2M/1M divider **BUT** you need to be aware of the consequences (including a very slow response time) and you'll need a correction factor to convert the reading back to a true voltage!
_fin_
Paul

PaulZC, 2020-06-24:
Hi @adamgarbo ,
We've included the 2M/1M divider on the new V10 PCB. I'll leave this issue open until the new boards go public.
Thanks again for the suggestion!
Paul