sparkfun/OpenLog_Artemis #12: MS8607 library references need to be updated to current library name

bengervais, 2020-05-31:
Master branch won't compile with current sparkfun MS8607 library due to incorrect name references in code

(V1.3)

To reproduce, clean install arduino IDE and install all libraries from links in Openlog_Artemis master branch.  Compile will fail on mentions of MS8607_Library.h in various sketches.

Should be replaced with SparkFun_PHT_MS8607_Arduino_Library.h to allow compilation.  (I'm not yet comfortable with submitting pull requests on github, hence why submitted here, sorry)

nseidle, 2020-06-01:
Yup! Thanks for this!

I'm working pretty exclusively on the v14 firmware branch now so you'll see it update there first.

PaulZC, 2020-06-24:
Hi @bengervais ,
Thanks again for raising this issue. It has been fixed in the new V14 branch. But I'm going to leave this issue open until V14 is merged into the master branch.
I'm sure you have worked out the fix for yourself, but the correct #include is here if you need it:
https://github.com/sparkfun/OpenLog_Artemis/blob/v14/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino#L160
and
https://github.com/sparkfun/OpenLog_Artemis/blob/v14/Firmware/OpenLog_Artemis/settings.h#L2
Best wishes,
Paul

PaulZC, 2020-06-28:
Hi @bengervais  ,
Thanks again for opening this issue. This has been corrected in v14 of the firmware.
I'm going to close this issue but please re-open if you have any further problems!
Very best wishes,
Paul

closed 2020-06-28 by PaulZC