sparkfun/OpenLog_Artemis #15: Units of logged data

hordurk, 2020-06-23:
Can you clarify what the units (and scales) are of the logged data?

It would be good if this was documented somewhere

nseidle, 2020-06-23:
It depends on what you've got enabled for logging. There's potentially a data/time, IMU (accel, gyro, mag), temp in C, measurement Hz, and then any number of units based on what sensors you've got attached. There is a 'helper' line displayed at the top of the log that will show you the location and units for each data.

I you have a particular line or lines that you need help with, please feel free to comment and we'll point you in the direction of a datasheet that should get you settled.

Documentation is extremely sparse as OLA is currently a test product. We are actively writing the hookup guide for the red version (v10) but that is still a few weeks away.

closed 2020-06-23 by nseidle

hordurk, 2020-06-24:
Sorry, I specifically meant the IMU quantities: acceleration, gyro and mag values.
The helper line does not show the units for these quantities (at least not on mine).

adamgarbo, 2020-06-24:
@hordurk 

**ICM-20948 units:**
* Acceleration: milli-G (mg)
* Magnetomer: micro Tesla (μT)
* Gyroscope: degrees per second (DPS)
* Temperature: Celsius (°C)

See: 
* https://learn.sparkfun.com/tutorials/sparkfun-9dof-imu-icm-20948-breakout-hookup-guide
* https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/4d5f843753ec01c0c16484c97b6d7b6e459d260d/src/ICM_20948.cpp#L17-L93