sparkfun/OpenLog_Artemis #18: Firmware X04-v14_BETA Testing

adamgarbo, 2020-06-29:
Hi @PaulZC,

Putting the v14 firmware through its paces! Tests are being conducted with the OLA connected to the SparkFun SAM-M8Q board. Please see below for comments/suggestions/bugs.

**General**
* I would suggest slightly increasing the timeout on the menus. This would help new users that may be unfamiliar with the various menu options.

**9) Output Measurement Count: Enabled**
* The measurement count is a great addition. However, it appears to be missing a corresponding column title.

For example, even though enabled it wasn't included after `output_Hz`
* `rtcDate,rtcTime,aX,aY,aZ,gX,gY,gZ,mX,mY,mZ,imu_degC,gps_Date,gps_Time,gps_Lat,gps_Long,gps_Alt,gps_SIV,gps_FixType,gps_GroundSpeed,gps_Heading,gps_pDOP,output_Hz,`

**2) If sensor read time is greater than 2s, turn off Qwiic bus power: No**
* This menu option could be more clear. Is "sensor read time" the duration between samples?

**Bug**
* If I disable the above menu option so the Qwiic bus power does not turn off between samples and increase the logging duration to 10 s, it appears that all measurements except for the GPS are removed after the first sample. Confirmed on the log file on the SD card. Please disregard if this is simply a quirk due to the way power is handled as we transition to the next hardware rev.

```
09:31:34.500 -> 18/10/2015,00:01:30.76,82.52,-219.73,-1120.61,11.05,0.48,-3.49,-25.65,35.10,-51.30,39.64,18/10/2015,00:00:33.000,0,0,0,0,0,0,0,9999,0.97,1,
09:31:45.667 -> 18/10/2015,00:01:41.94,18/10/2015,00:00:44.000,0,0,0,0,0,0,0,9999,0.16,2,
09:31:56.559 -> 18/10/2015,00:01:53.10,18/10/2015,00:00:55.000,0,0,0,0,0,0,0,9999,0.13,3,
09:32:07.473 -> 18/10/2015,00:02:04.01,18/10/2015,00:01:06.000,0,0,0,0,0,0,0,9999,0.12,4,
09:32:18.636 -> 18/10/2015,00:02:14.92,18/10/2015,00:01:17.000,0,0,0,0,0,0,0,9999,0.11,5,
09:32:29.557 -> 18/10/2015,00:02:26.08,18/10/2015,00:01:28.000,0,0,0,0,0,0,0,9999,0.11,6,
09:32:40.440 -> 18/10/2015,00:02:36.99,18/10/2015,00:01:39.000,0,0,0,0,0,0,0,9999,0.10,7,
09:32:51.603 -> 18/10/2015,00:02:47.90,18/10/2015,00:01:50.000,0,0,0,0,0,0,0,9999,0.10,8,
09:33:02.544 -> 18/10/2015,00:02:59.07,18/10/2015,00:02:01.000,0,0,0,0,0,0,0,9999,0.10,9,
09:33:13.448 -> 18/10/2015,00:03:09.98,18/10/2015,00:02:12.000,0,0,0,0,0,0,0,9999,0.10,10,
```

**8) Synchronize RTC to GPS**

It's great to see this functionality added to the OLA. While I haven't yet looked at the code that makes this feature possible, I would suggest adding a warning message if the OLA synchronizes the RTC with the GPS before a fix is able to be established (i.e. bad date/time). 

Will keep testing!

Cheers,
Adam

adamgarbo, 2020-06-29:
Another interesting bug. If the OLA is set to sample every 2 seconds, it will only record a single measurement.

Behaviour observed when running the OLA in stand-alone mode or connected to the SAM-M8Q. All settings reset to default save for the Log Rate.

```
09:59:38.695 -> Menu: Configure Terminal Output
09:59:38.695 -> 1) Log to microSD: Enabled
09:59:38.731 -> 2) Log to Terminal: Enabled
09:59:38.731 -> 3) Set Serial Baud Rate: 115200 bps
09:59:38.731 -> 4) Set Log Rate in Hz: 0.500000
09:59:38.731 -> 5) Set Log Rate in seconds between readings: 2
09:59:38.731 -> 6) Enable maximum logging: Disabled
09:59:38.731 -> 7) Output Actual Hertz: Enabled
09:59:38.731 -> 8) Output Column Titles: Enabled
09:59:38.731 -> 9) Output Measurement Count: Disabled
09:59:38.731 -> x) Exit
09:59:39.783 -> 06/29/2020,13:59:40.45,47.85,-110.35,-1028.32,12.24,2.53,-8.79,-26.85,35.10,-53.70,49.13,inf,
```

PaulZC, 2020-06-30:
Hi @adamgarbo ,

Sincere thanks for this!

Please see the notes in PR #19 for the answers to your points:

Please let me know if you think the 15s menu timeout is reasonable (given that that is how long it takes for logging to start if the interval is >= 2 secs)?

Also, if you have time, can you please try logging some analog values too? Do the IMU readings still disappear when the interval is 10s?

Thanks again!
Paul

adamgarbo, 2020-07-01:
Hey @PaulZC 

Spent some time testing the OpenLog_Artemis-X04-v14_BETA_2020-06-30.bin firmware this morning.

- [X] The 2-second interval bug is no more!
- [X] The 15-second menu timeout feels much better.
- [X] The `count` column title looks good!
- [X] The message when synchronizing the RTC to the GPS is super helpful. I saw the u-blox library PR that added the gpsDateValid/gpsTimeValid flags. I remember talking with you about that several months ago!
- [ ] IMU Bug
  - It appears that the bug with the IMU is still present with the combination of the following settings:

`5) Set Log Rate in seconds between readings: 3` (Basically any value >2 seconds)

`2) Turn off Qwiic bus power between readings (>2s): No`

You can see in the output that just the IMU measurements are disabled. The analog measurements persist.
```
rtcDate,rtcTime,analog_11,analog_12,analog_13,analog_32,aX,aY,aZ,gX,gY,gZ,mX,mY,mZ,imu_degC,output_Hz,count,
01/07/2020,12:31:18.80,0.33,0.26,0.30,0.39,51.76,-54.20,-989.26,0.50,-19.05,0.99,22.20,5.10,-86.40,29.53,100.00,1,
08:31:19.063 -> ⸮01/07/2020,12:31:21.95,0.05,0.02,0.04,0.03,0.63,2,
08:31:22.193 -> ⸮01/07/2020,12:31:25.08,0.05,0.02,0.04,0.03,0.48,3,
08:31:25.346 -> ⸮01/07/2020,12:31:28.22,0.05,0.02,0.04,0.03,0.42,4,
08:31:28.471 -> ⸮01/07/2020,12:31:31.36,0.05,0.02,0.04,0.03,0.40,5,
08:31:31.626 -> ⸮01/07/2020,12:31:34.50,0.05,0.02,0.04,0.03,0.38,6,
```

Happy to do more testing!

Cheers,
Adam

adamgarbo, 2020-07-01:
Hi @PaulZC,

I discovered an issue relating to data logging. The column title header is not being written to the .TXT file while other unnecessary information (e.g. SD card online) is. The following output is from the microSD card file after resetting the OLA settings to default. The first 5 lines appear to be incorrectly written to the log file.

```
SD card online
Data logging online
Serial logging online
IMU online
No Qwiic devices detected
07/01/2020,13:35:38.42,324.71,76.17,-942.87,9.72,-22.15,-20.89,8.25,32.55,-63.60,29.63,333.33,
07/01/2020,13:35:38.52,536.13,218.26,-988.28,11.96,-15.34,-9.91,11.55,34.50,-65.10,29.87,19.42,
07/01/2020,13:35:38.62,308.11,-10.25,-1023.44,11.02,-14.62,-12.55,13.05,34.65,-66.45,42.90,14.78,
07/01/2020,13:35:38.72,295.41,31.74,-993.16,6.27,1.66,-1.82,14.40,33.75,-67.35,33.12,13.20,
07/01/2020,13:35:38.82,271.48,-207.52,-1004.88,11.58,7.69,-6.14,13.80,33.90,-65.25,29.87,12.41,
07/01/2020,13:35:38.92,239.75,0.00,-1015.62,13.05,6.74,-8.77,13.20,33.30,-65.85,44.43,11.93,
07/01/2020,13:35:39.02,297.36,87.40,-927.73,7.14,-0.11,-8.50,11.10,33.75,-66.30,34.71,11.61,
07/01/2020,13:35:39.12,463.87,65.43,-980.47,9.42,0.48,-8.47,11.55,34.50,-64.65,42.09,11.38,
07/01/2020,13:35:39.22,490.23,240.23,-986.33,7.18,3.98,-5.30,12.60,35.40,-66.45,43.24,11.21,
07/01/2020,13:35:39.32,402.34,166.50,-1034.67,8.52,1.13,-1.11,12.00,33.00,-66.30,47.26,11.07,
07/01/2020,13:35:39.42,275.39,83.01,-961.43,4.77,4.74,0.39,12.15,33.90,-65.70,33.03,10.97,
```

Cheers,
Adam

PaulZC, 2020-07-02:
Hi @adamgarbo ,
Sincere thanks (again) for all the BETA testing!
Today's BETA (PR #20) resolves the extra-messages-in-the-log-file issue.
Also included is a new feature where you can open a new log file every (e.g.) 10 seconds - should you want to. I added it mainly for test purposes - so I could create a large number of files quickly - but decided to keep it in as a useful feature.
Still to do: fix the IMU not-working-after-power-down issue.
Thanks again,
Paul

adamgarbo, 2020-08-28:
I thought I'd test for this issue with the new v10 hardware revision and I can confirm that the IMU bug is still present.

The settings to replicate this behaviour are:
* Default OLA v15 firmware
* Menu 1) Option 5) Set Log Rate in seconds between readings: 5
* Menu 7) Option 1) Turn off Qwiic bus power between readings (>2s): No

There's also some strange behaviour with the Serial Monitor going on, which I may create a separate issue for. The OLA is not able to re-establish a connection to the Serial Monitor when the log rate is >=2 seconds. Additionally, when connected via USB, if the Qwiic bus power remains on between readings, the OLA will enter a reset loop where it continually triggers the serial soft-reset.

```
rtcDate,rtcTime,aX,aY,aZ,gX,gY,gZ,mX,mY,mZ,imu_degC,output_Hz,
01/01/2000,00:12:37.30,547.36,-44.92,-847.66,-3.82,-1.69,-3.07,-74.55,58.80,-55.20,31.26,inf,
01/01/2000,00:12:42.94,0.35,
01/01/2000,00:12:48.51,0.27,
01/01/2000,00:12:54.05,0.24,
01/01/2000,00:12:59.60,0.22,
01/01/2000,00:13:05.14,0.22,
01/01/2000,00:13:10.69,0.21,
01/01/2000,00:13:16.23,0.21,
01/01/2000,00:13:21.78,0.20,
01/01/2000,00:13:27.33,0.20,
01/01/2000,00:13:32.87,0.20,
01/01/2000,00:13:38.42,0.20,
```

PaulZC, 2020-08-29:
Hi @adamgarbo ,

This is really useful - thank you.

I hadn't forgotten about this issue - but there are only 24 hours in a day! ;-) I'll dig into this as soon as I can.

All the best,

Paul

PaulZC, 2020-09-03:
After being power-cycled, the IMU says it is online (myICM.status == ICM_20948_Stat_Ok) but ICM_20948_data_ready is returning ICM_20948_Stat_NoData...

PaulZC, 2020-09-03:
Hi @adamgarbo ,

OK! I _think_ the IMU startup-after-sleep problem is fixed!

The IMU seemed to [need an extra 25ms to get its act together](https://github.com/sparkfun/OpenLog_Artemis/commit/30d705118667cf1aa4bec7307a99b664c90230d8#diff-3735031c49e2fb57cfeee5ca407276a8R624-R627) after wakeFromSleep.
I suspect that wakeFromSleep happens a bit quicker than setup and that the IMU wasn't quite ready with its first reading when getData checked dataReady.

Please kick the tyres and let me know if you see anything else sketchy!
Thanks!
Paul

adamgarbo, 2020-09-03:
Thanks, Paul!

What a lazy IMU! Will test and close this issue if all seems okay!

Cheers,
Adam

closed 2020-09-03 by adamgarbo