sparkfun/OpenLog_Artemis #21: Streaming some data over bluetooth at the same time ?

JPB1JPB1, 2020-07-03:
Hello

Great, but can we send a maximum of data over bluetooth too ? and code example at the same time ?
Br
J

PaulZC, 2020-07-04:
Hi @JPB1JPB1 ,
Bluetooth support on the Artemis is almost complete - but has not yet been released.
We will of course be adding substantial Bluetooth features to the OLA as soon as possible.
Thanks for the interest!
Best wishes,
Paul

closed 2020-07-04 by PaulZC

ryanneve, 2020-12-15:
Any update on BT? Would be nice if console could be on BT rather than just USB-C.

aerovenky, 2026-01-02:
@PaulZC 
Thanks a lot for the update,
I’m very interested in helping add BLE support to the OpenLog Artemis, at least for wireless console access to begin with and possibly for data streaming later. Before I start experimenting, is there any previous work, experimental branch, or notes you can share about attempts to enable BT on OLA?