sparkfun/OpenLog_Artemis #27: Firmware v15/v16 unable to export compiled Binary

adamgarbo, 2020-08-13:
Hi @nseidle,

Just a heads up that when attempting to export a compiled Binary from the v15/v16 firmware I run into the error:

```
/Users/adam/Documents/GitHub/OpenLog_Artemis/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino: In function 'void enableCIPOpullUp()':
OpenLog_Artemis:552:11: error: 'MISO' was not declared in this scope
   padMode(MISO, cipoPinCfg, &retval);
           ^~~~
exit status 1
'MISO' was not declared in this scope
```

Is MISO missing a corresponding definition?
https://github.com/sparkfun/OpenLog_Artemis/blob/b1eea9a1b4404d847f5af14be32d4e7bea92b54d/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino#L553

Cheers,
Adam

nseidle, 2020-08-13:
Hi Adam - I'm currently working on an SCD30 bug in v16 and can compile.

I tested with master branch of OLA. Still compiles.

I've using the main branch of [Arduino_Apollo3](https://github.com/sparkfun/Arduino_Apollo3). You might be using the board manager in Arduino which... I'm not sure what branch that pulls in.

Tell me more so I can try to replicate.

adamgarbo, 2020-08-13:
Hi Nathan,

Digging into this now.

Arduino 1.8.13
Using board 'artemis' from platform in folder: /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.1.2
Using core 'arduino' from platform in folder: /Users/adam/Library/Arduino15/packages/SparkFun/hardware/apollo3/1.1.2

Is `MISO` defined in the HAL somewhere?

Cheers,
Adam

nseidle, 2020-08-13:
MISO is defined in the variant file: https://github.com/sparkfun/Arduino_Apollo3/blob/master/variants/redboard_artemis_atp/config/variant.h#L78

adamgarbo, 2020-08-13:
Ahh... I see! 

You're compiling with the Redboard Artemis ATP, while I was using the SparkFun Artemis Module. After switching to the ATP I was able to compile.

Paul and I discussed the difference between the board definitions a while back and which was best to use. At the time, both worked but I'll stick with the ATP moving forward (at least until an OLA board definition exists).  

Cheers,
Adam

closed 2020-08-13 by adamgarbo