sparkfun/OpenLog_Artemis #29: Increase GNSS lat/lon logger precision to match ZED-F9P capabilities

n8ur, 2020-08-25:
The current logger reports latitude and longitude with 7 decimal places when receiving an NMEA GGA sentence, which is good for a little more than 1 cm.  That's pretty good, but the dual-frequency receivers can do better under the right conditions, and when the NMEA HIGH_PREC config is set, the F9 receivers report DD MM.xxxxxxx (minutes with 7 decimal places), which is close to 9 digits of decimal degree precision.

Suggest that the code look for the number of places contained in the NMEA sentence and adjust the output precision accordingly.

I don't see the parsing code in the main repo, so I am guessing that it lives in a library somewhere...

Thanks,
John

PaulZC, 2020-08-26:
Hi John ( @n8ur ),

Thank you for raising this issue.

The OLA does indeed use the SparkFun u-blox library to communicate with u-blox GPS/GNSS modules. But it does not parse NMEA messages. In fact we deliberately disable them. Instead we use u-blox's UBX binary protocol to retrieve the data:
https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/OpenLog_Artemis/Sensors.ino#L264

The relevant function calls are ```getLatitude``` and ```getLongitude```. Here's the relevant part of the library:
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/src/SparkFun_Ublox_Arduino_Library.cpp#L3038-L3060

Sadly, those functions are limited to seven decimal places as the UBX-NAV-PVT message returns the Lat and Lon as integers in degrees^-7.
The only way forward would be to use the getHighRes functions instead:
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/src/SparkFun_Ublox_Arduino_Library.cpp#L2918-L2948

I'm going to leave this issue open because it is a perfectly valid request and we will look into it as soon as we can.

For now, you might want to take a look at the dedicated GNSS Logger code for the OLA, which does support high precision positioning (UBX-NAV-HPPOSECEF and UBX-NAV-HPPOSLLH). But, sadly, the issue there is that the data is logged in u-blox UBX format and may be difficult for you to work with.
https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger

Best wishes,
Paul

n8ur, 2020-08-26:
Thanks for the quick reply, Paul!

Now I get what you're doing.  I've spent painful time parsing UBX binary, so can use the gnss logger code and process that format for the time being.  But adding the high-res capability would sure be nice!

BTW, I assume that whatever configuration you're writing to the ublox is to the ram layer only, and reverts when the receiver is rebooted?

Thanks!

John

PaulZC, 2020-08-26:
Hi John,

Another good question!

The OLA code saves (only) the IO port configuration to battery-backed RAM:
https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/OpenLog_Artemis/autoDetect.ino#L510

(Previously we were using ```saveConfiguration``` to save _all_ of the settings to BBR, but changed that recently to only save the IO port settings using ```saveConfigSelective```. This prevents the NMEA messages from restarting on the I2C port each time the power is cycled, but allows all of the other settings to go back to their default configuration.)

All the best,
Paul

n8ur, 2020-08-26:
OK, that makes sense.  My vote would be to make that a config option ("Save modified GNSS config to receiver?").  In my own case, I'm using the logger intermittently and would like to avoid having to reconfigure when I go back to another application.  This isn't a big deal, though.

PaulZC, 2020-10-13:
Hi John ( @n8ur ),

After a lot of thought, I'm going to close this request.

The main issue is that the OLA firmware is designed to work cleanly with _all_ u-blox GNSS modules with little or no configuration. To implement this request, we would need to interrogate the module, work out what its capabilities are (i.e. does it support high resolution positioning) and then adapt the way it logs lat and lon accordingly. It's possible - but it is not a clean solution and would need to be updated each time a new module becomes available. (The M9 family is coming soon!)

High resolution logging is supported in the GNSS Logger branch, and I believe that's the correct home for it.

Best wishes,
Paul

closed 2020-10-13 by PaulZC

n8ur, 2020-10-13:
I understand your thinking.  Thanks for considering the request.

BTW, the M9N has been shipping for a while and is very impressive -- 
tests on one unit show significantly better location performance than 
the M8N, and it costs a whole bunch less (in quantity).

John
----

PaulZC, 2020-10-13:
Thank you John,
All the best,
Paul