sparkfun/OpenLog_Artemis #30: Investigate serial logging issue identified in the GNSS Logger branch

PaulZC, 2020-08-26:
Investigate the serial logging issue identified by @adamgarbo in the GNSS Logger branch:
https://github.com/sparkfun/OpenLog_Artemis_GNSS_Logger/issues/8

Possible culprits:
- Serial RX data is stored in a 4KB RingBuffer. Is the buffer large enough to cope with RAWX data?
- Is the buffer large enough to store incoming serial data while normal OLA sensor data is being written to the sensor dataLog file?

PaulZC, 2020-08-26:
Hi @adamgarbo ,

I suspect we're just asking too much of the OLA when we have it writing RAWX data to the serial log file while also writing IMU data to the dataLog file at 10Hz. I just tried a quick test where I had serial logging enabled but the IMU logging completely disabled and I only saw one sync error:
- Menu 3, Option1 disables the IMU logging
- Menu 1, Options 1 and 2 will stop the OLA logging 'empty' timestamps to the SD card and to the terminal
- Use menu 4 as normal to enable serial logging at 230400 baud

Again, if you have time, can you please try the same test and see if most of your errors go away?

If that's the root cause, then I'm not completely sure on the best way forward. I could add an extra large RingBuffer to hold the serial data. But, I'd probably have to add a timer interrupt to automatically move data from the normal serial buffer into the larger ring buffer. And that then means more work to disable and re-enable it when going into and out of deep sleep...

Humm. I shall keep digging!

All the best,
Paul

adamgarbo, 2020-08-26:
Hi @PaulZC,

Thanks for the update! 

A quick note that in my previous test, I was logging the UBX and IMU data at only 1 Hz, with just the GPS/GLONASS constellations enabled.

Happy to do a side by side comparison later!

Cheers,
Adam

PaulZC, 2020-08-27:
OK. Quick update:
I ran a test last night, logging 4Hz RAWX and SFRBX from all constellations to the OLA over serial at 230400 baud. The file is completely clean. No errors at all:

[image]

I did two things:
- I made sure that the sensor logging was completely disabled via the menus, so the OLA was only logging the serial data to SD card
- I hacked ```ap3_uart.h``` in the core to increase the size of the rx buffer to 32K.

[image]

[image]

I'll check what happens if I now re-enable IMU logging...

PaulZC, 2020-08-27:
Here goes... I've re-enabled IMU logging at 1Hz. To begin the ZED was sending SFRBX and RAWX at 1Hz. At 08:02:18 I used u-center to increase the RATE to 4Hz. That stops the ```Total chars received``` debug messages. (You only get those when the code has time to spare and is writing partial blocks to the SD card. At 4Hz the card is busy writing full blocks and the debug messages disappear. They come back if you drop the RATE to 1Hz again...) I'll let you know how it goes.

[image]

PaulZC, 2020-08-27:
OK. That didn't go so well. I tried a three hour test and the ubx data loses sync over 100 times...
So it does appear that disabling the IMU logging is critical. (Understandable really. There's going to be a big overhead for the SD card when swapping between the two log files.)
Right. Let's figure out how big the RX buffer needs to be...

adamgarbo, 2020-08-27:
Hi @PaulZC,

You've been busy! I'm glad you also conducted a test. For some reason, the OLA configured with the IMU and other data logging disabled did not record any serial data whatsoever. I wonder if I disabled a setting incorrectly.

However, the OLA configured to log both IMU and serial data did collect data for approximately 11 hours. Running the data through your integrity checker consistently resulted in 10 resyncs per hour. When processing the data with RTKLIB, there were a total of 40,727 observations and 861 errors (about 2%).

Cheers,
Adam

PaulZC, 2020-08-27:
Ah. I've been a muppet... I had forgotten to change the value of ```processed``` when rewinding the file back beyond the loss-of-sync. The results make a bit more sense now. The file resyncs at a location _before_ the loss of sync, which is what you'd expect with dropped bytes from a buffer overflow. I've corrected the tool...

With the AP3 UART RX RingBuffer size set to 16KB I got **one** re-sync when logging 4Hz RAWX & SFRBX from all constellations for just over two hours. I'll try 8KB next...

[image]

PaulZC, 2020-08-27:
One loss of sync when logging at 4Hz for about 90 minutes with an 8K RX buffer:

[image]

PaulZC, 2020-08-27:
And the vote is in. The default 4KB serial ring buffer isn't quite large enough. I was logging SFRBX and RAWX at 4Hz on all constellations for about 4.5 hours and I see 17 sync errors. So, we do need an additional buffer of 8KB or more...

[image]

adamgarbo, 2020-08-27:
Hi Paul,

Thanks for the update. To confirm, this is with absolutely no logging to `sensorDataFile`?

Will the increase to the UART RX RingBuffer have any trickle-down effects elsewhere on the OLA?

I'm currently running a RAWX/SFRBX test at 1 Hz and recording only the date, time and battery voltage to a separate log file (useful information for any project). Curious to see how it impacts the UBX sync losses.

Cheers,
Adam

PaulZC, 2020-08-27:
Hi Adam,

Correct - nothing being logged to the sensor dataLog:
Menu 1, Options 1, 2, 7 and 9 all disabled
Menu 3, Option 1 - IMU logging disabled
Menu 4, Option 1 - Serial logging enabled
Menu 4, Option 2 - Baud rate set to 230400
Menu 5, Options 1-4 & 6 all disabled

Yes, so here's the dilemma. We're already using quite a lot of RAM due to the number of sensor libraries we've got loaded. Adding another serial buffer will increase the RAM use further. So, well need to leave more RAM enabled during deep sleep, which will push the sleep current up. I haven't measured the increase yet...

I'm going to try an overnight 4Hz test with an 8KB buffer. I'll report back in the morning.

Cheers!

Paul

adamgarbo, 2020-08-28:
Hi Paul,

My most recent test of the OLA resulted in only a single sync error in over 6 hours of logging.

I was collecting RAWX/SFRBX messages from all constellations at 1 Hz, along with the date, time and battery voltage. The IMU was disabled. It appears that the OLA can handle the lower rates without any major issues, but once they're increased we really are pushing its limits. Keen to learn how your tests went.

Cheers,
Adam

PaulZC, 2020-08-28:
Many thanks Adam,

I've added an extra 8K RingBuffer to the OLA code. The code moves serial data from the serial RX buffer into that around the SD card writes. Ran an overnight test (10 hours) logging SFRBX and RAWX at 4Hz from all constellations at 230400 baud. Only one resync! And that happened after the first second or so. I’ve seen this quite a bit. It’s like the SD card stalls when you first start writing to it and then it gets its act together.

[image]

Right. OK. I’m happy with that. Let’s see if we can now log time/date/battery/IMU data at the same time. The test is:
- log SRFBX and RAWX at 1Hz from all constellations
- log time, date, battery voltage and IMU data at the same time

More later!

PaulZC, 2020-08-28:
Humm. Well that was unsuccessful. 100+ resyncs in about 8 hours of serial data... The sensor data looks fine though.
I'll try increasing the buffer size.

PaulZC, 2020-10-13:
I think we are possibly being a bit optimistic to expect the OLA to be able to log large quantities of RAWX data while also logging other sensor data - including IMU data. I'm also not sure that there is a solution which will keep all users happy.

I'm going to close this as I believe the firmware meets the needs of _most_ users. We can use the GNSS Logger build to tackle the niche cases. There's work to do there (I'm aware the GNSS Logger is not as up to date as it should be!).

closed 2020-10-13 by PaulZC