sparkfun/OpenLog_Artemis #31: Measurement count behaviour

adamgarbo, 2020-08-27:
Hi,

A super minor issue to look at on a rainy day.

I've observed that when logging data with the OpenLog Artemis with the measurement count option enabled `9) Output Measurement Count: Enabled`, after the menu is accessed from the Serial Monitor and the OLA returns to logging, the measurement count is reset. This can result in log files with nonmonotonic counts.

The counter looks to be reset in the main menu. I wonder if there's a better spot for this to happen.
https://github.com/sparkfun/OpenLog_Artemis/blob/691648ff7ee321dc482b184756c59b76b26f449c/Firmware/OpenLog_Artemis/menuMain.ino#L124

Also, when a new log file is created, should we be resting the measurement count? I'm of two minds.

Cheers,
Adam 

```
09:49:52.422 -> rtcDate,rtcTime,VIN,count,
09:49:52.908 -> 2020-08-27,09:49:52.08,5.12,1,
09:49:53.906 -> 2020-08-27,09:49:53.08,5.12,2,
09:49:54.926 -> 2020-08-27,09:49:54.08,5.13,3,
09:49:55.914 -> 2020-08-27,09:49:55.08,5.13,4,
09:49:56.908 -> 2020-08-27,09:49:56.09,5.13,5,
09:49:57.151 -> 
09:49:57.151 -> Menu: Main Menu
09:49:57.151 -> 1) Configure Terminal Output
09:49:57.151 -> 2) Configure Time Stamp
09:49:57.185 -> 3) Configure IMU Logging
09:49:57.185 -> 4) Configure Serial Logging
09:49:57.185 -> 5) Configure Analog Logging
09:49:57.185 -> 6) Detect / Configure Attached Devices
09:49:57.185 -> 7) Configure Power Options
09:49:57.185 -> h) Print Sensor Helper Text (and return to logging)
09:49:57.185 -> r) Reset all settings to default
09:49:57.185 -> q) Quit: Close log files and power down
09:49:57.185 -> d) Debug Menu
09:49:57.185 -> x) Return to logging
09:49:58.133 -> 2020-08-27,09:49:57.28,5.11,1,
09:49:59.132 -> 2020-08-27,09:49:58.28,5.12,2,
09:50:00.123 -> 2020-08-27,09:49:59.28,5.12,3,
09:50:01.135 -> 2020-08-27,09:50:00.29,5.11,4,
09:50:02.119 -> 2020-08-27,09:50:01.29,5.11,5,
```

nseidle, 2020-08-27:
The reading output rate (Hz) is tied to the measurement count and the amount of time since power on. If the user has the sub menus open for any time at all, the reading rate gets screwy, so we reset those two variables each time the menus are closed out. 

I agree tho that the # of readings should reflect the readings since POR. We'll need to find a fancy way of tracking the output rate. Can be done, just adds a bit of overhead.

PaulZC, 2020-09-08:
Released in [Firmware Version 1.6](https://github.com/sparkfun/OpenLog_Artemis/releases/tag/v1.6)

closed 2020-09-08 by PaulZC