sparkfun/OpenLog_Artemis #33: Feature request: Output logs over Serial1 or USB

nseidle, 2020-08-28:
From [forum](https://forum.sparkfun.com/viewtopic.php?f=172&t=53726&p=218446#p218446)

>It would be great if USB could be used to move the CSV files from the OLA to my PC instead of only being used for configuration.

I interpret this as:

* Create a menu option that allows the listing of logs, selection of a log, then outputting that log over USB and Serial1 (if that output is enabled - see request #32)

This opens a can of worms: We should probably implement an embedable command framework (AT command set?) akin to the original OpenLog for opening/editing/deleting logs. For now, let's just do a menu based system.

PaulZC, 2020-08-28:
Back in the good old days, we had terminal emulators that supported XMODEM and ZMODEM file transfer...
Is that what we need?

slowtoaccept, 2020-08-29:
Agree. Add a menu item to, for example, enable streaming the SD CSV file dataLog00000.TXT to specified serial output port. Would allow periodic programmatic transfer of stored data to another microcontroller. 

Also, add space management of the SD store.

PaulZC, 2020-09-03:
v16 now includes a new "SD Card File Transfer" menu:
- supports DIR/LS & DEL/RM
- supports single and multiple file transfer using ZMODEM (tested with Tera Term - works well!)
- supports streaming of a single file to the serial TX pin

Please see [the latest commits on the v16_Candidate branch](https://github.com/sparkfun/OpenLog_Artemis/commits/v16_Candidate)

There's more to do on v16. Estimated Time of Release: W/C Sept 8th

PaulZC, 2020-09-08:
Released in [Firmware Version 1.6](https://github.com/sparkfun/OpenLog_Artemis/releases/tag/v1.6)

closed 2020-09-08 by PaulZC