sparkfun/OpenLog_Artemis #35: Support SparkFun Qwiic MicroPressure Sensor

ryanneve, 2020-09-02:
It would be nice if the [SparkFun Qwiic MicroPressure Sensor SEN-16476](https://www.sparkfun.com/products/16476) with the [Honeywell MPR series sensor](https://cdn.sparkfun.com/assets/2/e/8/0/9/honeywell-mpr-datasheet.pdf) was supported.

I would like to use the OpenLog Artemis to log king-tide flooding events, and this is the most appropriate sensor you offer.

Arduino Library here:
[https://github.com/sparkfun/SparkFun_MicroPressure_Arduino_Library](https://github.com/sparkfun/SparkFun_MicroPressure_Arduino_Library)

PaulZC, 2020-09-03:
Hi @ryanneve ,
We'll look into this.
Very best wishes,
Paul

PaulZC, 2020-10-12:
Hi @ryanneve ,
Thank you for your patience!
The latest [V10-v17_BETA binary](https://github.com/sparkfun/OpenLog_Artemis/tree/master/Binaries) now includes support for the MPR MicroPressure sensor.
Can you please give it a try? You can find the upgrade instructions [here](https://github.com/sparkfun/OpenLog_Artemis/blob/master/UPGRADE.md).
I'll leave this issue open until v17 is released.
Best wishes,
Paul

PaulZC, 2020-10-13:
Resolved in v1.7. Closing...

closed 2020-10-13 by PaulZC

ryanneve, 2020-12-15:
Seems to work well. Thanks!