sparkfun/OpenLog_Artemis #38: Fetch log records on demand over serial

PaulHuizer, 2020-09-08:
I'm looking for a way to fetch recorded log records from OLA from another microcontroller.
The primary function of OLA is to offer datalogging for a variety of sensors. That is a very helpful building block for a larger application. However, if the larger application is running on another MCU, it would be very usefull if that MCU could access the recorded logs. The primary usecase I'm looking for is, that I want to use another MCU to send these recorded log files to a webserver for further processing.

That scenario would need features like: get current OLA timestamp, get list of files, fetch file, fetch file with n bytes offset, delete file ...
A suitable transport mechanism: serial?
This would also mean that the logged data would ideally be organised in a file structure that would fit this scenario, for example, create new file per sensor per hour, or per day ... The file structure could perhaps be a OLA config setting?

PaulZC, 2020-09-09:
Hi Paul ( @PaulHuizer ),
Please check the release notes for version 1.6 of the OLA firmware (released yesterday!):
https://github.com/sparkfun/OpenLog_Artemis/releases/tag/v1.6
You can now list and transfer files from the OLA through the terminal (the firmware now supports ZMODEM transfers) or copy them to the serial TX pin. The log files are now timestamped correctly too.
You can find the upgrade instructions here:
https://github.com/sparkfun/OpenLog_Artemis/blob/master/UPGRADE.md
Changing the file structure is something we might do in the future, but we don't have plans to implement that right now.
I'm going to close this issue as I think v1.6 should meet your needs, but please reopen if I've missed anything.
Please feel free to raise a fresh "Feature Request: implement directory structure based on..." issue if the single directory really doesn't work for you.
Very best wishes,
Paul

closed 2020-09-09 by PaulZC

PaulHuizer, 2020-09-09:
The functionality seems right, but sorry for the newby question: how can the list and transfer files features be accessed from another mcu?

PaulZC, 2020-09-09:
Ah, OK, sorry. I'd missed that point. Unless you can connect the OLA to a USB host port which can then open the OLA menus, there is currently no way to do this.
You may be able to do it by 'moving' the new SD Card File Transfer from ```Serial``` (USB) to ```SerialLog``` (the TX and RX pins).
Have a look at these lines:
https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/OpenLog_Artemis/zmodem_config.h#L19-L26
It _may_ be as simple as changing ```ZSERIAL``` from ```Serial``` to ```SerialLog```.
You would also need to add extra code in the main loop to 'open' the SD Card Menu in response to some kind of command sent to the RX pin.
I'm afraid this is too specific for us to consider including it in the OLA code. It would cause too much incompatibility with the way we currently use the TX and RX pins for serial logging.

PaulHuizer, 2020-09-09:
I'd hoped having the ability to fetch the data from the logger SD card in a 'MCU friendly' way would not be a 'too specific' feature.  The MCU I need in this application has specific capablities, but it cannot act as USB Host. The remaining option would be to modify the OLA firmware, but that seems like a lot of hassle and defeats the purpose of this readily available product for this usecase.  It is probably easier to connect the sensors direct and write output to the SD Card.

ryanneve, 2020-12-15:
Just to add another use case. I'd like to be able to access logger data wirelessly. I can hook a BT module to RX and TX, but there's no way to download files. I can stream to BT, but I'll have gaps when my logger is under water.