sparkfun/OpenLog_Artemis #39: readVin() before and after sleeping mode [BUG]

bastientaillandier, 2020-09-16:
### Subject of the issue
Reading Vin value throught readVin() function bug when ALO enter on sleep mode (after calling goToSleep() function).

### Your workbench
I log IMU data at 5Hz at the beggining and print readVin() value. With just an USB alim, I have about 5.3V.
I cut the jumper between MEAS pins to put a 3.7V battery for the alim. I connect the +pin of the alimentation to Vin
to measure the voltage of the battery. The voltage is about 4.2V.

After 5s, I change the logging frequency to 0.2Hz (once per 5s), then OLA go to sleep mode, and when it wake up the voltage
of readVin() is about 0.2V.

### Steps to reproduce
Connect OLA to USB, check readVin() value before and after sleep mode.

[image]

[image]

### Expected behavior
Same value before and after sleep mode

### Actual behavior
Strange value after sleep mode

PaulZC, 2020-09-16:
Hi Bastien (@bastientaillandier ),

Thank you for finding this issue! I can confirm that I see the same thing (using firmware v1.6 and Apollo3 1.2.0):

[image]

I am investigating!
Very best wishes,
Paul

PaulZC, 2020-09-16:
Hi Bastien (@bastientaillandier ),

OK. For reasons I don't completely understand, we need to call ```ap3_set_pin_to_analog``` after sleep to make the analog pins behave correctly:
https://github.com/sparkfun/OpenLog_Artemis/blob/release_candidate/Firmware/OpenLog_Artemis/lowerPower.ino#L323-L331

I have added a fix for this in the [release-candidate branch](https://github.com/sparkfun/OpenLog_Artemis/tree/release_candidate) and have copied the new BETA version of the firmware binary into the [master branch](https://github.com/sparkfun/OpenLog_Artemis/tree/master/Binaries).

Can you please test the fix and make sure it corrects your issue?
Very best wishes,
Paul

bastientaillandier, 2020-09-16:
Hi @PaulZC 

I try it myself and the bug doesn't appear anymore.
Thank you and congratulation !

PaulZC, 2020-09-16:
Hi Bastien,

That's great news - thank you for letting me know.
I am going to leave this issue open until we release v1.7.

Thank you,

Paul

PaulZC, 2020-10-13:
Resolved in v1.7. Closing...