sparkfun/OpenLog_Artemis #40: Feature Request: Add Analog Voltage Threshold Trigger

wangcheung, 2020-09-24:
I'm using the OLA to capture events and it would be great if there was an option to turn on event logging as a result of inputs on the analog inputs, more of a discrete datalog that would only log sample times on a logic low to high event or vice versa.

additionally, for another option it might be useful to only log RTC data at specific analog voltage levels. for ex, above 1.1 volts or less than .9 volts, ....

My specific use case is the first though, trying to only log RTC times for low to high events.

thanks

PaulZC, 2020-09-25:
Hi @wangcheung ,
Can you please provide a few more details?
Do you want the logger to log one time when the analog signal goes from low to high?
Or continuously, while the analog signal is above the threshold voltage?
Are you trying to reduce the amount of data which is logged to the SD card? Or, are you trying to save power?
v1.6 supports digital triggering on pin 11. Please see the Configure Terminal Output section in the [Hook-Up Guide](https://learn.sparkfun.com/tutorials/openlog-artemis-hookup-guide/configuration) for more details. Maybe you can use an op-amp as a comparator, to convert your analog signal to digital, and then use pin 11 to trigger the data collection?
Best wishes,
Paul

wangcheung, 2020-09-25:
I'm actually not looking to use analog but digital logging. I would like to log one data point (time stamp) when ever the input transitions from low to high (or high to low for an option but not at the same time). I am looking to reduce data on the SD card. I figured some people might also like the analog threshold option, but yes this could be accomplished with an op amp easy enough. I'll review the updated v 1.6 pin 11 triggering.
thanks

wangcheung, 2020-10-06:
pin 11 triggering works well, nice implementation. I guess the only improvement might be to add a debounce time but I could always just add a cap. Thanks, works well

closed 2020-10-06 by wangcheung