sparkfun/OpenLog_Artemis #42: Feature Request: Capability to Set ICM-20948 Full-Scale Range and Filtering Values

HowardRush, 2020-10-08:
### Subject of the issue
Please consider enabling the user to set the following programmable parameters of the ICM-20948 inertial measurement unit on the OLA board:
ICM-20948 Datasheet (https://cdn.sparkfun.com/assets/7/f/e/c/d/DS-000189-ICM-20948-v1.3.pdf) section and value:
10.2 GYRO_CONFIG_1, bit 5:3, GYRO_DLPFCFG[2:0], gyro low-pass filter configuration
10.2 GYRO_CONFIG_1, bit 2:1, GYRO_FS_SEL[1:0], gyro full-scale select
10.2 GYRO_CONFIG_1, bit 0, GYRO_FCHOICE, bypass or enable gyro DLPF (digital low-pass filter, I presume)
10.15 ACCEL-CONFIG, bit 5:3, ACCEL_DLPFCFG[2:0], accelerometer low-pass filter configuration
10.15 ACCEL-CONFIG, bit 2:1, ACCEL_FS_SEL[1:0], accelerometer full-scale select
10.15 ACCEL-CONFIG, bit  0, ACCEL_FCHOICE, bypass or enable accelerometer DLPF (digital low-pass filter, I presume)

My flying-machine application requires +/- 500 degrees/second gyro rates and +/- 16 g accelerations.  I am taking data at 100 Hz, which aliases noise from a motor and propeller, hence the need to filter signals upstream of the sampling.  Parameter values I would want are:
10.2 GYRO_CONFIG_1, bit 5:3, GYRO_DLPFCFG[2:0] = 1 or 2 or 3
10.2 GYRO_CONFIG_1, bit 2:1, GYRO_FS_SEL[1:0] = 01
10.2 GYRO_CONFIG_1, bit 0, GYRO_FCHOICE = 1
10.15 ACCEL-CONFIG, bit 5:3, ACCEL_DLPFCFG[2:0] = 1 or 2 or 3
10.15 ACCEL-CONFIG, bit 2:1, ACCEL_FS_SEL[1:0] = 11
10.15 ACCEL-CONFIG, bit  0, ACCEL_FCHOICE = 1

Thanks for considering this request.  I think other folks would also like to be able to set these parameters.
.

### Your workbench
* Are you using a microSD card? Yes, 1 or 32 GB, FAT 32
* At what frequency are you logging? 100 Hz
* What version of firmware are you using? the latest
* How is OpenLog Artemis wired to your sensor(s)? I'm currently only using the ICM-20948 that's on the board.
* How is everything being powered? 5V regulated power aboard the airplane, USB for setting values on the ground.

PaulZC, 2020-10-12:
Hi @HowardRush,
Thanks for raising this feature request.
It looks like it should be fairly straight forward to implement. (Famous last words!)
The backbone will be this code from ```startupDefault```:
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/ICM_20948.cpp#L656-L687
using these enums:
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_ENUMERATIONS.h#L154-L170
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_ENUMERATIONS.h#L187-L203
I'll give it a go tomorrow and will let you know how I get on.
Best wishes,
Paul

PaulZC, 2020-10-13:
Hi @HowardRush ,

The latest [V10-v17_BETA binary](https://github.com/sparkfun/OpenLog_Artemis/tree/master/Binaries) lets you set the IMU full scale and digital filter bandwidths via Menu 3 (Configure IMU):

[image]

Can you please give it a try? You can find the upgrade instructions [here](https://github.com/sparkfun/OpenLog_Artemis/blob/master/UPGRADE.md).

I'll leave this issue open until v17 is released.
Best wishes,
Paul

PaulZC, 2020-10-13:
Resolved in v1.7. Closing...

closed 2020-10-13 by PaulZC

HowardRush, 2020-10-13:
Thanks, Paul, I’ll try it.

 

Howard