sparkfun/OpenLog_Artemis #45: High gyro noise

smichpresel, 2020-10-16:
Hello to the sparkfun / OpenLog_Artemis team.

I enjoy using your nice device, however is seems that the 'no motion' gyro readings are very high. I am receiving readings in the range of +-10 degrees / second. Based on the gyro datasheet I would be expecting noise approximately one order of magnitude smaller.

By manually trying a constant turn rate - the scale of the measurement readings seems correct.

### Setup
Artemis OpenLog v1.6 (but same observed on previous versions)
Tested 4 different Artemis OpenLog  boards
Bare board connected to USB

## Reproduce
Results can be directly seen using arduino Serial plotter

## Things that didn't help
- compiling and flashing new Artemis OpenLog v1.6 using arduino toolchain
- setting DLPFs using the gyro lib (filter does help a little but still

[image]

PaulZC, 2020-10-16:
Hi @smichpresel ,

Version 1.7 of the firmware now lets you set the DLPF through the Configure IMU Logging menu. Can you please try updating to 1.7, set the Gyro DLPF to a low frequency, and post the results?

Many thanks,

Paul

smichpresel, 2020-10-16:
Thanks Paul for the fast response!

Updating to v1.7 resolves the issue. 

Already the no DLPF setting reduces the noise to expected levels and setting the DLPFs allow to go +-0.5dps (50Hz cuttoff).

PaulZC, 2020-10-16:
Hi @smichpresel ,
Great! Glad that's fixed your issue. Thanks for letting me know.
Best wishes,
Paul

closed 2020-10-16 by PaulZC