sparkfun/OpenLog_Artemis #46: Feature request:  OLA SLEEP AND WAKE AT SPECIFIED TIMES

alino93, 2020-10-20:
### Subject of the issue
Hi!
I have a project in which we are logging data from the movement of patients using IMU. I'm using the on-board IMU.
I cannot have a battery larger than 1Ah since it will become heavy and big to carry around for the patients.
I need to somehow extend the battery life to 5 days. right now it only works for max 2 days. The patients suffer from movement problems so I cannot ask them to turn the boards on and off every day.

One solution that came to my mind and I think could be a good feature on OLA is to sleep and wake at specified times during a day. (using RTC).
For example, I need to log data from 10/19/2020 to 10/25/2020 from 9 AM to 3 PM each day so the OLA should go to sleep at other times to save battery. This could be a great feature for projects that there is no access to the board to turn it off.

Thanks for considering this! I have seen several guys in the forum looking for a way to extend battery life and this might be a solution for some of them.

### Your workbench
* Are you using a microSD card? yes a 32 GB Fat32
* At what frequency are you logging? 100 Hz
* What version of firmware are you using? v1.7
* How is OpenLog Artemis wired to your sensor(s)? onboard IMU
* How is everything being powered? 1Ah 3.7v lipo battery

adamgarbo, 2020-10-20:
Hi @alino93,

Really interesting to read about your application for the OpenLog Artemis!

I believe RTC alarm functionality would be a great feature to have on the OLA and is in fact the main reason I added the necessary functions to the [Apollo3 RTC library](https://github.com/sparkfun/Arduino_Apollo3/blob/4885462145201549519799acdc59374a81c211b7/libraries/RTC/src/RTC.h#L32-L36):

```
void getAlarm(); //Query the RTC for the current alarm time/date
void setAlarm(uint8_t hund, uint8_t sec, uint8_t min, uint8_t hour, uint8_t dayOfMonth, uint8_t month); // Set alarm time
void setAlarmMode(uint8_t mode); //Set the RTC alarm repeat interval
void attachInterrupt(); //Attach the RTC alarm interrupt
void detachInterrupt(); //Detach the RTC alarm interrupt
```

At present, the OLA primarily uses the RTC for timestamping logged data and a microsecond counter for all timing related operations. Starting/stopping logging at specific times would require that the start/stop datetimes be stored in memory and updated each time after the alarm triggers. This is because the Apollo3 only has a single RTC alarm register. While certainly feasible, there may be considerable effort involved with integrating the RTC to work alongside the microsecond counter.

That being said, I've been able to successfully add RTC alarm functionality on my fork of the OLA to have it create new log files at specific dates and times, and to timestamp log files to have unique file names:

Configure RTC:
https://github.com/adamgarbo/OpenLog_Artemis/blob/631ec2a3249357012c743b1716abf8f0d5edc9c1/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino#L827-L841

Check for RTC alarm & create new log file:
https://github.com/adamgarbo/OpenLog_Artemis/blob/631ec2a3249357012c743b1716abf8f0d5edc9c1/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino#L530-L554

Cheers,
Adam

alino93, 2020-10-20:
So, it is not easy to implement this feature on the OLA. Therefore, I should not wait for this feature in the near future. Am I correct?
Could you explain more why it needs considerable effort to update the alarm each time it triggers?
Thanks!

PaulZC, 2020-10-21:
Hi @alino93 ,

Thank you for raising this issue (after our [forum discussion](https://forum.sparkfun.com/viewtopic.php?p=219920#p219888)).

I'm afraid this is not an easy request, and we won't be able to implement it soon.
The OLA code is currently quite simple. If the logging interval is > 2 seconds, the code wakes, takes a measurement, and goes back to sleep for a fixed period until it is time to wake again.
To implement your request, we will need to make the code:

- Set an RTC alarm.
- Sleep.
- Wake at the RTC alarm.
- Log continuously at 100Hz for perhaps 6 hours.
- Set the next RTC alarm.
- Sleep.

While still being able to do what most users want to do:
- Wake
- Take a single measurement
- Sleep
- Repeat

It is a major change...
Best wishes,
Paul

alino93, 2020-10-21:
Paul,
Okay, that makes sense, thanks for your explanation!

How about thinking of this in another way: I saw in the forum that the IMU supports something called wake on motion. Is that possible to use the IMU to wake the OLA and goes to sleep again when it is not moving? Or goes to sleep in a fixed period of time every time it wakes?

Thanks!

PaulZC, 2020-10-21:
Hi @alino93 ,
Yes, that approach could work well and might be less work than implementing the RTC alarms.
I will think about this some more...!
Best wishes,
Paul

ryanneve, 2020-12-15:
Can the RTC alarm wake up the OLA from a deep sleep (not just sleep) so that the RST button doesn't have to be pressed?
The Apollo3 datasheet talks about the Wake-up Interrupt Controller (WIC) and ALM interrupt control "_Once in the Deep Sleep Mode, the PMU, in conjunction with the Wake-Up Interrupt Controller, waits for a wakeup event._ ", but I couldn't tell if this means an alarm can wake from Deep Sleep.

If so, you just need a way to trigger pin 32 functionality to go to sleep, and to set a RTC alarm which will wake the  OLA from deep sleep.

Several mechanisms come to mind. 

- A daily ON (RTC alarm) and OFF (Go into Deep Sleep) time.
- A schedule file on the SD card, either with or without dates.
- An alarm set via the console menu (read a command from the console)

In my case, I only want to log pressure during high tides for which I have a schedule. I'd love to be able to leave out the logger for months, but only log during the peak of high tide.

PaulZC, 2020-12-18:
Hi @alino93 & @ryanneve,

I have looked into the "wake on motion" interrupt and I have had the IMU generating motion interrupts on a different project, but the code is not quite there yet. It needs a bit more work. (It is code for a new product which uses the same IMU and which is not public - yet.)

I will implement "wake on motion" - but it will not be soon. Sorry! And just to be clear, to keep the code changes manageable, I'm proposing that the OLA will wake on a motion interrupt, log _one_ reading to SD card, and go back into deep sleep. It will be similar to the "external trigger" feature we added in #36, except the trigger will come from IMU motion..

The sleep schedule that Ryan is proposing is a _lot_ of work. Especially if we throw a schedule file on SD card into the mix. I'm afraid I will not be able to implement that.

Best wishes,
Paul

alino93, 2020-12-18:
Hi, @PaulZC,

Thank you for considering this. But now that you are working on this, could you consider to log more than one reading? For example, it takes a constant value like 100 readings or 10 seconds in the menu, and every time it wakes, logs 100 readings then goes back to sleep.
Because I am logging motion data which is longer than one data!

Thank you,
Ali

PaulZC, 2021-01-29:
Hi @alino93 & @ryanneve ,
Apologies for the very slow response on this!
Ryan (@ryanneve) submitted a Pull Request containing some very useful code changes. I wasn't able to merge his code changes as-is but they did make me think about if it would be possible to give you your "wake at specified times" functionality. I think I've come pretty close!
You can now decide to switch into a new slow logging mode at specified times. Please have a look at the new entries in the Terminal Logging menu. If you're interested, the code changes are [here](https://github.com/sparkfun/OpenLog_Artemis/commit/61c1118a16d418b133f2eab575cf0a24d7d1d5cb).
The [v1.9 BETA binary](https://github.com/sparkfun/OpenLog_Artemis/tree/master/Binaries) includes the changes too. I'd be grateful if you can give it a try and let me know if you find any gremlins!
Use option 17 to enable "use RTC for slow logging". The default start and stop times should be close to what you need but you can of course change them. To begin with, I'd recommend leaving the slow logging interval set to 5 minutes. You can increase it to maybe 30 minutes but I'd recommend giving 5 a try first.
Please note: there is no error checking on the combination of start and stop times and the interval. It is possible to set the interval to be longer than the on time, in which case the OLA will probably never return to normal logging.
Please experiment with the settings and let me know if it does what you need.
If you get stuck, simply connect the OLA to your computer and open the Serial Monitor. The Main Menu will appear and you can use "r" to completely reset the settings if required.
Very best wishes,
Paul

alino93, 2021-01-29:
Hi @PaulZC,

Thank you for the new option. I have tested it with 5 min intervals as you said for 1 hour and it seems to work fine. But The option that I need is to log at 100 Hz not just log one data in 5 min intervals. I am sorry If I was not clear, I need to log at 100 Hz from 7:00 to 21:00 for example. Is that possible to implement?

Best regards,
Ali

PaulZC, 2021-01-30:
Hi Ali (@alino93),

I've just updated the v1.9 BETA binary. It now lets you enter the slow logging start and end times as hour and minute, instead of minute-of-day. It's a bit more intuitive.

I think the code is doing exactly what you need - but please do check. I just did a quick test where I was logging 'fast' or 'normal' IMU data at close to 100Hz (86Hz actual), started slow logging at 10:02, ended slow logging at 10:03, and it went back to 'fast' logging:

[image]

Something to be aware of is that going into slow logging does mess up the logHertz value. You will see that it is stuck at 36Hz after the slow logging. That is because the logHertz is averaging both fast and slow readings. It is reporting the true average, but it is confusing. I don't think there is anything I can do about that but I will think about it some more...

For your application, you need to start slow logging at say 15:00 or 21:00, and stop slow logging at 09:00. This should extend your battery life by 2-4 times?

Anyway, I hope this gives you the features you need. Good luck with your project - I hope it is successful for you. Please let me know if you need more help.
Best wishes,
Paul

alino93, 2021-02-01:
Hi @PaulZC,
Thank you very much for this.

I have tested the beta version again for two days and it works very well as expected. It would be nice if the files were stored separately for each day. 

I have also encountered a new problem that I should probably open a new thread for it. In summary, it seems that the clock loses accuracy during deep sleep, I have noticed that two synced sensors (start at the same time) if log data and stay on might lose clock sync for about 30 ms/day (~ 1ms/hr) which is acceptable for my work but if I turn them off (using pin 32) the RTC clocks would be off much faster, like 300 ms/day (~ 10ms/hr) which is a lot and not acceptable for our project. Is there a solution? 

Thank you and best regards,
Ali

closed 2021-02-01 by alino93

adamgarbo, 2021-02-01:
Hi @alino93,

A drift of 300 ms/day is equivalent to approximately +/- 3.5 ppm and 30 ms/day is <0.5 ppm. These are impressive results for the Artemis! Even dedicated RTCs usually don't deliver better than +/- 1.5 ppm. I'm not sure much more can be expected from the OLA without adding an external source to continuously synchronize the clocks.

I've never actually looked at how the Artemis maintains its clock when powered off, so I'd be curious to see where the additional drift is coming from.

alino93, 2021-02-01:
Hi @adamgarbo,

I am comparing the cross-correlation between signals of two OLAs with the same input at the same time, Maybe my method is not very accurate. 
For example, I get 1.68 s time difference between the two OLAs, and after 77 hrs I get 1.60 s time difference. 
So the RTCs seem to have 0.08 seconds drift compared to each other. (they might have a higher drift compared to an external clock) 

I have got similar results after other trials.

[image]

adamgarbo, 2021-02-01:
Very interesting graphs!

The OLA relies on the 32 kHz crystal and there are a number of sources of error that can produce differences in timing (e.g. inherent differences in the crystal oscillators and the onboard capacitors). There are some good discussions on the SparkFun forums about this: https://forum.sparkfun.com/viewtopic.php?f=170&t=53962

Robin also seems to have created a way to calibrate the RTC on the cheap with a GPS:
https://forum.sparkfun.com/viewtopic.php?f=170&t=54814