sparkfun/OpenLog_Artemis #47: Feature Request: Add IMU Euler Angles

max-itsl, 2020-10-21:
### **Subject of the issue**
Good day!
Please consider adding menu option to make OLA output “fused” IMU data in Euler Angles format:
**Roll (x), Pitch (y) and Yaw (z)**

Now OLA’s IMU outputs raw sensor data. By default it looks like this:
**rtcDate,rtcTime,aX,aY,aZ,gX,gY,gZ,mX,mY,mZ,imu_degC,output_Hz,**
Where
aX,aY,aZ – 3 axis accelerometer data (milli g)
gX,gY,gZ – 3 axis gyro data (Degrees per Second)
mX,mY,mZ – 3 axis magnetometer (micro Tesla)
https://github.com/sparkfun/OpenLog_Artemis/blob/release_candidate/SENSOR_UNITS.md#ICM-20948-IMU

Real world example:
01/02/2000,17:51:23.50,0.00,-45.41,1009.77,0.29,0.15,0.54,-0.90,-11.55,-37.95,28.56,19.42,
01/02/2000,17:51:23.60,3.91,-27.83,1007.81,-1.03,0.68,2.56,-0.45,-10.35,-36.15,28.17,14.78,
01/02/2000,17:51:23.70,6.35,-30.27,1019.04,2.79,0.41,0.69,-0.15,-10.05,-35.85,28.41,13.20,
I attach XLS file with more test data and some graphs (see second Excel sheet). I’ve put OLA on my desk, in a small box, and rotated it first over Y-axis, then over Z-axis, and then over X-axis. The test was not perfect, but it gives understanding what kind of data IMU outputs in relatively stationary conditions.
https://drive.google.com/file/d/1neTTJznOVbHkL8Lq28kz-av9kDZ-GDB4/view?usp=sharing

**Why do we need roll, pitch, and yaw angles?**
Because some applications by default use this format. For example, photogrammetry software (f.ex. Agisoft Metashape) accept only this format of photo orientation. With Euler angle output you will be able to attach OLA to a camera and get precise orientation data, which saves 40% project computation time (for large projects it can be days!).
Here is an example of DJI Mavic Mini drone orientation data. **This is what I would like to receive from OLA:**
https://drive.google.com/file/d/1klcJAtoMlrqTopzrh9Jx1AAVcEs6Agl5/view?usp=sharing 

**I’ve also found some information, maybe it will help:**

1. 
https://cdn.sparkfun.com/assets/7/f/e/c/d/DS-000189-ICM-20948-v1.3.pdf
ICM-20948 datasheet says that IMU has integrated Digital Motion Processor (DMP) with following features:
• **Offloads computation of motion processing algorithms from the host processor.** The DMP can be used to minimize power, simplify timing, simplify the software architecture, and save valuable MIPS on the host processor for use in applications.
• The DMP enables ultra-low power run-time and **background calibration** of the accelerometer, gyroscope, and compass, maintaining optimal performance of the sensor data for both physical and virtual sensors generated through sensor fusion. This enables the best user experience for all sensor enabled applications for the lifetime of the device.
• DMP features simplify the software architecture resulting in quicker time to market…”

2.
https://invensense.tdk.com/developers/support-center-faq/
Euler angles are a representation of an angular frame, as are quaternions and rotation matrices.
Euler angles are a set of three angles, corresponding to pitch, roll, and yaw.
Euler angles may be constructed according to many conventions. The axes of and ordering of ordering of pitch, roll, and yaw vary by convention. **The Embedded MotionApps Platform makes Euler angles available in the three most common conventions with MLGetEulerAnglesX, MLGetEulerAnglesY, and MLGetEulerAnglesZ.** See the Embedded MotionApps Platform Functional Specification for specific details on the definition of these conventions…

What items 1 and 2 mean? I think they mean this:
•	ICM-20948 is advanced IMU that has internal motion processor that **can provide solution straight in Euler angle format**
•	ICM-20948 DMP constantly performs background self-calibration, **no need to provide IMU calibration data manually**
•	InvenSense provide developers with **ready to use getters** to extract this data: MLGetEulerAnglesX, MLGetEulerAnglesY and MLGetEulerAnglesZ

IMHO, it is awesome, because normally to convert raw data to Euler angles (perform sensor fusion) we have to use some external MPU board (like Arduino) to receive sensor data, perform calculations, using one of existing algorithms (Mahony, Madgwick or other) and output solution. And this process normally requires sensor calibration procedure before getting actual measurements.
Here we have only to use getters and receive ready-to-use Euler angle data.

**If working with ICM-20948 DMP is not an option, I kindly ask to consider adding IMU fusion capabilities to OLA’s MPU as it has plenty of power.**

More reading about this topic:
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/1 Here the DMP question seems to be solved. **They get 40Hz yaw/pitch/roll over i2c**. The code seems to contain a solution to parse the DMP output

https://github.com/micropython-IMU/micropython-fusion Here "fusion.py" file contains Madgwick algorithm implementation for IMU raw data fusion to Euler angles

https://github.com/arduino-libraries/MadgwickAHRS Arduino implementation of Madgwick algorithm

https://learn.adafruit.com/how-to-fuse-motion-sensor-data-into-ahrs-orientation-euler-quaternions?view=all
https://learn.adafruit.com/ahrs-for-adafruits-9-dof-10-dof-breakout/sensor-fusion-algorithms
https://github.com/PaulStoffregen/NXPMotionSense
https://www.pjrc.com/store/prop_shield.html

My workbench
* Are you using a microSD card? – low latency (15ms) Sandisk Extreme, 32 GB, MicroSDHC, Class 10, UHS-I, U3. FAT32
* At what frequency are you logging? - From 10Hz to 100Hz
* What version of firmware are you using? - OpenLog_Artemis-V10-v17.bin
* How is OpenLog Artemis wired to your sensor(s)? – Using internal IMU
* How is everything being powered? - USB
* Are there any additional details that may help us help you? – Described above

Thank you 
Regards
Max.P

aedge72, 2020-10-31:
i'm also using the OpenLog Artemis and have the same question: how do i access the DMP data?  Tried the code posted here sparkfun/SparkFun_ICM-20948_ArduinoLibrary#1  but it's not working.  You would think that the manufacturer would provide easy to follow instructions to access all of the functionality, for amateurs like me.

aedge72, 2020-10-31:
to be more precise with my question, i want to download the quaternion raw data (w,x,y,z).   That would be very helpful. thanks.

lebarsfa, 2020-12-20:
FYI, I added the support to the OpenLog Artemis to the AHRS firmware originally designed for the old Razor IMUs : https://github.com/lebarsfa/razor-9dof-ahrs. This firmware computes the Euler angles without using the DMP.

aedge72, 2020-12-22:
Excellent I’ll check it out. Thanks!

ormapper, 2020-12-30:
+1 for this request please!

isirg, 2021-02-12:
+1 for this request!

mattevs, 2021-04-26:
+1
Op from the IMU of the Euler angles or quaternions would be fab!
Thanks for your great work :)

PaulZC, 2021-04-26:
Hi @max-itsl / @mattevs / all,
We've finally got the [ICM20948 DMP](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/releases/tag/v1.2.5) working nicely, so I will begin working on this request soon. It will take quite a bit of work because the standard IMU library functions and the DMP work very differently. I will need to add a separate block of code for the DMP with a completely different set of menu options. You may lose the IMU temperature logging (I'm not sure yet).
If you want a sneak preview of the DMP features, you will find three new stand-alone examples in the [release_candidate branch](https://github.com/sparkfun/OpenLog_Artemis/tree/release_candidate/Firmware/Test%20Sketches). I have tested them using v1.2.1 of the SparkFun Apollo3 boards, with the SparkFun RedBoard Artemis ATP selected.
I will update you when I have more news.
Best wishes,
Paul

PaulZC, 2021-04-26:
Hi All,
I've started adding DMP support to the OLA. It is little more than proof-of-concept at the moment, but, if you would like to give it a try, please download and install the [v1.11 BETA](https://github.com/sparkfun/OpenLog_Artemis/tree/master/Binaries).
If you open Menu 3 (IMU Logging) you will see a new option (12) to enable the DMP. Once enabled, you get the option to log either Game Rotation Vector (6-axis Quaternion) or Orientation Vector (9-axis Quaternion plus Heading Accuracy); plus the 'raw' DMP accelerometer, gyro and/or compass data.
The Quaternions are unitary (+/- 1). The DMP only outputs Q1-Q3. The Q0 value can be computed from this equation: Q0^2 + Q1^2 + Q2^2 + Q3^2 = 1. I'm not sure if the [sign of Q0](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/55) matters, but if you find a clever way to calculate it, please let me know. I'm not sure what the Accel, Gyro and Mag units are; the logged data is just an exact copy of the DMP int16_t data.
Please kick the tyres and let me know what you find.
Best wishes,
Paul

mattevs, 2021-04-29:
Hi Paul,

That’s great - Artemis Openlogs now on order and I will try next week!

Thanks
Matt

This email and any attachments to it may be confidential and are
intended solely for the use of the individual to whom it is 
addressed. If you are not the intended recipient of this email,
you must neither take any action based upon its contents, nor 
copy or show it to anyone. Please contact the sender if you 
believe you have received this email in error. QinetiQ retains 
personal data relating to our customers and partners for the 
purposes of conducting a business relationship, communicating
and marketing to them as well as to providing invitations to 
upcoming events.  
Please see our Privacy Notice ( https://www.qinetiq.com/Privacy-Policy )
for further information.  In accordance with our Privacy Notice, you
have the right to withdraw your consent at any time. QinetiQ may 
monitor email traffic data and also the content of email for 
the purposes of security. QinetiQ Limited (Registered in England
& Wales: Company Number: 3796233) Registered office: Cody Technology 
Park, Ively Road, Farnborough, Hampshire, GU14 0LX  https://www.qinetiq.com.

Uygarkuzuoglu, 2021-05-05:
Hi. I tried V1.11 beta but ı cant see yaw pitch roll data ?

PaulZC, 2021-05-05:
Hi @Uygarkuzuoglu ,
At the moment you can only log the Quaternion data (Q1-Q3) from the DMP. For now, you will need to manually calculate the Euler angles yourself. There is some [example code](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example7_DMP_Quat6_EulerAngles/Example7_DMP_Quat6_EulerAngles.ino#L225-L246) in the ICM-20948 library which you can use.
Best wishes,
Paul

PaulZC, 2021-06-01:
Hi @max-itsl / All,

I've added support for the ICM-20948 DMP in version 1.11 of the firmware. Please see [here](https://github.com/sparkfun/OpenLog_Artemis/blob/master/UPGRADE.md) for the upgrade instructions.

If you open Menu 3 (Configure IMU), you will see a new option (12) to enable the DMP. You then have the choice of logging: six-axis quaternions (Quat6); nine-axis quaternions (Quat9); raw accel, gyro and compass measurements.

Quat6 only uses measurements from the accel and gyro. You will need to select Quat9 to log the absolute orientation - with respect to geomagnetic North.

Please note that the DMP requires calibration, for Quat9 especially. You need to revolve the OLA slowly around all three axes, and hold it stationary for a few seconds in all six orientations, before the quaternion data becomes stable. I'm able to power up the OLA in a random orientation, perform the calibration, and have it output the exact same quaternions afterwards.

You can manually calculate the Euler angles (tip/tilt/yaw) from the Quaternions. Please see the new Quat6 stand-alone example for more details:

https://github.com/sparkfun/OpenLog_Artemis/blob/1859ea6b64e6adc7025ed0d3cee2c493006f55fc/Firmware/Test%20Sketches/IMU_DMP_Quat6/IMU_DMP_Quat6.ino#L199-L231

If you want to run the stand-alone examples, please make sure you have version 1.2.6+ of the ICM-20948 library installed. Version 1.2.5 contained a small but critical flaw. Please see the [release notes](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/releases/tag/v1.2.6) if you are interested.

I am closing this as I believe this issue is resolved, but please re-open if you identify any issues with the new firmware.
Best wishes,
Paul

closed 2021-06-01 by PaulZC

PaulZC, 2021-06-16:
Hi @Uygarkuzuoglu ,
Do you mean: how do you calibrate the compass when using the DMP? If so, the DMP runs a calibration procedure in the background, continuously. You need to slowly rotate the sensor around all three axes, and hold it still for a few seconds in all six orientations. After that, the compass data and Quat9 Quaternions will be valid. You don't need to call a function to start the calibration.
Best wishes,
Paul

Uygarkuzuoglu, 2021-06-16:
Thank you so much. How do I know if the calibration was successful ?

Uygarkuzuoglu, 2021-06-16:
[image]

PaulZC, 2021-06-16:
I can't tell what data is being shown in your screenshot. Please open the menu, press "h", wait a second or two, and then take another screenshot. The "h" tells the code to print the "helper" text, showing what each column of data is.

Uygarkuzuoglu, 2021-06-17:
[image]

Uygarkuzuoglu, 2021-06-17:
I want to calculate roll - pitch - yaw parameters of each camera .

[image]

PaulZC, 2021-06-17:
Hi @Uygarkuzuoglu ,
There are some very strange values in your DMP quaternion data. The DMP uses unit quaternions and so should give values in the range +/-1.0. I see values of 1.4 and 1.5 in your data. Those values must be errors of some kind. I also see some very large values in the magnetometer readings: -16000 and 20000.
Please make sure you are using version 1.11 of the OLA firmware.
Also, do you have the OLA close to any magnets or motors? Maybe ferrous metals? That would explain the strange magnetometer readings.
Best wishes,
Paul

Uygarkuzuoglu, 2021-06-17:
I am using 1.11 version. it is on the desk in office and power from c-type usb cable.

PaulZC, 2021-06-17:
Hi @Uygarkuzuoglu ,
OK. I see the same thing. I think it is due to the 1Hz sample rate:

[image]

If I increase to 10Hz, the data is clean:

[image]

The DMP is running at 55Hz, so I wonder if the 1Hz sampling is causing the FIFO to overflow. I will investigate this - as soon as I have I have time.
Best wishes,
Paul

PaulZC, 2021-06-17:
Hi @Uygarkuzuoglu ,
I have opened a new issue #87 to track this.
Thank you for identifying this issue,
Paul

Uygarkuzuoglu, 2021-06-17:
[dataLog00025.TXT](https://github.com/sparkfun/OpenLog_Artemis/files/6668649/dataLog00025.TXT)

55 Hz

Uygarkuzuoglu, 2021-06-17:
Sorry , i am not an expert on arduino . I uploaded IMU_DMP_Quat6/IMU_DMP_Quat6.ino to artemis board but ı get this message 

" Initialization of the sensor returned: Data Underflow
Trying again.. "

How to calculate Yaw - Pitch - Roll ?

[image]

PaulZC, 2021-06-17:
Hi @Uygarkuzuoglu ,
You can not run the examples from the ICM-20948 library directly on OpenLog Artemis. It is not a standard Arduino board.
There are separate examples for the OLA:
https://github.com/sparkfun/OpenLog_Artemis/tree/main/Firmware/Test%20Sketches
Please try IMU_DMP_Quat6.ino.
But you have some work to do:
Make sure you are using version 1.2.1 of the Apollo3 boards. Version 2.1.0 will not work.
Make sure you have RedBoard Artemis ATP selected as the board type.
Make sure you have modified the ICM-20948 library as described in the code comments.
Best wishes,
Paul

Uygarkuzuoglu, 2021-06-17:
I am using this board ;

[image]

My setup ;

[image]

Sorry ı dont understand " 1.2.1 version of the apollo3" ,  I am using OpenLog_Artemis-V10-v111.bin

Uygarkuzuoglu, 2021-06-17:
[image]

adamgarbo, 2021-06-17:
Hi @Uygarkuzuoglu, 

The Artemis Apollo3 Core has two different versions: 1.x and 2.x. Certain features do not work on v2.x, so make sure your Boards Manager looks like the following:

[image]
Also, did you make the necessary changes to the ICM-20948 library?

```
 * ** Important note: by default the DMP functionality is disabled in the library
 * ** as the DMP firmware takes up 14301 Bytes of program memory.
 * ** To use the DMP, you will need to:
 * ** Edit ICM_20948_C.h
 * ** Uncomment line 29: #define ICM_20948_USE_DMP
 * ** Save changes
 * ** If you are using Windows, you can find ICM_20948_C.h in:
 * ** Documents\Arduino\libraries\SparkFun_ICM-20948_ArduinoLibrary\src\util
 ```

Cheers,
Adam

Uygarkuzuoglu, 2021-06-17:
Hi @adamgarbo 

There is not it .

[image]

adamgarbo, 2021-06-17:
Hi @Uygarkuzuoglu,

That is the Library Manager. You need to open the **Boards Manager**.

Cheers,
Adam

Uygarkuzuoglu, 2021-06-17:
Thank you so much ! You are my heros today :)

[image]

Uygarkuzuoglu, 2021-06-17:
@adamgarbo How to write roll pitch yaw to SDCARD ? There is no log file in sdcard now .

adamgarbo, 2021-06-17:
Hi @Uygarkuzuoglu,

I'm afraid I don't have an answer for that one. Your best bet may be to ask on the SparkFun forums: https://forum.sparkfun.com/

This GitHub issue is not the best place for general project advice/troubleshooting.

Cheers,
Adam

Uygarkuzuoglu, 2021-06-17:
Ok. Thx.

ormapper, 2021-06-17:
I've been tracking this thread and was about to give this a try but I'm
confused as to the board orientation. I see the XYZ screen printed onto the
SD card side of the board, is that considered the "up" side?

Uygarkuzuoglu, 2021-06-17:
I think yes.

[image]