sparkfun/OpenLog_Artemis #48: Update RTC commands to match Apollo3 Core v2.x

adamgarbo, 2020-10-24:
A quick reminder that we'll need to update the OpenLog Artemis's RTC setTime() commands in the firmware to reflect the new (standardized) order of hundredths, seconds, minutes, etc. that came in with v2.0 of the Apollo3 Core.

https://github.com/sparkfun/Arduino_Apollo3/blob/933524d33d3616ce1211d6ee191d1fbb7d5209cf/libraries/RTC/src/RTC.h#L28

I'll try to submit a PR to release_candidate to address this. Was the v17 binary compiled with v1.x or v2.x of the Apollo3 Core?

Cheers,
Adam

PaulZC, 2020-10-25:
v1.x

adamgarbo, 2021-02-26:
I don't believe there's currently a need to keep this issue open. The OLA's migration to v2.x of the Apollo3 Core is most likely several months away. Happy to assist when the time comes to migrate. I suspect RTC time functionality will be the least of our worries. 😉

closed 2021-02-26 by adamgarbo