sparkfun/OpenLog_Artemis #49: Improve OLA time stamp resolution when exrernally triggered.

naltieri01, 2020-10-27:
### Subject of the issue
Please consider improving the time stamp resolution of the OLA when it is externally triggered at pin 11. Presently the best time stamp resolution from the RTC is 10ms, please can this be improved to obtain a micro second time resolution.

### Your workbench
* microSD card? FAT
* Using pin 11 to externally trigger on rising edge.
* Version 1.6  firmware are you using? 
* Uses SparkFun GPS Breakout - NEO-M9N, U.FL (Qwiic) for initial synchronization of OLA RTC, all on board sensors disabled.
* Powered by 1Ahr lipo battery.
* Are there any additional details that may help us help you?

### Steps to reproduce
Apply 2V rising edge logic signal to pin 11 of OLA

### Expected behavior
OLA logs date and time when pin 11 is triggered 

### Actual behavior
Time stamp reaolution recorded is limited to 10ms.

PaulZC, 2020-11-27:
Hi Nick (@naltieri01 ),
Thank you for your patience!
I have added a "logMicroseconds" option to the time stamp menu. The new code is here:
https://github.com/sparkfun/OpenLog_Artemis/commit/6ee2be76ca2e20a4e986a726cc35a450fd1c593e
Please note that "micros" will reset if the Artemis goes into deep sleep - I'm afraid there is nothing I can do about that. But you have at least got the microsecond time information you wanted.
I will leave this issue open until v1.8 is released - or there is a BETA version you can try.
Best wishes,
Paul

PaulZC, 2020-12-01:
Hi Nick (@naltieri01 ),
Firmware version 1.8 has just gone live. You can find update instructions here:
https://github.com/sparkfun/OpenLog_Artemis/blob/master/UPGRADE.md
And the new binary file is here:
https://github.com/sparkfun/OpenLog_Artemis/tree/master/Binaries
Please let me know if this fixes the issue. I've tested it here - but you never know!
All the best,
Paul