sparkfun/OpenLog_Artemis #50: Allow data collection rates faster than GPS

MattEttus, 2020-10-31:
### Subject of the issue
GPS modules like SAM-M8Q output data at 10Hz.  When logging from those, all other data collection is slowed down to the same rate, no matter what the setting is

### Your workbench
* Attempting to log at 50 Hz, but limited to 10Hz b/c of GPS
* Artemis OpenLog v1.17
* How is OpenLog Artemis wired to your sensor(s)?  Qwiic connector
* How is everything being powered?  Battery
* Are there any additional details that may help us help you?
From PaulZC in the forums -- "It would be possible to do this using the "autoPVT" function in the u-blox library. But it's not supported on the OLA at the moment..."

### Steps to reproduce
Use GPS and attempt to log IMU at 50 Hz

PaulZC, 2020-11-01:
Thanks Matt,

This _should_ be straight forward to implement using autoPVT.

Cheers,
Paul

PaulZC, 2020-11-27:
Hi Matt (@MattEttus ),

Thank you for your patience!

I have added the "autoPVT" feature. The code modifications are here:
https://github.com/sparkfun/OpenLog_Artemis/commit/44c319b6ea2743436dcac25bfaa7bef565254c69

In the image below, you can see that the ZED-F9P is updating every 100ms, but the OLA is successfully logging data at 25Hz (every 40ms):

[image]

You will find the new option in the "Configure Attached Devices" menu for the GNSS.
I will leave this issue open until v1.8 is released - or at least until there is a BETA version you can test.
This should resolve the TOW issue too.
Very best wishes,
Paul

PaulZC, 2020-12-01:
Hi Matt,
Firmware version 1.8 has just gone live. You can find update instructions here:
https://github.com/sparkfun/OpenLog_Artemis/blob/master/UPGRADE.md
And the new binary file is here:
https://github.com/sparkfun/OpenLog_Artemis/tree/master/Binaries
Please let me know if this fixes the issue. I've tested it here - but you never know!
All the best,
Paul

closed 2020-12-01 by PaulZC