sparkfun/OpenLog_Artemis #51: Logging dropouts

MattEttus, 2020-10-31:
### Subject of the issue
When logging IMU, 2 Analog voltages, RTC time, and GPS to an SD card, I get long logging dropouts, sometimes 3-4 seconds, and every now and then a measurement out of order

### Your workbench
* Are you using a microSD card? FAT32 16 GB
* At what frequency are you logging? 10Hz
* What version of firmware are you using? Artemis OpenLog v1.17
* How is OpenLog Artemis wired to your sensor(s)?  Qwiic
* How is everything being powered?  Battery
* Are there any additional details that may help us help you?
screenshot of log dropouts attached

[image]

MattEttus, 2020-11-01:
The time glitch is a separate issue, see #53.  The dropouts are still real.

PaulZC, 2020-11-01:
Hi Matt,
Please try using an empty (freshly formatted) SD card. Does that help?
Sometimes the SD card writing can stall if the card has a lot of files on it already.
Best wishes,
Paul

MattEttus, 2020-11-02:
Freshly formatted card with no other files on there doesn't help.   Slowing the logging down to 5Hz doesn't help, nor does using a brand new card from a different manufacturer.

PaulZC, 2020-11-02:
Hi Matt,
Please see my reply to #52 - thanks!
Paul

PaulZC, 2020-11-27:
Hi Matt (@MattEttus ),
I'm hopeful that the "autoPVT" feature will correct this issue for you. Let's see...
Cheers,
Paul

MattEttus, 2020-12-03:
v1.8 seems to have fixed the problem with GPS time going backwards on the turnover over the seconds.

I do still get issues with whole times being skipped.  For example, if logging rate is 50Hz you expect to get 5 samples of each GPS time, since it updates at 10 Hz.  Sometimes you get 4 and sometimes 6, but that is ok and expected.  But every now and then I'll get ten samples of time X in a row, and then the next samples will be of time X+200ms, thus skipping 100ms.  Not sure if this is the GPS's fault, doing some sort of adjustment or not.  In the example below, GPS time stays at 02:32:02.400 for 200ms (20 samples) and then jumps to 28,02:32:02.600 for 4 samples, and then continues on to xxx.700.

The good news is the time does at least match up with the iTOW and it doesn't ever go backwards.  Maybe it's best to just follow the RTC time for these short intervals and leave GPS time for long term syncing.

01/01/2000,00:36:46.40,22.95,-16.60,-1006.35,02:32:02.300,354740300,47.19,252,
01/01/2000,00:36:46.42,9.28,-10.25,-998.05,02:32:02.400,354740400,47.08,253,
01/01/2000,00:36:46.44,13.67,1.46,-1005.86,02:32:02.400,354740400,47.21,254,
01/01/2000,00:36:46.46,0.98,-13.67,-1003.91,02:32:02.400,354740400,47.21,255,
01/01/2000,00:36:46.48,13.67,9.77,-1008.30,02:32:02.400,354740400,47.23,256,
01/01/2000,00:36:46.50,23.44,-5.86,-1018.07,02:32:02.400,354740400,47.23,257,
01/01/2000,00:36:46.52,9.77,-2.44,-1000.98,02:32:02.400,354740400,47.25,258,
01/01/2000,00:36:46.54,0.98,6.84,-1010.25,02:32:02.400,354740400,47.13,259,
01/01/2000,00:36:46.56,13.18,2.44,-1006.84,02:32:02.400,354740400,47.27,260,
01/01/2000,00:36:46.58,21.48,-10.74,-1003.42,02:32:02.400,354740400,47.27,261,
01/01/2000,00:36:46.60,18.07,4.39,-1011.23,02:32:02.400,354740400,47.29,262,
01/01/2000,00:36:46.62,24.41,-10.74,-1009.28,02:32:02.600,354740600,47.17,263,
01/01/2000,00:36:46.64,14.65,-4.39,-1009.77,02:32:02.600,354740600,47.31,264,
01/01/2000,00:36:46.66,26.37,-3.91,-1013.18,02:32:02.600,354740600,47.31,265,
01/01/2000,00:36:46.68,13.67,-9.28,-996.58,02:32:02.600,354740600,47.33,266,
01/01/2000,00:36:46.72,22.95,6.84,-1018.55,02:32:02.700,354740700,47.11,267,

PaulZC, 2020-12-03:
Hi Matt,
Ah, yes. This is a feature of how autoPVT works. If an I2C PVT message gets corrupted or missed, then autoPVT will continue to give you the old data until new data arrives. We believe this is better than to have the function block while waiting for new (valid) data. The 200ms case just means that unfortunately two messages were missed or corrupted.
I'm going to close this issue as I believe we have resolved it for you, but please re-open if you need more help or see any more weirdness...
Best wishes,
Paul

closed 2020-12-03 by PaulZC