sparkfun/OpenLog_Artemis #52: OLA Deletes OLA_settings.txt file

MattEttus, 2020-11-01:
### Subject of the issue
OLA deletes its own settings file on occasion, and ignores it on others.  Possibly related to writing the file from a computer.

### Your workbench
* Are you using a microSD card? Hitachi 16GB FAT32
* At what frequency are you logging? 10Hz
* What version of firmware are you using? Artemis OpenLog v1.17
* How is OpenLog Artemis wired to your sensor(s)? N/A
* How is everything being powered? Battery
* Are there any additional details that may help us help you?

### Steps to reproduce
Put SD card in computer, confirm there is an OLA_settings.txt file there.  Edit.  Place back in OLA, let it run for a while, and then reread the card in the PC.

### Expected behavior
File should not be deleted

### Actual behavior
File is deleted

PaulZC, 2020-11-01:
Hi Matt,
Have you edited the settings file on your computer?
There are only three ways the settings file can be deleted:
From the main menu - using the "r" option.
Or when new settings are about to be written to the card.
Or if sizeOfSettings is set to -1.
Maybe you are trying to copy different settings? And the settings you are copying has sizeOfSettings set to -1?
Best wishes,
Paul

MattEttus, 2020-11-02:
Nope, did none of those things.  But now I have discovered that even without touching the SD card, slightly different settings can get used on each boot.  Sometime a particular item (gps_iTOW) is turned on and some times it isn't.

PaulZC, 2020-11-02:
Hi Matt,
This is really weird. I've not seen anything like this before.
Two suggestions:
Can you post your settings file and I'll see if I can replicate the issue.
Also, can you try enabling debug messages. There is a hidden debug menu - enter "d" at the main menu. Do the debug messages provide any clues?
Many thanks,
Paul

PaulZC, 2020-11-27:
Hi Matt (@MattEttus ),
Are you still seeing this issue? I can't duplicate it here and really have no idea what could be causing it. You are - to the best of my knowledge - the only user who has seen this issue...
Are you by any chance removing the SD card while the OLA is still powered on? Maybe that is corrupting the settings file?
If you open the main menu and select the "q" option, that will put the OLA into a safe state where the log and settings files are closed and the SD power is disabled. It is then safe to remove the SD card.
Thanks,
Paul

MattEttus, 2020-12-03:
I haven't seen this happen again since then, but haven't been using it that much.  I'm ok with closing.

PaulZC, 2020-12-03:
Thank you Matt,
Paul

closed 2020-12-03 by PaulZC