sparkfun/OpenLog_Artemis #53: GPS time does not match GPS iTOW

MattEttus, 2020-11-01:
### Subject of the issue
When using a SAM-M8Q, the GPS time does not match the iTOW.  While gps_iTOW proceeds monotonically, gps_TIME glitches backwards by a second every time the integer second rolls over.  This is likely due to some kind of rounding error.  See plot below

### Your workbench
* Are you using a microSD card?  16GB Hitachi FAT32
* At what frequency are you logging? 5 Hz
* What version of firmware are you using? Artemis OpenLog v1.17
* How is OpenLog Artemis wired to your sensor(s)? Qwiic
* How is everything being powered? Battery
* Are there any additional details that may help us help you?

### Steps to reproduce
Log all GPS values and rtcTime and some others at a speed of at least 5 Hz (maybe happens at lower speeds too?)

### Expected behavior
gps_Time should be monotonic increasing unless Sparkfun is selling Flux Capacitors

### Actual behavior
gps_Time Glitches backwards by 1 second at every integer second
[image]

PaulZC, 2020-11-01:
Hi Matt,
In the "collect sensor data" code we collect the seconds then the milliseconds:
https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/OpenLog_Artemis/Sensors.ino#L268
I suspect that on occasion, exactly at the second interval, we are getting the 'old' second followed by the new 'millisecond'.
iTOW is a single 32-bit number representing time in milliseconds, so it is impossible for the same to happen there.
I will investigate this but it might be tricky to fix. Or not without stalling the data collection...
Cheers,
Paul

PaulZC, 2020-11-27:
Hi Matt,
The "autoPVT" feature has corrected this issue. I see no discontinuities in the UTC data (vs. TOW).
I'll leave this issue open until v1.8 is released - or there is a BETA version you can try.
Best wishes,
Paul

PaulZC, 2020-12-01:
Hi Matt,
Firmware version 1.8 has just gone live. You can find update instructions here:
https://github.com/sparkfun/OpenLog_Artemis/blob/master/UPGRADE.md
And the new binary file is here:
https://github.com/sparkfun/OpenLog_Artemis/tree/master/Binaries
Please let me know if this fixes the issue. I've tested it here - but you never know!
All the best,
Paul

closed 2020-12-01 by PaulZC