sparkfun/OpenLog_Artemis #58: Custom firmware uploaded to OLA Board doesn't run

CodeStrumpet, 2020-12-15:
When I upload firmware compiled with the Artemis RedBoard ATP board selected in Arduino the firmware seems to upload successfully but the code does not seem to run. 

Is there something specific that needs to be done when compiling from The Arduino IDE?  Or should I be using the MBED IDE when compiling custom firmware for the OLA board? 

Any ideas would be most appreciated!

Cross-post here:  https://forum.sparkfun.com/viewtopic.php?f=169&t=54557

nseidle, 2020-12-15:
ATP is the correct board to use but OLA doesn't yet support v2 of the Artemis Arduino core. Please use v1.2.1.

closed 2020-12-15 by nseidle

CodeStrumpet, 2020-12-16:
Thank you. I just installed v1.2.1 through the Arduino Boards manager and am now able to upload my new firmware.

For anyone else who might end up here, I had to blow out my ~/Library/Arduino15 directory on Mac and then re-install the Apollo library at 1.2.1 to get it to compile (after downgrading from version 2.x.x I was getting errors about the board not being found). On Windows it worked right after downgrading to the lower version through the boards manager.

nseidle, 2020-12-16:
Happy to help and glad to hear you got it working. Also, thanks for reporting back 'gotchas', that helps a ton.

CodeStrumpet, 2020-12-21:
@nseidle -- when using v1.2.1 of the Artemis Arduino core are there any workarounds to be able to use BLE? 

I have read many forum posts that mention v2.0.0+ is required for BLE so it seems like a longshot, but given that the OLA board isn't yet supported in the newer versions of the core I figured I would ask.

Alternatively, is there a newer development version of the Artemis Arduino Core I could use that supports the OLA board? If not, any idea you could give about the development timeline on that front would be super helpful.

nseidle, 2020-12-21:
We are waiting for v2 to mature a bit before we begin to port OLA over.

CodeStrumpet, 2020-12-22:
Okay, thanks for the clarification!

decamun, 2023-01-20:
IS OLA still compiled on v1.2.1, or is it now compiled on v2?

PaulZC, 2023-01-20:
Hello Devin (@decamun ),

The OLA code now uses Apollo3 v2.2.1 (the latest) and RedBoard Artemis ATP as the board type.

Please see the following for more info:

https://github.com/sparkfun/OpenLog_Artemis/issues/149#issuecomment-1386830863

https://github.com/sparkfun/OpenLog_Artemis/issues/149#issuecomment-1386724466

Best wishes,
Paul