sparkfun/OpenLog_Artemis #59: Feature Request: Allow disabling of voltage regulator when entering deep sleep.

ryanneve, 2020-12-17:
### Request

There seem to be provisions to allow the OLA to turn the AP2112K regulator on and off. There's a split pad between D25 and the regulator's enable and pin 25 is defined as PIN_VREG_ENABLE.
It might be nice to have the option of disabling the 3v3 output when the OLA is in Deep Sleep. This would mean being powered by the RTC battery

### Use case
My sampling rate is very low (every few minutes), but I want wireless access to the data, so I've got the OLA powering a BlueSMiRF module through the AP211K via the 3v3 pin. This will draw a lot of power, even when the OLA is in deep sleep if I can't turn it off.

### Issues
How long will the RTC battery last when the OLA is in Deep Sleep? Looks like 14 uA from a 1mAhr battery.
How quickly does the RTC battery re-charge?
What happens if the RTC battery dies while the regulator is disabled? 

In my case, I need some time when powering up to re-establish a Bluetooth connection, but I think I can do this by setting the Qwiic bus power up delay to a high enough number.

PaulZC, 2020-12-18:
Hi Ryan,

Here's the dilemma:

The RTC battery can only provide very small amounts of current, in the <100uA range. It's enough to keep the Artemis' RTC running while the Artemis is in deep sleep but it isn't enough to power the Artemis even with the Artemis running from its 32kHz clock (instead of the usual MHz clock source). The RTC battery voltage dips and the Artemis goes into brown out or a full power on reset.

We tried implementing a low battery with automatic restore function on OLA (and it didn't work well):
We wanted the OLA to be able to detect when the power had been removed (easy - using either the battery voltage divider or the low voltage supervisor) and go into deep sleep.
We then wanted the OLA to wake up periodically, check if power had been restored, and if it had then wake itself up properly.
The issue we had is that even partially waking the Artemis (bringing it out of deep sleep using an RTC or Watchdog interrupt, and have it run from 32kHz to keep the power draw low) drew too much current from the RTC battery - causing the Artemis to brown-out or do a full power-on reset.
We couldn't successfully have the Artemis wake up partially, check even one IO pin, and go back into deep sleep if power had not been restored. We tried - I spent a good couple of days looking into this - but we had to give up.
The only thing that works completely reliably is to have the Artemis go into deep sleep as soon as the power fails, and have the user wake it up again with a reset when power has been restored.

So, the feature you're asking for will suffer from the same issue. When the Artemis wakes up (via an RTC interrupt), even if the very first thing it does is re-enable the regulator it will still cause a brown-out or full reset.

You're welcome to try! You will learn **a lot** by being able to compile and modify the OLA code for yourself and try out new things like this. If you get it working, we'll happily merge it into the OLA firmware.

I'm going to close this issue as I don't believe we can help you, but please re-open if you are willing to modify the code for yourself and need a few pointers.

Very best wishes,
Paul

closed 2020-12-18 by PaulZC

adamgarbo, 2020-12-18:
@ryanneve I would definitely agree with Paul that you will learn a huge amount by modifying the OLA code for yourself. I've done a considerable amount of work creating my own OLA firmware versions, as well as porting code to the MicroMod Data Logging Carrier Board (MMDLC). It was quite the learning experience!

Speaking of the MMDLC, you may wish to explore this board for your use case. I don't know what sensors you have connected to the OLA, but the MMDLC has two user-controllable AP2112 regulators: one to control power to the microSD card and any connected peripherals, and one to control power to the Qwiic bus. It may provide the added flexibility you require for managing power to peripherals connected to your project.

Cheers,
Adam

ryanneve, 2020-12-18:
Thanks Paul & Adam for the update and background. The MMDLC looks very interesting. The only sensor we have is the Qwiic MicroPressure, but Bluetooth is also part of our project for streaming data and wireless data downloads. My primary constraint is time, so I don't know if I would be able to jump to a new platform for this project.

I wonder if a second [bigger RTC battery](https://www.digikey.com/en/products/detail/seiko-instruments/MS920SE-FL27E/1889209) would help with the voltage dips? I couldn't find any packages that would be good for production, but it wouldn't be hard to hack in a bigger battery and try it. There's even a TP1 pad to use.

In my specific case, the reason I want to turn off the voltage regulator is to power down the external Bluetooth module. I can probably put the BlueSMiRF to sleep and wake it up with commands on the serial line or console although this would be very specific to my application. Any idea if BT on the Artemis will be implemented soon?