sparkfun/OpenLog_Artemis #60: Feature request: Option to set pin 32 wake up timer using RTC alarm

ryanneve, 2020-12-17:
### Subject of the issue
It's pretty easy to pull pin 32 low to put the OLA in Deep Sleep mode, but waking up the OLA is harder. you probably don't want to hold RST low. If you want to sample based on an environmental switch, it would be nice if there was a way to exit deep sleep without resetting. 
This could be done by optionally setting a RTC wake alarm when pin 32 is pulled low. Just like when sleeping for long sampling rates, but for a longer period.
When the pin32 alarm wakes the OLA back up, if pin 32 is still low, it will go back to sleep again. Will interrupt be generated if it's low the entire time or would pin value need to be checked on startup?

### Other considerations.
You'd probably want to clear the RTC alarm if the OLA wakes up early due to the RST button.

### Example use case
If someone only wanted to sample when it was light out, they could hook a photo-switch up to pin 32 and have the pin32 sleep timeout set to 10 minutes. The OLA would sleep most of the night, waking every 10 minutes to see if pin 32 was still low. Come morning, it would stay awake, sampling at whatever rate desired.

PaulZC, 2020-12-18:
Hi Ryan,

To summarize, you would like the OLA to stay in deep sleep while pin 32 is held low, and "wake" again and start logging when pin 32 is high?

I would encourage you to have a go at implementing this yourself. You will learn a **huge** amount by being able to modify the code yourself. When you get it working successfully, send us a Pull Request and we will very happily merge it - thank you.

I'm going to leave this issue open. It's a reasonable request. There will be some work to do to make sure the new feature works cleanly with the existing Pin 32 code ("Stop Logging" and "Analog Read"), but not an excessive amount. We don't have time to look at this right now though, so the quickest way forward would certainly be to have a go at implementing the code changes yourself.

Very best wishes,
Paul

ryanneve, 2020-12-18:
Happy to have a go at it. It's been a while since I did this sort of thing, but I'll try. 

I don't think pin 32 would have to be held low, but it could. I'm not 100% clear on interrupts, but it seems to me if pin 32 is held low the entire time, when the OLA wakes and re-attaches the interrupt in **wakeFromSleep()**, it wouldn't generate another interrupt to run **stopLoggingISR()** if the pin was already low. Perhaps if **wakeFromSleep()** checked the status of PIN_STOP_LOGGING to determine how **stopLoggingSeen**  should be set initially.

ryanneve, 2021-01-04:
@PaulZC I've forked the repository, and made the changes. It seems to work as intended. I've never done a pull request before, so apologies if I don't do it correctly.

PaulZC, 2021-01-30:
Hi @ryanneve ,
Thanks again for submitting your PR! Please see #64 for full details.
I’ll leave this issue open until v1.9 goes live.
Best wishes,
Paul

PaulZC, 2021-02-03:
Resolved in [Version 1.9](https://github.com/sparkfun/OpenLog_Artemis/releases/tag/v1.9). Closing...

closed 2021-02-03 by PaulZC