sparkfun/OpenLog_Artemis #61: Feature request: Add support for SparkFun Triple Axis Accelerometer Breakout - MMA8452Q (Qwiic)

drewzilla37, 2020-12-19:
### Subject of the issue
I would like to be able to log multiple accelerometers.
Use case: putting them on items that are coupled together to be able to assess vibration and natural frequencies.

### Your workbench
1GB micro SD (FAT32)
10 Hz
1.8
wired to Mux abd 2 sensors via qwiic cables
usb-c through power bank

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.

### Expected behavior
Get accelerometer data for as a column in the .txt that is generated

### Actual behavior
sensors are detected but not recognised. No data is coming through.

drewzilla37, 2020-12-21:
I've found a workaround for now although not ideal.
I'm using an Arduino Nano with the Mux board and the two MMA8452Q and printing the data to serial in a csv format. I have the TX pin of the Nano feeding the RX pin of the OLA which is recording it.
I'm encountering an issue occasionally where only a few lines are being recorded in the serialLog file then it just stops.

PaulZC, 2021-01-29:
Hi @drewzilla37 ,
Apologies for the delay...
How is your project going? Do you still want us to add the MMA8452Q as a supported device on the OLA?
Best wishes,
Paul

drewzilla37, 2021-01-29:
Project is on hold still having the intermittent issues with serial log just stopping.
Yes, can you please add support for it?

PaulZC, 2023-01-14:
Hi @drewzilla37 ,

Apologies for the _very_ slow response on this. It's 2 years since you opened this issue! Sorry about that.

I've added support for the KX134 high-g accelerometer in v2.5 of the firmware. It isn't the MMA8452Q, but I hope will still meet your needs.

Very best wishes,
Paul

closed 2023-01-14 by PaulZC

drewzilla37, 2023-01-14:
Thank you Paul! Happy new year!