sparkfun/OpenLog_Artemis #63: Serial Logging Timestamps

DennisMelamed, 2021-01-04:
### Subject of the issue
Is it possible to add an option to timestamp entries to the serial logfile? I am recording data coming from a separate microcontroller over serial, but would like the this data to be timestamped to align to the IMU data being collected by the OLA.

### Your workbench
* Are you using a microSD card? If so, what size? How is it formatted (FAT, FAT32, or exFat)?
FAT32, 32GB card
* At what frequency are you logging? For example, 10Hz, 1Hz, once every 5 seconds, etc.
IMU logging at ~100 Hz, serial data coming in ~ 1 Hz
* What version of firmware are you using? This is shown at boot, for example: "Artemis OpenLog v1.10"
V1.8
* How is OpenLog Artemis wired to your sensor(s)?
Onboard IMU, serial line from peripheral microcontroller UART TX pin to OLA RX pin
* How is everything being powered?
OLA powered from usb-c, peripheral microcontroller via 3.3V pin on OLA
* Are there any additional details that may help us help you?
N/a

### Steps to reproduce
Wire other device to provide serial data to OLA RX pin once per second. Viewing serial log file, there is no timestamp for each reading.

### Expected behavior
Each line of serialLog****.TXT should have a timestamp attached to it that allows when the serial data was received to be aligned to when the IMU is sampled

### Actual behavior
Serial log file contains no timestamps.

PaulZC, 2021-01-05:
Hi Dennis (@DennisMelamed ),
This would be quite a big change to the code. The code would need some kind of 'idle timer' which sets the delay before a new timestamp is generated. And the feature would need to be switchable, as some (most?) users will expect the log file to _only_ contain the serial data.
I will leave this issue open - I think you might be the second person to ask for this feature - but I won't have time to implement it soon. If you have time to modify the code yourself, please go ahead and send us a Pull Request.
Best wishes
Paul

ryanneve, 2021-01-05:
@DennisMelamed & @PaulZC,
Would it be safe to assume that incoming serial data has line breaks, and you could just parse incomingBuffer and insert a RTC timestamp whenever a line break is seen? This would probably work best with serial data which is being generated slowly. Many of the serial output instruments I've worked with won't go faster than 2 Hz.

DennisMelamed, 2021-01-05:
@ryanneve & @PaulZC ,
I was thinking along similar lines, perhaps with a configurable "start of measurement" token. I'll try and put together a PR, I have some experimental code that seems to work for this.

PaulZC, 2021-01-28:
I do like the "start/end of measurement" token idea. I'll have a go at including this and make it default to the NewLine character. I'll work on this over the next few days, building on your PR Dennis, and will include it in release 1.9 (due out next week).
Thanks!
Paul

PaulZC, 2021-01-29:
Thanks again @DennisMelamed for this suggestion and Pull Request.
I've implemented your PR [here](https://github.com/sparkfun/OpenLog_Artemis/commit/c7c3ad716d0034631669bcc0032be5ac9fcafdf0). I had to change it a bit: I added the settings to the settings file; added the timestamp token; and changed the way the timestamp is read back out of serialTimestamp. But it is still essentially your code that is doing the work.
Closing...

DennisMelamed, 2021-01-29:
Glad it was helpful! Having the string just get smaller instead of a boolean flag makes much more sense. Thanks for all of your work on this project!

PaulZC, 2021-01-29:
The [v1.9 BETA binary](https://github.com/sparkfun/OpenLog_Artemis/tree/master/Binaries) now includes the changes too. I'd be grateful if you can give it a try and let me know if you find any gremlins!
Very best wishes,
Paul

PaulZC, 2021-02-03:
Resolved in [Version 1.9](https://github.com/sparkfun/OpenLog_Artemis/releases/tag/v1.9). Closing...

closed 2021-02-03 by PaulZC

DennisMelamed, 2021-02-03:
Hi @PaulZC  - Sorry for the delay, work took over. I'm actually seeing some issues w/ the new firmware. I was unable to compile it due to a missing library `SparkFun_SDP3x_Arduino_Library.h` which I don't seem to be able to find anywhere, but that's probably an issue on my end. I tried uploading the firmware blob for v1.9 through the GUI on Mac OS, but it seems to fail to initialize the SD card and the onboard IMU:

```
10:47:12.657 -> SD init failed (second attempt). Is card present? Formatted?
10:47:12.657 -> Config file read failed: SD offline
10:47:12.657 -> Artemis OpenLog v1.9
10:47:12.799 -> ICM-20948 failed to init.
10:47:12.799 -> SD card offline
10:47:12.799 -> Datalogging offline
10:47:12.799 -> Serial logging offline
10:47:12.799 -> IMU offline
10:47:12.799 -> Identifying Qwiic Muxes...
10:47:12.799 -> Identifying Qwiic Devices...
10:47:12.833 -> No Qwiic devices detected
10:47:12.833 -> rtcDate,rtcTime,output_Hz,
```

This might again be a mis-config on my end, but everything seemed to work immediately upon upload with the previous versions so I'm wondering what changed.

PaulZC, 2021-02-03:
Hi @DennisMelamed ,
Oops. Yes. Sorry about that. I've made the SDP3X library public:
https://github.com/sparkfun/SparkFun_SDP3x_Arduino_Library
It won't be available through the Library Manager just yet, you'll need to manually download it for now.
You might have to use option "r" to reset the settings file to the default once you have the new firmware installed?
Also please check you are using the V10 version (not X04) - assuming you have the red version of OLA (not the black SparkX version)?
Please let me know how it goes.
Best wishes,
Paul

DennisMelamed, 2021-02-03:
Hi @PaulZC :
Thanks! Things seem to be working! 

A purely cosmetic thing: when the character which triggers a timestamp is a line feed, the way the code is set up (line 462 in `OpenLog_Artemis.ino`) causes a blank line to show up. I.e. both the new line sent over serial and the newline added by the OLA are included in the file. Not a big deal, easy enough to strip blank lines from a file. 

Possible solution might be to set the `incomingBufferSpot` pointer back one character so the character triggering the timestamp is overwritten by the newline, but some use cases could rely on that character being present. 

Thanks again for all of your work on this!
Dennis

PaulZC, 2021-02-03:
Thank you Dennis,
Paul