sparkfun/OpenLog_Artemis #66: BLE enabled on the current hardware (OLA v1.18)

u382514, 2021-01-16:
Just got a hold of this little gem of technology and was wondering if, at some point, the firmware can/will be updated to open up BLE and that the current hardware will be able to handle it or if it's going to require a new buy. 

Thanks!

u382514, 2021-01-17:
Just following up: I was able to get BLE enabled by upgrading the core to 2.0.3 and the hardware indeed accepted it. Just getting a basic example compiled using the ICM 20948 library was quite an effort though. You folks will have your work cut out for you porting OLA.

closed 2021-01-17 by u382514

PaulZC, 2021-01-18:
Hi Brandon (@u382514 ),
That's great news - glad that is working for you,
Paul

userqt, 2023-01-20:
@u382514 , I am curious how did you manage to make the BLE working on the OLA? I am not familiar with how can the core be updated. Would it be possible if you can share some info on this?

u382514, 2023-01-21:
Ooof, that was so long ago. It involved multiple bottles of scotch and a bruised and battered ego. What I remember is dealing with a lot of compile errors and just going one by one until I got it to work with ArduinoBLE. Then I slowly wrote the OLA code back in to the bare minimum requirements I needed which was the onboard ICM20948 and the SD Card. I ended up writing my own implementation of ICM20948 for the DMP and probably some other valid reason that escapes me.  

Seems like they've long since upgraded to a core that can handle BLE but don't seem to see any progress on implementing BLE yet in the firmware? Adding ArduinoBLE now should be fairly trivial I would think.  

I may be able to go in some time this weekend and do a quick upgrade to one the OLA boards I have lying around and report back. Such a great board I bought way more than I needed.

u382514, 2023-01-21:
I was able to get this to work with no problems. A whole lot of very nice updates have occurred to this repo since 2021. 

1.  First off, I downloaded a fresh version of Arduino IDE
2. Next, I had to install SparkFun Apollo3 Boards (version 2.2.1). To do that I needed to add a url for the board to show up for install. Go to settings and add the following URL to the "Additional boards manager URLs": https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/main/package_sparkfun_apollo3_index.json
3. Go to Tools -> Boards -> Board Manager and filter on Apollo3. Install SparkFun Apollo3 Boards version 2.2.1.
4. I plugged my OpenLog board into the usb-c port, and selected the port. For the board I used RedBoard Artemis ATP
5. I then downloaded this whole repository and opened up the OLA_IMU_Basics sketch under Firmware -> Test Sketches
6. I made sure this compiled and worked on the OpenLog board. Worked like a charm
7. I then made the changes  to the OLA_IMU_Basics sketch to enable BLE and send over a Temp via BLE (full sketch below). Compiled and worked like a charm 
8. To test I used an app called BluetoothLE on my iPhone

I hope this helps. 

```cpp
/****************************************************************
 * Testing basic functionality of the ICM 20948 on the OpenLog Artemis.
 * Select "SparkFun Apollo3" \ SparkFun RedBoard Artemis ATP" as the board.
 * 
 * Based on:
 * 
 * Example1_Basics.ino
 * ICM 20948 Arduino Library Demo 
 * Use the default configuration to stream 9-axis IMU data
 * Owen Lyke @ SparkFun Electronics
 * Original Creation Date: April 17 2019
 * 
 * This code is beerware; if you see me (or any other SparkFun employee) at the
 * local, and you've found our code helpful, please buy us a round!
 * 
 * Distributed as-is; no warranty is given.
 ***************************************************************/

// OLA Specifics:
const byte PIN_IMU_POWER = 27; // The Red SparkFun version of the OLA (V10) uses pin 27
//const byte PIN_IMU_POWER = 22; // The Black SparkX version of the OLA (X04) uses pin 22
const byte PIN_IMU_INT = 37;
const byte PIN_IMU_CHIP_SELECT = 44;
const byte PIN_SPI_SCK = 5;
const byte PIN_SPI_CIPO = 6;
const byte PIN_SPI_COPI = 7;

#include <ArduinoBLE.h>
#include "ICM_20948.h"  // Click here to get the library: http://librarymanager/All#SparkFun_ICM_20948_IMU

#define USE_SPI       // Uncomment this to use SPI

#define SERIAL_PORT Serial

#define SPI_PORT SPI    // Your desired SPI port.       Used only when "USE_SPI" is defined
#define CS_PIN PIN_IMU_CHIP_SELECT // Which pin you connect CS to. Used only when "USE_SPI" is defined. OLA uses pin 44.

#define WIRE_PORT Wire  // Your desired Wire port.      Used when "USE_SPI" is not defined
#define AD0_VAL   1     // The value of the last bit of the I2C address. 
                        // On the SparkFun 9DoF IMU breakout the default is 1, and when 
                        // the ADR jumper is closed the value becomes 0

#ifdef USE_SPI
  ICM_20948_SPI myICM;  // If using SPI create an ICM_20948_SPI object
#else
  ICM_20948_I2C myICM;  // Otherwise create an ICM_20948_I2C object
#endif

//U382514 - Added Service and Characteristics
BLEService myBLEService("19B10000-E8F2-537E-4F6C-D104768A1214");
BLEFloatCharacteristic myCharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite | BLENotify);

void setup() {

  //U382514 - Add BLE Setup
  if (!BLE.begin()) {
[... 204 lines cut ...]
#if defined(ARDUINO_ARCH_MBED) // updated for v2.1.0 of the Apollo3 core
bool enableCIPOpullUp()
{
  //Add 1K5 pull-up on CIPO
  am_hal_gpio_pincfg_t cipoPinCfg = g_AM_BSP_GPIO_IOM0_MISO;
  cipoPinCfg.ePullup = AM_HAL_GPIO_PIN_PULLUP_1_5K;
  pin_config(PinName(PIN_SPI_CIPO), cipoPinCfg);
  return (true);
}
#else
bool enableCIPOpullUp()
{
  //Add CIPO pull-up
  ap3_err_t retval = AP3_OK;
  am_hal_gpio_pincfg_t cipoPinCfg = AP3_GPIO_DEFAULT_PINCFG;
  cipoPinCfg.uFuncSel = AM_HAL_PIN_6_M0MISO;
  cipoPinCfg.ePullup = AM_HAL_GPIO_PIN_PULLUP_1_5K;
  cipoPinCfg.eDriveStrength = AM_HAL_GPIO_PIN_DRIVESTRENGTH_12MA;
  cipoPinCfg.eGPOutcfg = AM_HAL_GPIO_PIN_OUTCFG_PUSHPULL;
  cipoPinCfg.uIOMnum = AP3_SPI_IOM;
  padMode(MISO, cipoPinCfg, &retval);
  return (retval == AP3_OK);
}
#endif
```

userqt, 2023-01-21:
This is amazing, thank you so much. It saved me a lot of time. I was able to compile and upload the sample you provided, it worked right away. I used an android app called nRF Connect. I was able to see it listed as MYBLESERVICE and was able to connect to it - awesome!!!

If I may ask you:

How do I make sure that I received the correct data? As an experiment I did this: `myCharacteristic.writeValue(1234567890);` In the app, I see that it received the following: (0x) 06-2C-93-4E. Is this binary displayed as hex? How do I decode this? 

The other thing I am not sure is how do I get this data to my application? At the moment, I am using the USB and listening to the serial port: "COM3", baud 115200. Now with BLE, I am not sure how to read data from it.

What I would ideally like to do is send text just like the IMU prints stuff to the console in the arduino IDE, but writeValue is able to write only float.

Again thanks a lot and enjoy your day.

u382514, 2023-01-22:
You're receiving back a byte array. So you'll have to look at whatever language/framework you're working in to determine how to convert that byte array back into the data it represents. 

For instance, if I was building a Flutter app (Dart), using the Float Characteristic I'm returning for the temp I might convert it like this: 

```dart
final bytes = Uint8List.fromList(byteArrayDataFromBLECharacteristic);
final bytesData = ByteData.sublistView(bytes);
double celcius = bytesData.getFloat32(0, Endian.little);
double farenheit = (celcius * (9 / 5)) + 32;
```

In the example I gave I'm using a Float Characteristic specifically, but there are other incarnations you can use. You'll have to go through the ArduinoBLE documents to see what they all do. If I wanted to send a string through, I would change the Arduino Sketch to probably use a generic BLECharacteristic like below.

```cpp
//BLEFloatCharacteristic myCharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite | BLENotify);
BLECharacteristic myCharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite | BLENotify, 12); 

//... code skipped for brevity

//myCharacteristic.writeValue(myICM.temp());
char buf[12];
String helloWorld = "Hello World";
helloWorld.toCharArray(buf, 12);
myCharacteristic.writeValue(buf);
```

and then in my Flutter app I would convert it like so...

```dart 
final helloWorld = utf8.decode(byteArrayDataFromBLECharacteristic);
```

userqt, 2023-01-22:
I am trying to send three floats to my application (I am using C#). In arduino I used:
`char res[40];
sprintf(res, "%.1f,%.1f,%.1f", 1.2, 5.6, 2.5);
myCharacteristic.writeValue(res);`

And then in C#:
`Encoding.UTF8..GetString(bytes)`

This however, returns me `%.1f,%.1f,%.`

How can I send those floats correctly?

userqt, 2023-01-22:
Ok, nvm, I was able to make it work. The byte array was not complete, coz I had a size limit less than I needed. Seems I forgot to increase the size of the characteristic during my experiments.

Thank you for the help @u382514.

u382514, 2023-01-23:
Awesome! Glad to be of service. Check out ArduinoJson. I've had pretty good luck getting data in a structured format using this package to use in apps to cut down on coding time. This makes even more of a difference when SENDING data to the device via BLE. This package magically handles encoding just about anything needed for a Json format (including floated) without having to deal with other work-a-arounds. Obviously everything comes at a cost and BLE can be quite slow when it comes to real-time updates. Pros and cons for convenience.

userqt, 2023-01-23:
@u382514 , also very good suggestion, I will take a look into the ArduinoJson, would defintely need it. 

I am experiencing some lags and I am not sure what is the cause of that. I see that there is a delay(500). If I disable it then the board turns off after connection.

Is there some specific way to handle rapid flow of data from the board to my application? The data (I am sending quaternion transformations) should always be available to the characteristic and the app should always just read it. I could add an optmization to send the data if there is change for example, but I am afraid that won't be enough. It is not clear to me if the loop method is the place for writing values or I have to listen for events with a handler in my application? I am quite new to all this. I need the board to operate as a controller operated by the user in a constant manner and send the data produced during the user input.

Do you perhaps know or have some directions how this can be done?

u382514, 2023-01-23:
These devices have a lot of noise by nature, so it's very common to add filters (like the Kalman Filter) to smooth out the data. Sometimes it's better to do it directly on the board and sometimes it's better to do that in your application. I usually do it in the application because I store the real-time data on the SD Card for offline analytics (more granular and accurate). 

I use BLE for real-time UI changes like graphs, tiles, etc at relatively constant intervals. Trying to use the BLE information of an IMU in a 3d viewer for instance may just not work so well with the BLE limitations. BLE just isn't going to get that kind of throughput/accuracy that direct to SD/offline processing is going to give you. You can buffer data in the loop and at set intervals flush through BLE. I'm assuming you're using stream/subscribing to the characteristic in your application, so what you see is pretty much what you get. 

If you need more throughput you're going to have to pair the board up with a wifi board or something like that.

userqt, 2023-01-23:
I figured it out, it was a setting in my application plugin, which I had to enable, in order to make the communication between the two more frequent.