sparkfun/OpenLog_Artemis #68: built in ICM init fails

better-adapted, 2021-01-18:
### Subject of the issue
I need to modify the OpenLog code - but when I build it the ICM init() function fails "ICM-20948 failed to init.=4"

### Your workbench
Sparkfun Apollo core 1.2.1

### Steps to reproduce
downloaded the needed libs, all normal files except:
**zmodem_crc16.cpp**
_//#include <avr/pgmspace.h>

#ifndef PROGMEM 
#define PROGMEM 
#endif_

### Expected behavior
normal demo should run with the ICM unit

### Actual behavior
ICM-20948 failed to init.=4

adamgarbo, 2021-01-18:
I'd first try using the SparkFun Redboard Artemis ATP board definition when using "Export compiled Binary". This is what was used to compile the current firmware binaries.

Cheers,
Adam

better-adapted, 2021-01-18:
Hello Adam ( i am called adam too lol)
yes i did that as i cehcked the top of the code, but same issue, adding debug to the ICM_20948::startupDefault() now

PaulZC, 2021-01-27:
Hi @better-adapted ,
Apologies for the delay.
Have you tried the standard OpenLog Artemis code (using the [pre-compiled binary](https://github.com/sparkfun/OpenLog_Artemis/tree/master/Binaries))? Does your ICM work OK with that?
Best wishes,
Paul

better-adapted, 2021-01-27:
yes strangely it works with the pre-compiled

I do get errors with Arduino build system - around the zmodem referencing "PROGMEM", MISO and other areas

Are you able to build the fimware? i have all the repo's referenced pulled down and tried building against different cores and modules but still no joy

PaulZC, 2021-01-29:
Hi @better-adapted ,
I've just re-tested the [OLA_IMU_Basics example](https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/Test%20Sketches/OLA_IMU_Basics/OLA_IMU_Basics.ino) and it runs OK. Can you please try that? If it runs OK for you, then you know the issue is in your code...
I'm going to close this - but please re-open if you are still having problems.
Best wishes,
Paul

closed 2021-01-29 by PaulZC

Hamdi8, 2021-05-19:
I get this error in the OLA_IMU_BASIC code . 

**OLA_IMU_Basics:221:37: error: 'AP3_GPIO_DEFAULT_PINCFG' was not declared in this scope**

bool enableCIPOpullUp()
{
  //Add CIPO pull-up
  ap3_err_t retval = AP3_OK;
  am_hal_gpio_pincfg_t cipoPinCfg = AP3_GPIO_DEFAULT_PINCFG;
  cipoPinCfg.uFuncSel = AM_HAL_PIN_6_M0MISO;
  cipoPinCfg.eDriveStrength = AM_HAL_GPIO_PIN_DRIVESTRENGTH_12MA;
  cipoPinCfg.eGPOutcfg = AM_HAL_GPIO_PIN_OUTCFG_PUSHPULL;
  cipoPinCfg.uIOMnum =AP3_SPI_IOM ;
  cipoPinCfg.ePullup = AM_HAL_GPIO_PIN_PULLUP_1_5K;
  padMode(MISO, cipoPinCfg, &retval);
  return (retval == AP3_OK);
}

Hamdi8, 2021-05-19:
hello , 
I get this error when i run OLA_IMU_BASICS code :  

**OLA_IMU_Basics:221:37: error: 'AP3_GPIO_DEFAULT_PINCFG' was not declared in this scope**

bool enableCIPOpullUp()
{
  //Add CIPO pull-up
  ap3_err_t retval = AP3_OK;
  am_hal_gpio_pincfg_t cipoPinCfg = AP3_GPIO_DEFAULT_PINCFG;
  cipoPinCfg.uFuncSel = AM_HAL_PIN_6_M0MISO;
  cipoPinCfg.eDriveStrength = AM_HAL_GPIO_PIN_DRIVESTRENGTH_12MA;
  cipoPinCfg.eGPOutcfg = AM_HAL_GPIO_PIN_OUTCFG_PUSHPULL;
  cipoPinCfg.uIOMnum = AP3_SPI_IOM;
  cipoPinCfg.ePullup = AM_HAL_GPIO_PIN_PULLUP_1_5K;
  padMode(MISO, cipoPinCfg, &retval);
  return (retval == AP3_OK);
}
Who can give me some help ,  please ?
Hamdi

PaulZC, 2021-05-19:
Hi Hamdi,
This looks like a problem with the Apollo3 Boards. Please try version 1.2.1 of the Apollo3 Boards. You can select this through the Boards Manager. The examples have not yet been tested or updated for the latest version (2.1.0) of Apollo3.
Best wishes,
Paul

Hamdi8, 2021-05-19:
hello ,
Thanks for your answer .
I already  paste the JSON package URL into the 'Additional Boards Manager URLs' field in Arduino preferences  ; https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/master/package_sparkfun_apollo3_index.json
and choose "RedBoard Artemis ATP"  in the Board Manager . 
I want to notice that i'm using 'DEV-16832 OpenLog Artemis' .
Even when i tried to correct the code , and make it run , i find this error : 
**error: 'padMode' was not declared in this scope**

my function : 

bool enableCIPOpullUp()
{
  //Add CIPO pull-up
  ap3_err_t retval = AP3_OK;
  am_hal_gpio_pincfg_t cipoPinCfg ;
  cipoPinCfg.uFuncSel = AM_HAL_PIN_6_M0MISO;
  cipoPinCfg.eDriveStrength = AM_HAL_GPIO_PIN_DRIVESTRENGTH_12MA;
  cipoPinCfg.eGPOutcfg = AM_HAL_GPIO_PIN_OUTCFG_PUSHPULL;
  cipoPinCfg.uIOMnum = 0;
  cipoPinCfg.ePullup = AM_HAL_GPIO_PIN_PULLUP_1_5K;
   //am_hal_gpio_pinconfig ( MISO , cipoPinCfg ) ; 
  padMode(MISO, cipoPinCfg, &retval);
  return (retval == AP3_OK);
}

Hamdi ,

PaulZC, 2021-05-19:
Hi Hamdi,
Yes, but **which version** of the Apollo3 boards have you selected?
I think you are seeing this problem because you are trying to use version 2 of the Apollo3 boards, maybe 2.1.0? The problem will go away if you select version 1.2.1 from the drop-down list in the Boards Manager.
Best wishes,
Paul

Hamdi8, 2021-05-19:
Hi Paul , 
I don't the version in the drop-down list in the Boards Manager. I can find only the type of the board and i select "RedBoard Artemis ATP". 
Sincerly ,
Hamdi,

adamgarbo, 2021-05-19:
Hamdi,

Make sure your Boards Manager looks like the following:

[image]

You can click on "Select Version" and choose 1.2.1 from the list. If you can't see this, then that's a different problem entirely and you may need to look at removing the Apollo3 Core and starting over.

Cheers,
Adam

Hamdi8, 2021-05-20:
Hello, 
Thanks a lot for your answer .
The problem is solved . 
Sincerly, 
Hamdi