sparkfun/OpenLog_Artemis #73: Feature Request: Add a GNSS data-logging example for RAWX etc...

PaulZC, 2021-02-18:
Based on this forum post, it would be useful to have a stand-alone example showing how to use the OLA as a "RedBoard" for GNSS data logging

adamgarbo, 2021-02-18:
I wonder if an OpenLog Artemis board definition might be the best way to allow users to make use of the OLA for such purposes. An approach similar to the MicroMod ecosystem could work well, where the control circuitry is made available through custom definitions (e.g. G1, G2, etc. on the MMDLCB).

PaulZC, 2021-04-26:
GNSS_Data_Logging example has been added to the [release_candidate branch](https://github.com/sparkfun/OpenLog_Artemis/tree/release_candidate/Firmware/Test%20Sketches).
Sadly no custom board definition - yet. That will need to wait until v2.0 of the core is a bit more stable.
Cheers,
Paul

PaulZC, 2021-06-01:
Resolved in v1.11. The new GNSS example is here:

https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/Test%20Sketches/GNSS_Data_Logging/GNSS_Data_Logging.ino

closed 2021-06-01 by PaulZC