sparkfun/OpenLog_Artemis #74: Some questions about the board

idea--list, 2021-02-20:
This is not an issue, have just some questions as i find this board really interesting for future projects:

- Can we turn off all the LEDs to avoid them draining battery? (on the Thing Plus the red status LED is always on by HW design)
- Does this board expose at least 1 SPI bus or we can attach I2C sensors only?
- On which Arduino core is the OLA firmware based? V1.x or V2.x?
- Can we program this board with the official Mbed OS releases or this board is meant only for the Arduino platform (i ask because i can not find pin definitions in Mbed for this board but for all the other Artemis boards)

adamgarbo, 2021-02-20:
Hi @idea--list,

I'm sure @PaulZC will chime in here, but to answer your questions...

> * Can we turn off all the LEDs to avoid them draining battery? (on the Thing Plus the red status LED is always on by HW design)

Yes. The PWR and STAT LEDs are tied to GPIO pins and can be controlled through software. The TX/RX LEDs will light up with Serial traffic, and the CHG when charging a battery.

> * Does this board expose at least 1 SPI bus or we can attach I2C sensors only?

I believe both the microSD and IMU run on SPI so you're likely going to be limited to I2C sensors. Paul can confirm this.

> * On which Arduino core is the OLA firmware based? V1.x or V2.x?

 V1.2.x

> * Can we program this board with the official Mbed OS releases or this board is meant only for the Arduino platform (i ask because I can not find pin definitions in Mbed for this board but for all the other Artemis boards)

The OLA uses the SparkFun Artemis Reboard ATP board definition to compile its firmware as it doesn't yet have its own. I'm unfamiliar with Mbed OS, so please let me know what you find. 

Cheers,
Adam

PaulZC, 2021-02-20:
Hi @idea--list ,
@adamgarbo is completely correct.
The board is _very_ compact and it wasn't possible to break out an SPI bus directly. A hack, if you're desperate, would be to use a [micro SD sniffer](https://www.sparkfun.com/products/9419) to access the pins.
Best wishes,
Paul

closed 2021-02-20 by PaulZC

idea--list, 2021-02-20:
@PaulZC 
That is a great idea!
Would the sniffer also allow me reading an SPI sensor at 500Hz and log the data to the SD card?

PaulZC, 2021-02-20:
500Hz sounds high... But in principle, yes. Don't forget you'll be trying to get both your sensor traffic and your SD data through the same bus. You will almost certainly need to buffer your sensor data while you do the SD write in appropriate block sizes (usually 512 bytes). You will also need to use one of the 'analog' pins as the Chip Select for your sensor.
"Your mileage may vary", but it's worth a try?
All the best,
Paul
PS. thanks for your contributions on the forum. Much appreciated.

PaulZC, 2021-02-20:
The sniffer can be slightly oversized on delivery - due to the cut edges on the board.
If it doesn't slide neatly into the SD socket, carefully file _**a little**_ off the edges with an emery board.
Eyeball the pads first. You may need to remove more on one edge than the other.
Don't force it into the socket or you'll wreck the tiny Geneva mechanism inside.

idea--list, 2021-02-20:
This board is getting more and more interesting for me!
I mean right now i am happy using the Artemis Thing Plus + an [EPD board](https://www.adafruit.com/product/4195) that also adds an SD-slot to my prototype but i still lack an IMU with this combo. The OpenLog Artemis has that also on board...and my SPI sensor has a FIFO anyway :wink:

PaulZC, 2021-02-20:
Cool. Have fun!