sparkfun/OpenLog_Artemis #75: Examples Not Working

cneves-idmind, 2021-02-22:
### Your workbench
I'm using the OpenLog_Artemis Board, with no SD Card and no adittional sensors. It is logging only to Serial at 10Hz.
I updated the repository today, the sketch and firmware files are version 19. The board is powered by USB-C from a Desktop.

### The issues
Following the update on the [ICM20948 library](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary) (in order to use DPM to output an orientation quaternion), I have been going through the code in this repository. 
**Issue 1:** After updating the library, I have compiled and uploaded the OpenLog_Artemis sketch. There is no serial output and no blinking leds.
**Issue 2:** I compiled and uploaded OLA_IMU_Basics.ino. The serial output is:
```
ICM_20948::checkID: ICM_20948_chICM_20948::checkID: ICM_20948_check_id returned: Wrong ID
ICM_20948::startupDefault: checkID returned: Wrong ID
ICM_20948_SPI::begin: startupDefault returned: Wrong ID
Initialization of the sensor returned: Wrong ID
Trying again...

```

### Expected behavior
Since the sketches were not changed, I expected the output to remain the same.

In addition, I have not been able to get LEDs to blink in a simple sketch.
Any help getting the update library for the ICM to work is greatly appreciated.

Thanks in advance,
Carlos

PaulZC, 2021-02-22:
Hi Carlos (@cneves-idmind ),
Thank you for reporting this issue.
I've just tested both the OLA firmware (1.9) and the OLA_IMU_Basics example and they both run OK, both with the DMP loaded and without.
Can you please check any changes you made to the code very carefully? It may not be possible to use the standard IMU functions (to get accel, gyro and compass data) and the DMP (to get quaternions) simultaneously - I have not tested this.
Please check you have "SparkFun RedBoard Artemis ATP" selected as the board.
Please also check that you are using version 1.2.1 of the SparkFun Apollo3 Boards package. The OLA code will not run on the new Mbed v2 of the core.
You might find it useful to add:
```
if (settings.printDebugMessages) myICM.enableDebugging();
```
above line 865 in OpenLog_Artemis.ino
```
myICM.begin(PIN_IMU_CHIP_SELECT, SPI, 4000000); //Set IMU SPI rate to 4MHz
```
You can enable the debug messages via the 'hidden' debug menu. Type "d" at the main menu to access the debug menu.
Best wishes,
Paul

PaulZC, 2021-02-22:
Also, do you have the black (SparkX) or the red (SparkFun) version of the OLA?
You will need to make some changes to the code to let it run on the older SparkX version.
Best wishes,
Paul

cneves-idmind, 2021-02-23:
Thanks for the quick reply.

I had the wrong board selected ("RedBoard Artemis" instead of the "RedBoard Artemis ATP").
I was able to get OLA_IMU_Basics to work, and get quaternion output like in the ICM Library example.

(I will have to test the settings and try to understand how the quaternion only has 3 values, but that is another issue).
Thanks

closed 2021-02-23 by cneves-idmind

PaulZC, 2021-02-23:
Hi Carlos (@cneves-idmind ) ,
Thanks for letting me know. Not a problem.
InvenSense normalize the quaternions so that
Q0^2 + Q1^2 + Q2^2 + Q3^2 = 1
You can calculate Q0 with sqrt( 1 - ((Q1*Q1) + (Q2 * Q2) + (Q3 * Q3)))
All the best,
Paul