sparkfun/OpenLog_Artemis #76: Keypress won't open menu until sample time

whipple63, 2021-02-24:
Keypress won't open menu until sample time
When pressing any key to open the menu during slow sample rates, the key is ignored until it is pressed very near the sample time.

### Your workbench
* Are you using a microSD card? If so, what size? How is it formatted (FAT, FAT32, or exFat)? 32G Fat
* At what frequency are you logging? For example, 10Hz, 1Hz, once every 5 seconds, etc.: Once every 10s, but will be longer...
* What version of firmware are you using? This is shown at boot, for example: "Artemis OpenLog v1.10": 1.9
* How is OpenLog Artemis wired to your sensor(s)? 1 qwiic micro pressure sensor
* How is everything being powered? battery
* Are there any additional details that may help us help you?

### Steps to reproduce
This should be easily replicated, but just in case starting from default:
•	57600 Baud
•	Log accelerometer (turn off gyro, turn magnetometer)
•	Log VIN (battery) voltage
•	Enable measurement counts
•	Turn off ‘output actual hertz’
•	Set the clock
•	Change pressure sensor to use kPa
•	Set up when to open new log files 3600s

### Expected behavior
Ideally the menu would come up - even if after a short wait.  It shouldn't wait until next sample.  I plan on dropping sample period down to 30 minutes when all is set up.  I realize that I can reset the device to access the menu, but I will be using this remotely on a data collection device.

### Actual behavior
The menu won't come up unless a key is pressed very near the sample time (i.e. hold a key until it pops up).

PaulZC, 2021-02-24:
Hi @whipple63 ,
The issue here is that the Artemis is placed into "deep sleep" when the sample period exceeds 2 seconds. In deep sleep it can't respond to data arriving via UART. I sympathize, but I can't think of a way of implementing what you're asking for...
Sorry!
Closing...
Paul

closed 2021-02-24 by PaulZC

whipple63, 2021-02-24:
How about allowing for a period of time after the sample where the system listens for a character?