sparkfun/OpenLog_Artemis #78: Refined low-power code

adamgarbo, 2021-03-26:
Hi @PaulZC,

The other Paul, Robin, and I have recently been revisiting the Artemis low-power code. It turns out that we've been incorrectly using the `am_hal_pwrctrl_memory_deepsleep_powerdown()` function (see: https://forum.sparkfun.com/viewtopic.php?f=169&t=55163#p224296). 

Because the Artemis isn't necessarily writing to the SRAM starting at the first memory location, retaining the lowest 64K is not advisable (see: https://forum.sparkfun.com/viewtopic.php?f=169&t=50904&p=224374#p224374). We've basically been powering down the first 64K of SRAM and have been lucky that no code was actually written to those banks.

Based on these findings, the recommended method of powering down the memory looks to be:

```
//Power down cache, flash, SRAM
am_hal_pwrctrl_memory_deepsleep_powerdown(AM_HAL_PWRCTRL_MEM_ALL); // Power down all flash and cache
am_hal_pwrctrl_memory_deepsleep_retain(AM_HAL_PWRCTRL_MEM_SRAM_384K); // Retain all SRAM
```

This will power down all the cache (? uA) and flash (~110 uA) but retain all of the SRAM. The power consumption difference between 32K and 384K is <1 uA, which is negligible, even for me. 

Additionally, upon wake, it is not necessary to power up the memory and the following line of code can be safely removed as it doesn't actually do anything:

```
//Power up SRAM, turn on entire Flash
am_hal_pwrctrl_memory_deepsleep_powerdown(AM_HAL_PWRCTRL_MEM_MAX);
```

Cheers,
Adam

PaulZC, 2021-03-27:
Many thanks Adam - great detective work!
I must admit I saw a similar issue on the Artemis Global Tracker and changed the [low power code](https://github.com/sparkfunX/Artemis_Global_Tracker/blob/master/Software/examples/Example16_GlobalTracker/Example16_GlobalTracker.ino#L2244-L2247) there so it no longer disabled the RAM.
Does your revised code work on both v1.2.1 and v2.x?
All the best,
Paul

adamgarbo, 2021-03-27:
Hi Paul,

I've extensively tested the low-power code on v1.2.1 as well as v2.0.6. It was good to get confirmation from Robin, who directly watched all of the bits change, and also look at the build files to determine that the code was written to the SRAM around the 90K byte locations. This completely explained why it was failing to recover from sleep when only the first 64K SRAM was retained.

Cheers,
Adam

PaulZC, 2021-03-28:
Nice one - thank you Adam,
I'll roll this into the next update.
Cheers!
Paul

PaulZC, 2021-03-30:
Fixed in v1.10

closed 2021-03-30 by PaulZC