sparkfun/OpenLog_Artemis #79: OLA locks up after sleeping while logging to TX pin

ryanneve, 2021-03-29:
### Subject of the issue
Logging fails if **Output serial data to TX pin** is Enabled and sample rate > 2 seconds (so that OLA goes to sleep between samples).

### Your workbench
OLA connected to PC via USB-C and also connected to same PC via TX pin and BlueSMiRF module. Only sensor is a microPressure board.

### Steps to reproduce
If I set sample rate to 1 second, I can see samples on both USB-C and TX pin simultaneously. If I then set sample rate to 3 seconds, after menu exit I see one more sample on TX and two more samples on USB-C. The last sample seen on USB-C does not end up in dataLogXXXXX.txt file. The last sample is also much less than 3 seconds after the previous sample (see below)
OLA is non-responsive, and the only way to recover is pressing the reset pin,

### Expected behavior
Should not lock up.

### Actual behavior
OLA locks up and is non-responsive.
```
x) Return to logging
rtcDate,rtcTime,VIN,aX,aY,aZ,imu_degC,kPa,output_Hz,count,
03/29/2021,13:49:17.87,5.21,-171.39,-9.28,-1013.18,27.65,102.0902,100.00,5913,
03/29/2021,13:49:18.33,5.33,-167.97,0.49,-1007.32,27.17,102.0923,4.26,5914,
```

Let me know if you need any other details of configuration.

PaulZC, 2021-03-29:
Hi Ryan,
Thanks for reporting this.
Which version of the code are you using?
And are you using the firmware binary - or have you compiled your own code?
Thanks,
Paul

PaulZC, 2021-03-29:
Ah... Looks like I might have missed a call to beginSerialOutput. Here is where it gets called in the setup code:
https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino#L362
We should call it after this line in wakeFromSleep too:
https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/OpenLog_Artemis/lowerPower.ino#L452
I’ll try and update the code tomorrow.
Cheers,
Paul

ryanneve, 2021-03-29:
Paul, looks like you have it figured out, but it's stock 1.9.

PaulZC, 2021-03-30:
Fixed in v1.10

closed 2021-03-30 by PaulZC