sparkfun/OpenLog_Artemis #81: Feature Request: SparkFun Pulse Oximeter and SparkFun Qwiic Button board

LSMEDEL, 2021-04-08:
Request for Qwiic button and Oximeter Sparkfuns Boards

I use machine learning to analyze the data from the sd card, how ever the data must be labeled (for example for human activity recognition), so a nive adition would be the Qwiic buttons, so the data can be properly labeled.

The oximeter data from the sparkfun site, is an important feature to log, specially in this pandemic to analyze the data.

So far I wrote an Android App to log the labeled data (as a second system), but then I have to add the data manually and sometimes I get a lost data.

Could it be possible to add button and oximeter features? if so how

PaulZC, 2021-05-29:
Hi @LSMEDEL ,

Thank you for your patience.

I am adding support for the Oximeter to the OLA, together with support for the Qwiic Button. Can you please tell me if the Mode2 Oxygen R value is important to you? I.e. do you need me to implement both Mode1 and Mode2?

Best wishes,
Paul

LSMEDEL, 2021-05-30:
Hi Paul,

Thanks for replying, the Mode2 Oxygen R value is important, because it can
help to detect hyperventilation (for example in epilepsy among others), due
the low frequency the biometrics variables change (3 Hz for example) Mode2
should be enough.

The main functions for the quicc button are to classify information, for
example Matlab that requires all being labeled, so a button or buttons can
help to this purpose in real time.

Thank you so much for your support, and so sorry I reply to you late.

Best wishes,
Medel

Ps. I got 12 OLAS, this is one of the best boards I ever tried.

PaulZC, 2021-05-31:
Hi Medel (@LSMEDEL ),

Thank you for the feedback. I have the oximeter working nicely in Mode1. It will be straightforward to change this to Mode2 and add the R value.

The oximeter needs two IO pin connections - for RESET and MFIO - so I have added a special feature to the OLA code for that. You can now tell the code that you will have an oximeter connected, and then it dedicates pins 32 and 11 for RESET and MFIO. This means you cannot use them for analog logging etc., but you can still use the TX and RX pins for that. I hope this is OK for your application? If you need me to add a feature so that it uses the TX or RX pins instead for RESET and MFIO, please let me know.

I am struggling a little with the Qwiic Button. It uses a technique called clock stretching on the Qwiic I2C bus, and that causes problems for the Artemis processor. I can provide more details if you are interested. I am going to upgrade the code so it uses the new version of the Apollo3 core (v2.1.0) to see if it corrects this issue. I will let you know how it goes.

Best wishes,
Paul

LSMEDEL, 2021-05-31:
Hi Paul,

That's perfect with the Oximeter, I don't use the pins 11 and 32 at the
moment, but if any I have others OLA for that. It really is perfect.
It kind of surprises me with the issues with Qwiic Button, ironically I
thought of it as only a button, if it is too much trouble, never mind, your
work with the Oxymeter is really helpful.

To label the data I usually use an app I wrote for Android, the tricky part
is that it has to be nicely synchronized with the RT clock by hand.

Thank you so much.

PaulZC, 2021-06-01:
Hi @LSMEDEL ,

Thanks again for this suggestion. This feature has been added in v1.11. Please follow the [upgrade instructions](https://github.com/sparkfun/OpenLog_Artemis/blob/master/UPGRADE.md) to update your firmware.

If you open Menu 6 (Detect / Configure Attached Devices), you will see a new option to enable oximeter detection. Select that option, press "y" to confirm. The oximeter will then be detected and started. As discussed, you will need to connect: pin 32 to the oximeter RESET pin; and pin 11 to the MFIO pin.

I wasn't able to add support for the Qwiic Button... The [clock stretching issue](https://github.com/sparkfun/Arduino_Apollo3/issues/400) was causing difficulties. There is a work-around you can use: if you are not using the TX and RX pins, enable those for analog logging. Connect a 1K resistor between the TX pin and 3V3. Do the same for the RX pin. Then connect a toggle switch between the TX pin and GND. Do the same for the RX pin. The analog pins will see 3.3V ("high") when the switches are open, 0V ("low") when the switches are closed. The state of the switches will be logged as analog readings in your data. It is not an elegant solution, but it will work! ;-)

I'm closing this issue but please re-open if you find any issues with the new feature.
Best wishes,
Paul

closed 2021-06-01 by PaulZC