sparkfun/OpenLog_Artemis #82: OLA_settings.txt corrupted after editing attached Device Settings

ryanneve, 2021-04-21:
### Subject of the issue
Having trouble with OLA_settings.txt getting corrupted, and OLA locking up.

### Your workbench
OLA running 1.10 firmware.
Logging every 5 seconds.
MPR MicroPressure Sensor
Samsung 32GB uSD card freshly formatted with SD card formatter.

### Steps to reproduce
- Insert uSD card formatted with SD card formatter.
- Power up OLA via USB-C, connect and go into main menu
- Select 6) Detect / Configure Attached Devices
- Select 1)  MPR MicroPressure Sensor (0x18)
- Change units from PSI (default) to kPa.
- Enter x to exit menu.
- Select 2) Configure Qwiic Settings

### Expected behavior
Should save changes and go into Qwiic settings. Should never lock up.

### Actual behavior
OLA locks up on final step above.
After power cycle:

- OLA_deviceSettings.txt does not appear to have been altered to reflect change in units.
- OLA_settings.txt is corrupt. Entering cat OLA-settings.txt will lock up OLA.
- If card is removed and inserted into PC, OLA_settings.txt can not be read.

If I delete OLA_settings.txt and power cycle, the OLA will take one sample, then lock up.

My OLA_settings.txt looks like this:

cat OLA_settings.txt

sizeOfSettings=152
olaIdentifier=282
nextSerialLogNumber=1
nextDataLogNumber=22
usBetweenReadings=5000000
logMaxRate=0
enableRTC=1
enableIMU=1
enableSD=1
enableTerminalOutput=1
logDate=1
logTime=1
logData=1
logSerial=0
logIMUAccel=1
logIMUGyro=0
logIMUMag=0
logIMUTemp=1
logRTC=1
logHertz=0
correctForDST=0
americanDateStyle=1
hour24Style=1
serialTerminalBaudRate=115200
serialLogBaudRate=9600
showHelperText=1
logA11=0
logA12=0
logA13=0
logA32=0
logAnalogVoltages=1
localUTCOffset=0
printDebugMessages=0
powerDownQwiicBusBetweenReads=0
qwiicBusMaxSpeed=100000
qwiicBusPowerUpDelayMs=250
printMeasurementCount=1
enablePwrLedDuringSleep=1
logVIN=1
openNewLogFilesAfter=0
vinCorrectionFactor=1.47
useGPIO32ForStopLogging=0
qwiicBusPullUps=1
outputSerial=0
zmodemStartDelay=20
enableLowBatteryDetection=0
lowBatteryThreshold=3.40
frequentFileAccessTimestamps=0
useGPIO11ForTrigger=0
fallingEdgeTrigger=1
imuAccDLPF=0
imuGyroDLPF=0
imuAccFSS=0
imuAccDLPFBW=7
imuGyroFSS=0
imuGyroDLPFBW=7
logMicroseconds=0
useTxRxPinsForTerminal=1
timestampSerial=0
timeStampToken=10
useGPIO11ForFastSlowLogging=1
slowLoggingWhenPin11Is=0
useRTCForFastSlowLogging=0
slowLoggingIntervalSeconds=300
slowLoggingStartMOD=1260
slowLoggingStopMOD=420
resetOnZeroDeviceCount=0

PaulZC, 2021-04-22:
Hi Ryan (@ryanneve ),
Thanks for reporting this. I will look at it as soon as I can.
Best wishes,
Paul

PaulZC, 2021-04-23:
Hi Ryan (@ryanneve ),
It looks like this is linked to ```useTxRxPinsForTerminal```. I do see the issue if ```useTxRxPinsForTerminal=1```, but not if ```useTxRxPinsForTerminal=0```. I'll keep investigating.
Thanks,
Paul

PaulZC, 2021-04-23:
Oooh! This is a good one... Spot the difference:
[image]

PaulZC, 2021-04-23:
OK. I think the problem is caused by this [line of code](https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/OpenLog_Artemis/menuAttachedDevices.ino#L348). SerialPrintf2 is actually a macro, defined [here](https://github.com/sparkfun/OpenLog_Artemis/blob/master/Firmware/OpenLog_Artemis/OpenLog_Artemis.ino#L264). So that means the ```availableDevices++``` will be being performed _twice_ and will cause all kinds of chaos. When you enter 2, it will actually be trying to call a non-existent menu function, because ```availableDevices``` is 3, not 2.
If you enter "3", then it works...
Fixing...

PaulZC, 2021-04-23:
Hi Ryan (@ryanneve ),
Please try the [v1.11 BETA firmware](https://github.com/sparkfun/OpenLog_Artemis/tree/master/Binaries) and let me know if this fixes your issue.
Best wishes,
Paul

ryanneve, 2021-04-23:
@PaulZC , I'll try to test it today.  Thanks!  -Ryan

ryanneve, 2021-04-23:
@PaulZC , Loaded it, and it seems to work. Thanks!

closed 2021-04-23 by ryanneve