sparkfun/OpenLog_Artemis #90: Get more decimal places in DPM Quat6 and Quat9

toni-lv, 2021-07-04:
Hello,

I am using DPM to get quaternion data directly from the readings in firmware version v1.11. From quaternion data I get 4th quaternion data in order to use this rotation data. However due to decimal precision sometimes error is too big and it is not possible to get a valid rotation.

Is it possible to get more decimal places from log in order to avoid this behavior? Is any other way to get rotations in euler angles?

P.S. Everytime I start reading data I do calibration just in case accel and gyro calibration improves readings but it does not help.

Thanks for your help

PaulZC, 2021-07-04:
Hi @toni-lv ,
More decimal places is certainly possible. I’d need to edit (e.g.) this line:
https://github.com/sparkfun/OpenLog_Artemis/blob/b8bab7732d70572dad3ea398ba46e56a58277a93/Firmware/OpenLog_Artemis/Sensors.ino#L138
it will mean up-issuing the firmware, but it is an easy change.
How many decimal places do you need?
Best wishes!
Paul

PaulZC, 2021-07-04:
You will need to calculate the Euler angles yourself from the Quaternions. See #47 for more details.

toni-lv, 2021-07-04:
I was using quaternions in order to get sensor fusion done inside chip with DPM, so I get better orientation, that is why I was calculating Euler angles with this formula:  
q0 = sqrt( 1.0 - ((q1 * q1) + (q2 * q2) + (q3 * q3)));
But, I have noticed while calculating Euler angles from quaternion data sometimes this term ((q1 * q1) + (q2 * q2) + (q3 * q3)) is bigger than 1, and therefore q0 term cannot be calculated.
I think adding 2 additional decimal places can avoid rounding errors and previous term will be smaller than 1. Consequently q0 is calculated and Euler angles conversion should be straightforward to do (like most of quaternions readings). I think this maybe works, but maybe there is a smarter way to do this, if anyone knows.

PaulZC, 2021-07-05:
Hi @toni-lv ,
Please try the v1.12 BETA binary:
https://github.com/sparkfun/OpenLog_Artemis/tree/main/Binaries
Upgrade instructions are here:
https://github.com/sparkfun/OpenLog_Artemis/blob/main/UPGRADE.md
Best wishes,
Paul

toni-lv, 2021-07-05:
Hello @PaulZC, I have upload firmware and perfom gyro and acceleremoter calibration. Results improve, but I have still receiving some values of q1, q2 and q3 that when adding its squared are bigger than 1 but with very little difference (1.0000058753999999, 1.0000020555), my readings file is this: [dataLog00075.TXT](https://github.com/sparkfun/OpenLog_Artemis/files/6763046/dataLog00075.TXT)
I think this numerical issue should be solved in another way.

PaulZC, 2021-07-05:
Thank you @toni-lv ,
There is a comment in the InvenSense application note that "Q0 value is computed from this equation: Q0^2 + Q1^2 + Q2^2 + Q3^2 = 1. In case of drift, the sum will not add to 1, therefore, quaternion data need to be corrected with right bias values.".
Maybe there is a small drift / bias error - even after calibration?
Best wishes,
Paul

toni-lv, 2021-07-05:
Yes, I think there is still some bias. For my particular case I normalize data again when values are over 1, but I think this is not an ideal solution.

PaulZC, 2021-07-16:
Hi @toni-lv ,
I have included the five decimal places in the new version of the firmware: v2.0 (which uses v2.1.0 of the Apollo core). Please give v2.0 a try when you have time.
I'm going to close this issue as I don't think I can help any further with the drift / bias error.
Best wishes,
Paul

closed 2021-07-16 by PaulZC