sparkfun/OpenLog_Artemis #92: Wrong IMU orientation or top/bottom part of board confused?

toni-lv, 2021-07-14:
### Subject of the issue
I am retrieving orientation data from IMU with DMP mode enabled and after a while IMU flips 180 degrees. I tried a static reading but 180 degrees flip happens. I tried several configurations an calibrating IMU before, but this strange flip happens and once it is flipped all rotations are recorded properly.
I started reading how algorithm fuse use code from SparkFun 9DoF IMU Breakout - ICM-20948, and looking at it I found this difference on boards (image below).

[image]

As you can see IMU orientation it is drawn the same, but in ICM-20948 IMU is attached on top side of board and in OpenLog Artemis this is mount on bottom side of board. So my question is IMU direction fipped when calculating angles due to this difference, or  I was assuming badly and what I think it was bottom side it is really top side and my mounting was wrong and strange flip comes from another source of error?

PaulZC, 2021-07-14:
Hi @toni-lv ,

The important thing is the orientation of the accelerometers, magnetometers and gyros with respect to the ICM-20948 chip package:

[image]

The silkscreen on both boards is correct. Note the chip package pin 1 marker dot. In those product images, the package on the OLA is rotated by 90 degrees CCW compared to the ICM-20948 Breakout.

When using the DMP, the orientation of the magnetometer is corrected with a matrix multiply:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/486fdbfdf73d8d790aef1903f2d5ceca22a35f46/src/ICM_20948.cpp#L1305-L1325

The Quaternions will be using the accelerometer and gyro as the frame of reference.
To avoid confusion, please orientate your OLA bottom-up. ;-)
Best wishes,
Paul

toni-lv, 2021-07-14:
Yes, silkscreen is fine I was guessing on IMU chip mounted on top or bottom which maybe was different.
DMP algorithm show this behaviour with and without magnetometer into quaternion output.
So to make it clear top side is where connectors are attached (image below)?

[image]

adamgarbo, 2021-07-14:
Hi @toni-lv,

This is the equivalent orientation to the ICM-20948:
[image]

Cheers,
Adam

toni-lv, 2021-07-14:
Thanks for everything

closed 2021-07-14 by toni-lv

ormapper, 2021-10-18:
I tried following the orientation advice above and I was expecting the Roll Pitch Yaw orientations from the IMU_DMP_Quat6 and MU_DMP_Quat9 sketches to match simple aircraft orientations. However, orienting my board as shown below provides a match for Yaw (with values of +180/-180), but a nose up in this case produces an increase in Roll value not Pitch, and a wing tip up/down produces a change in Pitch, not Roll. I tried flipping the board over but all that seemed to change was the Yaw direction.
[image]