sparkfun/OpenLog_Artemis #93: Drifting of rotations with DMP Quat9 when steady

toni-lv, 2021-07-14:
### Subject of the issue
As initially stated on issue #92 I registered strange behavior when sampling with Openlog Artemis looking upwards. After performing some test I came up with this strange behavior, but if I let Artemis with bottom view as top view this kind of crazy rotation does not occur.

### Workbench
* 32 GB microSD card.
* 20 Hz sampling
* OpenLog_Artemis-V10-v112_BETA.bin, same happened to me when using  OpenLog_Artemis-V10-v111.bin
* USB wiring, similar behavior when using 1S battery.

### Steps to reproduce
Upwards: steady Openlog with Artemis upwards hold while sampling data.
Downwards: steady Openlog with Artemis downwards hold while sampling data.

### Expected behavior
It should register data and steady movement (some drift in yaw should be possible due to registering Quat9 close to a PC). Same should be when downwards but with initial conditions rotated 180 degrees.

### Actual behavior
When upwards it rotates 180 degrees. However when downwards no, and behaves as expected.

- Upwards data
[upwards.TXT](https://github.com/sparkfun/OpenLog_Artemis/files/6817165/upwards.TXT)

Video with quaternion rotation  with every data point in Blender
https://www.youtube.com/watch?v=kQouJCbC0Q0

- Downwards data
[downwards.TXT](https://github.com/sparkfun/OpenLog_Artemis/files/6817168/downwards.TXT)

Video with quaternion rotation  with every data point in Blender
https://www.youtube.com/watch?v=ccNfk01VkFI

PaulZC, 2021-07-15:
Hi @toni-lv ,
Thanks for reporting this, but I don't think there is very much we can do about it. There is nothing in the DMP configuration that could alter or control this behavior.
How are you calibrating the DMP? I recommend - as a minimum: rotating it slowly around all three axes; and holding it stationary for a few seconds in all six orientations.
Best wishes,
Paul

toni-lv, 2021-07-15:
Hello,
Yes I perform calibration with this procedure, but only difference appreciated was that during calibration rotations are registered properly, but as soon as they finish this 180º drift appears. That is why I start to try static tests. I will try to perform rotation calibration even more slowly to try this.
Thanks for your help

PaulZC, 2021-08-29:
Hi @toni-lv ,

I am going to close this issue as I think it has gone stale? But please re-open if you need more help with this.

Very best wishes,
Paul

closed 2021-08-29 by PaulZC

RoopanJK, 2023-01-20:
Hi @PaulZC ,

I'm using the board Sparkfun Openlog Artemis DEV-16382, the board is amazing and worked fine out of the box with default firmware. It's been 2 weeks and I was getting considerable accurate Yaw values without calibration. But now the yaw values drift to a particular angle from its position.

Look how the yaw points back to E everytime I rotate the IMU.
Need help. Thanks in advance.
[image]

PaulZC, 2023-01-20:
Hello @RoopanJK ,

Are you sure you are calibrating the DMP correctly? Rotate slowly around all three axes; and hold still in all six orientations for a few seconds.

Please also make sure you are in a 'clean' environment, well away from any magnetic or ferrous materials. Loudspeakers have strong magnets in them.

Best wishes,
Paul

RoopanJK, 2023-01-20:
Do you mean the typical sensors calibration (accel, gyro, mag) while mentioning DMP calibration?? And yes I tried calibrating it several times but same drift occurs.
The thing is this wasn't an issue until yesterday till then I did not do any calibration and the Yaw values were pretty accurate and even points towards true north with a slight deviation.

I am sure that I am in a 'clean' environment, no change in environment. No setup changes.

Only change was that I was carrying it to my mechanical area to drill holes in my plate to mount it only then after I noticed this problem. They did not mishandle the board but wasn't wearing any ESD resistance gear. Could ESD be of any potential issue here?

PaulZC, 2023-01-20:
Did you go past any strong magnetic fields? Like welding equipment perhaps?

Are you using steel screws? Could they be magnetized?

What material is your plate made from?

RoopanJK, 2023-01-20:
No I don't think I wasn't around any strong magnetic field environment for sure.
Yes they are steel screws, but currently testing just with the bare board, no screws attached.
My plate is of 4mm Acrylic sheet.

I don't see any physical issue caused here. But i am puzzled what would've caused this. The only possible reason I could think of could be ESD but the board was also carefully handled in that manner too.

Have similar issues been there earlier, or is this the first. The roll and pitch doesn't drift away. but just the yaw.

And If I do not use Drift Correction, the values are ok but they keep incrementing at a constant rate.

Thanks,
RoopanJK

PaulZC, 2023-01-20:
If the board is ~horizontal: roll and pitch depend most the accelerometer readings; yaw depends most on the magnetometer. Something is causing the magnetometer readings to change or drift. Usually this is caused by a stray magnetic field.

Could there be an error in your drift correction code?

You need to re-calibrate the DMP _every time_ you upload new code or disconnect the power.

RoopanJK, 2023-01-20:
Please elaborate on recalibrating the "DMP". Please bare with me.

Thanks,
RoopanJK

RoopanJK, 2023-01-20:
There are no errors in my drift correction code, used the one provided with the ROS packages repository worked in the past.

PaulZC, 2023-01-20:
"Rotate slowly around all three axes; and hold still in all six orientations for a few seconds."

I can only help you if you are using our code examples - unmodified. If you add additional code, it becomes impossible for me to help you.

RoopanJK, 2023-01-20:
I've used the code from this repository
https://github.com/ENSTABretagneRobotics/razor_imu_9dof
It worked fine out of the box, this sudden error is what that's making this more complicated.

Can you help me just extract the inertial values for yaw, without any magnetometer influence? I've tried turning off drift correction given in this code but the yaw keeps incrementing as such.

[image]

Notice how the values constantly increment even thought the sensor is static.

PaulZC, 2023-01-20:
You can use Quat6 instead of Quat9. Quat6 only uses the accelerometers and gyros. It will give you the orientation (including yaw) **relative** to when the DMP was started. It will drift over time.

ErwanODS, 2023-01-26:
Hello,

I have been working with OLA for several months for embedded devices on vehicules.
I encounter a major problem which is the re-calibration of the DMP after each restart and/or deconnection to the power supply.
I would need to start the OLA in a moving vehicule and not have to recalibrate the DMP ever time.
If I understand correctly it's not possible ?

When the OLA starts without being stabilized , the pitch and the roll are erroneous. 
Unfortunately correct values are obtained only after the OLA is still for a few seconds...

Thanks for your job and answers.
Erwan

ErwanODS, 2023-01-26:
Wouldn't it be possible to save all configuration parameters of the DMP in order to reload it at the next restart ?

PaulZC, 2023-01-27:
Hi Erwan (@ErwanODS ),

This is possible. We have a code example which shows how to save the DMP biases:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/examples/Arduino/Example11_DMP_Bias_Save_Restore_ESP32/Example11_DMP_Bias_Save_Restore_ESP32.ino

The question is: _when_ should be the biases be saved? Periodically? On "Quit"? Manually - via the menu?

Best wishes,
Paul

ErwanODS, 2023-02-01:
Hi PaulZC,

Thanks a lot for your answer and your code example.
I update SparkFun_ICM_20948_IMU library to the lastest version 1.2.11 and try to save the DMP biases in my code.

However multiples behaviours remain misunderstood :

- I calibrate IMU exactly like you mentionned in your example code : "Rotate the sensor around all three axes" and "Hold the sensor stationary in all six orientations for a few seconds". After a few tens of seconds the DMP generally updates compass biases only. Gyro and accelero biases are hold to 0. Is it normal ?
[image]
[image]

- These 3 biases are then saved in EEPROM after one minute (for my tests) but compass bias is continously updated by the DMP whether the imu is moving or not :
[image]

- From what I understand the DMP calculate the gyro bias and set it only after the IMU is hold stationary.
So if the DMP biases are restored when the IMU is started in a moving vehicle, as soon as the movements stop the gyro bias is updated automatically.

- Finally I can not understand how is updated the accelerometer bias because it always remains at 0.
I succes only one time to save values for its bias but I don't know how...

ErwanODS, 2023-02-01:
Finally if I make the IMU go back and forth on the Y axis while it is starting, the DMP transmits wrong pitch and roll values after its calibration. The biases are restored from the start and the DMP keeps them.

And when I stop the movements of the IMU, the values ​​of roll and pitch then converge slowly towards their initial value of 0, without the biases of the DMP having been modified by it...
Which would mean that there are other parameters that are updated, right?

https://user-images.githubusercontent.com/123651887/216133063-798088f2-411a-401d-9487-8f61fa353f33.mp4

Best wishes,
Erwan

PaulZC, 2023-02-02:
Hi Erwan (@ErwanODS ),

I am sorry. I do not have all the answers. The DMP is a 'mystery' in many ways...

There is a long discussion about the biases here which you may find useful:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/50

I hope this helps,
Paul

ErwanODS, 2023-02-15:
Hi @PaulZC,

I have read this long discussion and it was very interesting.

Unfortunately we cannot continue to use this IMU for our developments without manage DMP inialisation with successful.
There are probably additional calibration parameters that we don't know about or we don't know how to disable automatic DMP calibration...

Thanks a lot for your help
Erwan