sparkfun/OpenLog_Artemis #96: Units of Raw Accel data when DMP is enabled

toni-lv, 2021-07-19:
Hello,

While working with data output from DMP, I found up that units are not same as when not enabled (steady values at same position during more than a minute):
- No DMP when Z axis is pointing upwards (aX, aY, aZ): -235.84,-172.36,-965.33
- DMP when Z axis is pointing upwards (RawAX,RawAY,RawAZ): 30,-30,8178

As you can see firs units are as described in docs mG, but I do not know which units are being using with raw accelerometer data.
Do you have any ideas?

PaulZC, 2021-08-08:
Hi @toni-lv ,

Apologies for the very slow reply.

All I can say is that there are still many things we do not understand about the DMP and you are not the only person to have noticed this issue. Please see [this post](https://forum.sparkfun.com/viewtopic.php?p=227711#p227711) on the forum.

It does appear to be a scaling issue. There is something slightly strange about using 0x04000000 for ACC_SCALE to convert 4g to 2^25 (0x02000000). But 0x04000000 is the value from the InvenSense application note, and the value used by the InvenSense example...

I wish I had a better answer for you, but unfortunately I do not...

Very best wishes,
Paul

toni-lv, 2021-08-08:
Thanks a lot for your help.

PaulZC, 2021-08-29:
Hi @toni-lv ,

I am going to close this issue as I think it has gone stale? But please re-open if you need more help with this.

Very best wishes,
Paul

closed 2021-08-29 by PaulZC