sparkfun/OpenLog_Artemis #97: ICM-20948 support via I2C

joseangeljimenez, 2021-08-08:
Hello,

First, I must thank you for having created a great piece of hardware (SparkFun OpenLog Artemis) and firmware (OpenLog for Artemis). We have purchased about 10 boards and all of them work flawlessly.

Context
- I have developed a small prototype with an OpenLog Artemis board connected with a Qwiic cable to a small circular board (custom made) which includes two sensors: pressure (MS5837) and orientation (ICM-20948).

I have been looking into the source code and documentation of the ICM-20948. So far the prototype is working great using the OpenLog Artemis on board ICM-20948. However, I excepted to use the external ICM-20948 via I2C (instead SPI). Is this actually supported? Can you hint me on how to modify the source code to change from SPI to I2C for the ICM-20948?

Thanks for your great support,
Jose

PaulZC, 2021-08-08:
Hi José (@joseangeljimenez ),

Changing the OpenLog Artemis firmware to log data from an ICM-20948 via I2C would be difficult. The OLA firmware talks to the on-board ICM over SPI. There is completely different, complex code which builds up a linked-list of all of the devices connected to the Qwiic port so their data can be read. Adding the ICM to that list is feasible, but would be a lot of work. There would be many things to add, including a whole new configuration menu. Adding in the DMP would make things even more complicated. So, what you are asking for is feasible but is difficult and not something I can do soon. I will be concentrating on adding BLE support on the OLA next.

You may be able to write your own logging code using the ICM-20948 [example](https://github.com/sparkfun/OpenLog_Artemis/blob/main/Firmware/Test%20Sketches/OLA_IMU_Basics/OLA_IMU_Basics.ino) for the OpenLog Artemis. However it looks like that code might be out-of-date and may not work correctly with v2.1 of the Apollo3 core. I will also need to update the [SD example](https://github.com/sparkfun/OpenLog_Artemis/tree/main/Firmware/Test%20Sketches/SD/BasicFileTest) so it will run on v2.1 of the core. Adding the MS5837 will be straight forward, the library for that contains its own examples.

I will take an action to update all the examples so they work correctly on v2.1 of the core. I would hope to be able to do that within 1-2 weeks (I have a lot of other projects going on at the moment!). Apologies, but that's the best I can do.

Very best wishes,
Paul

PaulZC, 2021-08-18:
Hi José (@joseangeljimenez ),

I have updated the stand-alone examples so they will work with v2.1 of the Apollo3 (Artemis) core. One of the examples - Sensor_Autodetect \ AutoDetectWithMux - contains code which will detect an ICM20948 on I2C and print the accel, gyro, mag and temperature to the serial console. It does not log to SD card; the TIM_TM2_Data_Logging example contains SD-logging code which you could re-use.

This is not a complete solution for you, but at least you now have a good starting point for writing your own code.

Very best wishes,
Paul

closed 2021-08-18 by PaulZC